espressif/esp-at #598: 请问，我这边用AT+CIPSTART=”SSL“去做一个https请求时，使用被动接收模式，为什么最后一包数据会等两分钟后，一点一点上报上来

doudouladou, 2021-12-22:
流程如图所示
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

doudouladou, 2021-12-23:
文件大小为23k

doudouladou, 2021-12-23:
请求的文件大小为23k左右

ustccw, 2021-12-23:
What if you access the https resource by tools? such as `postman`.   
From the log you provide, it seems that that the server has not sent data to the module in last free two minutes (i am not sure about it).

You can get the latest code and enable the mbedtls & lwip (socket & tcp & ip) logs at menuconfig, then see what happened.

doudouladou, 2021-12-23:
[image]
postman 非常快就接收下来了

doudouladou, 2021-12-24:
我这边目前需要提供些什么信息@ustccw

doudouladou, 2021-12-27:
emmm，大佬帮忙看一下

ustccw, 2021-12-27:
what if active mode ([`AT+CIPRECVMODE=0`](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#at-ciprecvmode-query-set-socket-receiving-mode)), is the same result?

btw, as i mentioned before, you can get the latest code and enable the mbedtls & lwip (socket & tcp & ip) logs at menuconfig, then see what happened.

doudouladou, 2021-12-27:
ok， 我这边先试一下AT+CIPRECVMODE=0

doudouladou, 2021-12-27:
另外麻烦问下在哪里获取最新的代码，日志在哪里看，这方面有没有教程

ustccw, 2021-12-27:
latest code: https://github.com/espressif/esp-at
documentation: https://docs.espressif.com/projects/esp-at/en/latest/?badge=latest

doudouladou, 2021-12-27:
好的，谢谢你

doudouladou, 2021-12-27:
我这边用AT+CIPRECVMODE=0试了一下，全部24k数据在2s内收取下来了

ustccw, 2021-12-27:
okey, plz try with the latest code.

doudouladou, 2021-12-27:
@ustccw 大佬，我合成固件失败了，你能帮我合个最新代码的固件吗

doudouladou, 2021-12-27:
@ustccw 大佬，我这边测试好了，稍后提供log和AT交互

doudouladou, 2021-12-27:
[18_52_07_127---18_56_12_964 AT交互log.TXT](https://github.com/espressif/esp-at/files/7780001/18_52_07_127---18_56_12_964.AT.log.TXT)
[18_52_07_127---18_56_12_964 esp32log.TXT](https://github.com/espressif/esp-at/files/7780002/18_52_07_127---18_56_12_964.esp32log.TXT)

18：52：07 建立链接
18：52：12 发送数据请求
18：52：16 收到第一包数据
总共收到3包数据后， 18：52：28读完第三包上来的数据

 18：52：28 开始卡住，没有响应

18：54：16开始有数据上报，每次31字节
18：56：13收完全部数据

ustccw, 2021-12-27:
@doudouladou what is your `AT+GMR`.

btw, there are three useful ways to debug it.
- use `TCP` instead of `SSL`, it can show the packets from the log as much as clear.
- capture the TCP packets by `wireshark` or `omnipeek` or other tools.
- send `AT+CIPRECVLEN?` to query the actual tcp data when AT blocks

doudouladou, 2021-12-28:
[image]

doudouladou, 2021-12-28:
这个是使用AT+CIPSTART="TCP"来建立链接请求数据

esp32的log
[09_56_29_554---09_57_16_113   ESP32 LOG.TXT](https://github.com/espressif/esp-at/files/7782325/09_56_29_554---09_57_16_113.ESP32.LOG.TXT)

AT交互
[09_56_29_554---09_57_16_113  TCP AT交互.TXT](https://github.com/espressif/esp-at/files/7782326/09_56_29_554---09_57_16_113.TCP.AT.TXT)

ustccw, 2021-12-28:
From the logs you provided, you can search the string "`|  106  |  227  |   20  |  214  | (src)`", it is the packet from the ssl server. 
The matching 32nd and 33rd packets, their time gap is more than indeed one minute+ apart (10:04:58.257 - 10:03:29.183).

I think the root cause is that the reading speed (`AT+CIPRECVDATA`) is far lower than the network throughput speed, resulting in the low throughput or retransmission of the server. The low throughput or retransmission time may increase exponentially, so the last data will be received for a long time.
For example, the first packet received at `10:02:58.733`, but read it at `10:03:24.008`, this may cause the server to start the low throughput or retransmission.

You can try to test sending `AT+CIPRECVDATA=6000` every 100ms, let's see if the issue still exists.

doudouladou, 2021-12-28:
@ustccw 问题依旧存在，稍后我发下log与AT交互日志

doudouladou, 2021-12-28:
@ustccw 如下log所示
在11:38:02.149 建立SSL链接
11:38:07.506 发送请求数据
11:38:08.081 收到第一包数据
11:38:09.282 开始 每100ms发送一次AT+CIPRECVDATA=6000
11:38:13.972 卡住
11:40:07.936 开始收到后面的数据，每次31字节
11:40:10.251 开始 每100ms发送一次AT+CIPRECVDATA=6000
11:40:19.775 收完全部数据，链接结束

AT交互log
[11_38_02_149---11_40_19_855 AT交互log.TXT](https://github.com/espressif/esp-at/files/7782636/11_38_02_149---11_40_19_855.AT.log.TXT)

esp32 log
[11_38_02_149---11_40_19_855 esp32 log.TXT](https://github.com/espressif/esp-at/files/7782637/11_38_02_149---11_40_19_855.esp32.log.TXT)

doudouladou, 2021-12-29:
@ustccw 大佬帮忙看看

doudouladou, 2021-12-30:
@ustccw 大佬，帮看一下，急

ustccw, 2022-01-04:
@doudouladou i think you should ask the server why data comes to the client at such intervals.
From the logs, Line:2616, it shows server send this packet at 11:38:13.676, but the next packet after it from the server is at Line:5591 (11:40:07.782).

i think the better way for this issue, is that add someone who maintain the server and look into this case together.

doudouladou, 2022-01-04:
我换了个方式进行测试
我使用两种数据接受模式，SSL连接去请求百度
CIPRECVMODE=0的时候，10495个字节能完全接收下来
CIPRECVMODE=1的时候，能遇见两种情况，

ustccw, 2022-01-17:
@doudouladou We reproduced this issue, and will look into it in one month, for we are busy with another project.

doudouladou, 2022-01-29:
@ustccw 好的

doudouladou, 2022-03-16:
@ustccw 大佬，问题分析有进展吗

ustccw, 2022-03-16:
thanks for your patience, will look into it this week.

ustccw, 2022-03-22:
@doudouladou sorry that it's not an easy issue from my test, i'm trying to figure it out.

closed 2022-03-24 by igrr

doudouladou, 2022-03-25:
@ustccw  我试一下

ustccw, 2022-04-06:
@doudouladou have you tested? if the issue is still exists, please feel free to reopen this issue.

doudouladou, 2022-04-06:
@ustccw 被隔离了。。要晚点

doudouladou, 2022-04-15:
@ustccw 我试了，还是有一样的问题
[image]

ustccw, 2022-04-15:
could you please provide the server info and all the commands you input. i'll test your server asap.

doudouladou, 2022-04-15:
稍等

doudouladou, 2022-04-15:
[test.zip](https://github.com/espressif/esp-at/files/8494303/test.zip)

doudouladou, 2022-04-15:
在如上文件夹里

doudouladou, 2022-04-15:
https://cdn.openluat-luatcommunity.openluat.com/attachment/20211222114100362_WIFI_2.0.1_Luat_V0040_8955_SSL_TTS.bin

ustccw, 2022-04-19:
okey, let me check

doudouladou, 2022-07-06:
@ustccw 大佬，大佬，有找到问题吗

doudouladou, 2022-07-06:
读到的

ustccw, 2022-07-11:
由于 issue 被关闭，后面被错误忽略了。

这周晚点我将复现看下~

doudouladou, 2022-07-11:
好的

ustccw, 2022-07-12:
@doudouladou 我用最新的 ESP32-C3 固件测试了下，似乎没什么问题唉。发的命令依次是：
```
AT+RESTORE
AT+CWMODE=1
AT+CWJAP="688018.sh",""
AT+CIPRECVMODE=1
AT+CIPSTART="SSL","cdn.openluat-luatcommunity.openluat.com",443
AT+CIPSEND=233
// HTTP 请求见压缩文件夹里：hex-233bytes.bin

AT+CIPRECVDATA=6000
// 连续发送这个命令，会依次接收完这个文件，见压缩包里：cutecom.log
```

[esp32c3-https-test.zip](https://github.com/espressif/esp-at/files/9091271/esp32c3-https-test.zip)

你可以用压缩包的 ESP32-C3 固件 target.bin 测试看看吗，从 0x0 地址烧录。如果问题还存在，可以保存下串口的日志，发下。