espressif/esp-at #599: 采用SPI/SDIO接口的AT固件下载

wangli631, 2022-01-04:
你好，我尝试在VSCode中编译AT固件，但是很不幸，他失败了。请问我可以在什么地方下载到以及编译好的基于SPI/SDIO接口的AT固件

xiaokang-rays, 2022-01-05:
目前没有看到有发布除串口之外的二进制固件。
最好在Ubuntu下编译吧，windows环境的VSCode我也尝试过，也失败了。
Ubuntu先搭一个ESP-IDF的环境，一定要和你的ESP-AT对应的版本相同，ESP32-C3是4.3.0，我的用的ESP32-WROOM是4.2.0，后面再参考它这个编译就可以。
https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html

ustccw, 2022-02-09:
thanks for the feedback, i'll add this feature soon.

closed 2022-02-10 by igrr

ustccw, 2022-02-10:
@wangli631 @xiaokang-rays Please download the firmware from GitHub Actions: https://github.com/espressif/esp-at/actions/runs/1822235111#artifacts

You can take a look if there is any download issue: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_download_the_latest_temporary_version_of_AT_from_github.html