espressif/esp-at #600: Linux laptop build esp-at for esp32s2 processor using ESP-IDF 4.3.2 error (IDFGH-6553)

gregrasche, 2022-01-05:
build esp-at for esp32s2 processor using ESP-IDF 4.3.2

**Describe the bug**
I am trying to follow this guide to build esp-at for esp32s2 processor
https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp32c3/Compile_and_Develop/How_to_clone_project_and_compile_it.html     

ESP-IDF 4.2 was the first support of esp32s2.  The esp32s2 was chosen to use in our product.  4.3 has bug fixes to 4.2.  ESP-IDF version 4.3.2 was released in December 2021, so we went with that for the most bug fixes.  We wanted to stay away from ESP-IDF 4.4 as it is Beta now.

I need to build the 'AT Commands' project.  It calls out the use of 4.2 and when I try to build with 4.3.2 I get the following error:
  ./build.py menuconfig
  Traceback (most recent call last):
    File "./build.py", line 25, in <module>
      import xlrd
  ModuleNotFoundError: No module named 'xlrd'

Can I use ESP-IDF 4.3.2 to build the AT Command project for the esp32s2?  Is that a supported configuration?

**To Reproduce**
Steps to reproduce the behavior:
1. get ESP-IDF 4.3.2 
2. get esp-at project
3. run  ./build.py menuconfig
4. See error
~/esp/esp-at$ ./build.py menuconfig
Traceback (most recent call last):
  File "./build.py", line 25, in <module>
    import xlrd
ModuleNotFoundError: No module named 'xlrd'

**Expected behavior**
Build works.

**Environment Information:** 
IDF_TOOLS_EXPORT_CMD=/home/greg/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/home/greg/esp/esp-idf/install.sh
IDF_PYTHON_ENV_PATH=/home/greg/.espressif/python_env/idf4.3_py3.6_env
IDF_PATH=/home/greg/esp/esp-idf

## Environment

- Development Kit:      esp32-s2 on DevKitM
- Kit version DevKitM
- Module or chip used:  esp32-s2
- IDF version (run ``git describe --tags`` to find it):
v2.3.0.0_esp32c3-6-gc453e6b
- Build System:         [Make|CMake|idf.py]
./build.py menuconfig
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

gregrasche, 2022-01-05:
I just tried ESP-IDF 4.2 and it fails also.

~/esp/esp-at$ ./build.py menuconfig
Traceback (most recent call last):
  File "./build.py", line 25, in <module>
    import xlrd
ModuleNotFoundError: No module named 'xlrd'

gregrasche, 2022-01-06:
Is the esp32s2 being abandoned, where the esp32c3 is suggested instead?

xcguang, 2022-01-06:
Yes, ESP32-C3 is more low cost.

gregrasche, 2022-01-07:
We have chosen the esp32s2 MINI because it has more I/O and a camera DVP interface. We cannot use the c3 MINI

This https://github.com/espressif/esp-at   says that ESP-AT will regularly update the release/v2.2.0.0_esp32 branch for ESP32-S2. 

Questions:
1) are we safe going with esp32s2 and release/v2.2.0.0_esp32 if it supports the AT command we need?
2) which esp-idf is ep-at compatible with?
3) should we use release/v2.1.0.0_esp32s2, rather than release/v2.2.0.0_esp32 ?

closed 2022-01-22 by gregrasche