espressif/esp-at #602: Can we rely on AT+GMR to detect connected Espressif device?

MaJerle, 2022-01-14:
For my LwESP library, I try to detect connected device using AT commands.
So far I can do this by using `AT+GMR` command which returns `AT version` line, containing name of the Espressif device.

```c
/* 
 * AT version example string looks like (AT+GMR):
 *
 * ESP8266: "AT version:2.2.1.0(46d6c26 - ESP8266 - Aug  6 2021 06:50:15)"
 * ESP32C3: "AT version:2.3.0.0(e98993f - ESP32C3 - Dec 23 2021 09:03:35)""
 * ESP32: "AT version:2.2.0.0(c6fa6bf - ESP32 - Jul  2 2021 06:44:05)
 */
if (strstr(rcv->data, "- ESP8266 -") != NULL) {
    esp.m.device = LWESP_DEVICE_ESP8266;
} else if (strstr(rcv->data, "- ESP32 -") != NULL) {
    esp.m.device = LWESP_DEVICE_ESP32;
} else if (strstr(rcv->data, "- ESP32C3 -") != NULL || strstr(rcv->data, "- ESP32-C3 -") != NULL) {
    esp.m.device = LWESP_DEVICE_ESP32_C3;
} else {
    /* Error */
}
```

Ultimate question is, is this reliable way to determine connected Espressif device or shall it be done different way instead?

ustccw, 2022-01-17:
@MaJerle If only to distinguish whether it is an ESP chip, I think it is pretty feasible. see [AT+GMR](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Basic_AT_Commands.html#cmd-gmr) doc for more details.

btw, this cannot distinguish whether it is an ESP module or not, because some customers only use ESP chips to produce their own modules.

MaJerle, 2022-01-17:
@ustccw it would be to distinguish which ESP chip is connected, either ESP32, ESP32-C3, ESP8266, ...
Based on that, library knows if certain features are possible (or not), for example BLE, or AT+CWSTATE, etc.

If it is, or not, a module, is not important for the moment.

ustccw, 2022-01-17:
@MaJerle you can  send [AT+CMD](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Basic_AT_Commands.html#cmd-cmd) command to list all AT commands and types supported in current firmware.

About the version print, yes,  you can rely on this data format to get device info.

MaJerle, 2022-01-17:
Thanks, all clear.

closed 2022-01-17 by MaJerle