espressif/esp-at #604: SPI HANDSHAKE signal for esp-at on esp32c3

gregrasche, 2022-01-22:
In this DOC
https://github.com/espressif/esp-at/tree/master/examples/at_spi_master/spi/esp32_c_series  

The following signals  are called out for the esp32c3

Master(ESP32C3)
SCLK                      GPIO6
MISO                     GPIO2
MOSI                     GPIO7
CS                          GPIO10
HANDSHAKE         GPIO3

I am implementing a SPI interface on a different processor (STM) that will use the ESP provided AT commands over a SPI bus.  What is the propose of the HANDSHAKE signal? How is it used?

ustccw, 2022-02-09:
@xcguang PTAL

ustccw, 2022-02-09:
@xcguang PTAL

ESP-Coco, 2022-02-09:
Hi @gregrasche  please refer to this document : https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_implement_SPI_AT.html

closed 2022-02-12 by gregrasche