espressif/esp-at #606: factory builds for ESP8266 incompatible with ESP-01 modules

cm68, 2022-02-01:
## Environment

- Development Kit:      none
- Module or chip used: ESP-01
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         USB

## Problem Description
the single most common ESP module, the outrageously prolific ESP-01, available for years for less than a dollar US,
does not have a working build target. 

there are 2 issues:

1. the build target for the 8266, which presupposes a 2Mbyte flash chip, is too big for both the common 512KB and 1MB flash devices.
2. ESP-01 modules use a 26Mhz oscillator, so the bootloader messages and AT commands are at 72000 Baud, even if the ESP-01 has been ugraded with a larger flash device.

### Expected Behavior
flashing a ESP-01 acually produces a working ESP-01

### Actual Behavior
the ESP-01 build target is too large at 2MB
the ESP-01 build target interacts with the serial port at 72000 baud due to crystal mismatch

### Steps to repropduce

1. download factory_WROOM-02.bin from ESP8266-IDF-AT_V2.2.1.0.zip
2.  esptool.py --chip esp8266 --port /dev/ttyUSB0 --no-stub write_flash 0 factory_WROOM-02.bin

[image]
[image]

## Debug Logs

```
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c8:2b:96:00:36:ae
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x001fffff...
Erasing flash...
Wrote 2097152 bytes at 0x00000000 in 203.9 seconds (82.3 kbit/s)...

Leaving...
Hard resetting via RTS pin...
```

connecting via a terminal program is useless, and 2 above oscilloscope screenshots show that the baud rate 
is around 72000.  given that the crystal on the module is 26mhz, I would expect code built for a 40 Mhz crystal
to produce a baud rate of 74880, which is what we see.

```

ustccw, 2022-02-09:
Issue 1: there is a 1MB choice for ESP8266. please switch to `release/v2.2.0.0_esp8266` branch, and choose `Platform name` PLATFORM_ESP8266, `Module name` ESP8266_1MB in the compilation phase. the firmware of `ESP8266-IDF-AT_V2.2.1.0.zip` is for WROOM-02 module, not `ESP-01`.

Issue 2: `ESP8266/ESP8285` uses the 26 MHz oscillator by default. you can config it at `./build.py menuconfig` -> `Component config` -> `ESP8266-specific` -> `Main XTAL frequency`. 
if ESP-01 modules also use a 26Mhz oscillator, please check your UART configurations.

jandrassy, 2022-02-09:
[https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/files/ESP8266-1MB-tx1rx3-AT_V2.2.zip](https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/files/ESP8266-1MB-tx1rx3-AT_V2.2.zip)

cm68, 2022-02-19:
> https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/files/ESP8266-1MB-tx1rx3-AT_V2.2.zip
>
> thank you for building this and solving my immediate problem. how did you
> build it?
>
> I read the manual :-)

which manual?  there are so many docs, most of which refer to the ESP32 and
say nothing of the 8266.

which compiler suite did you use? on which platform?  what version?

on my ubuntu box, when I try to run menuconfig, it fails.  the esp8266 is
not well supported, certainly not well documented.
if you can point me to a document that says to do a git clone using this
url, etc, that would be great.  until then, maybe a little
less gate keeping would be better for the community.

jandrassy, 2022-02-19:
> which manual?

the one linked on the main page of this repository.

cm68, 2022-02-20:
Either.

the build is broken.

> ***@***.***:espressif/ESP8266_RTOS_SDK.git./build.py menuconfig

module_name ESP8266_1MB

platform_name=ESP8266,module_name=ESP8266_1MB
Setting IDF_PATH environment variable: /home/curt/src/esp/esp-at/esp-idf
Checking Python dependencies...
Python requirements from /home/curt/src/esp/esp-at/esp-idf/requirements.txt
are satisfied.
Executing action: menuconfig
Running make in directory /home/curt/src/esp/esp-at/build
Executing "make -j 18 menuconfig"...
[  0%] Performing build step for 'mconf-idf'
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make
rule.
make[4]:
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/check-lxdialog.sh:
No such file or directory
In file included from
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/mconf.c:23:
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:38:10:
error: #include expects "FILENAME" or <FILENAME>
   38 | #include CURSES_LOC
      |          ^~~~~~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:103:9:
error: unknown type name ‘chtype’
  103 |         chtype atr;     /* Color attribute */
      |         ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:200:16:
error: unknown type name ‘WINDOW’
  200 | int on_key_esc(WINDOW *win);
      |                ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:221:17:
error: unknown type name ‘WINDOW’
  221 | void attr_clear(WINDOW * win, int height, int width, chtype attr);
      |                 ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:221:54:
error: unknown type name ‘chtype’
  221 | void attr_clear(WINDOW * win, int height, int width, chtype attr);
      |                                                      ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:223:21:
error: unknown type name ‘WINDOW’
  223 | void print_autowrap(WINDOW * win, const char *prompt, int width,
int y, int x);
      |                     ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:224:19:
error: unknown type name ‘WINDOW’
  224 | void print_button(WINDOW * win, const char *label, int y, int x,
int selected);
      |                   ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:225:18:
error: unknown type name ‘WINDOW’
  225 | void print_title(WINDOW *dialog, const char *title, int width);
      |                  ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:226:15:
error: unknown type name ‘WINDOW’
  226 | void draw_box(WINDOW * win, int y, int x, int height, int width,
chtype box,
      |               ^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:226:66:
error: unknown type name ‘chtype’
  226 | oid draw_box(WINDOW * win, int y, int x, int height, int width,
chtype box,
      |
^~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:227:15:
[... 42 lines cut ...]
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/mconf.c:23:
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/mconf.c: In function
‘conf_choice’:
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:97:29:
error: ‘KEY_MAX’ undeclared (first use in this function)
   97 | #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
      |                             ^~~~~~~
/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/mconf.c:855:23: note: in
expansion of macro ‘ERRDISPLAYTOOSMALL’
  855 |                 case -ERRDISPLAYTOOSMALL:
      |                       ^~~~~~~~~~~~~~~~~~
      |                       ^~~~~~~~~~~~~~~~~~
make[4]: ***
[/home/curt/src/esp/ESP8266_RTOS_SDK/tools/kconfig/Makefile:173: mconf.o]
Error 1
make[3]: *** [CMakeFiles/mconf-idf.dir/build.make:146:
mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:1811: CMakeFiles/mconf-idf.dir/all]
Error 2
make[1]: *** [CMakeFiles/Makefile2:1845: CMakeFiles/menuconfig.dir/rule]
Error 2
make: *** [Makefile:430: menuconfig] Error 2
make failed with exit code 2
idf.py build ret: 2
idf.py build failed