espressif/esp-at #608: AT+BLEGATTCWR always writes same wrong value to CCCD

Staibii, 2022-02-10:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32|ESP32-C3-WROOM]
- Power Supply:         [external 3.3V]

## Problem Description
I am using two ESP modules communicating over BLE, sending AT commands via UART. One modul as BLE Client[ESP32-WROOM-32] and the other as Server[ESP32-C3-WROOM]. Everything works fine until I want to enabel notification or indication. Therefore I want to write 2 Bytes of Data to the corresponding CCCD, using the AT+BLEGATTCWR command. The Problem is that no matter how many Bytes or whiche value I send , the operation always writes the same two Bytes ('1''0') to the CCCD on the server side.   

### Expected Behavior
#### Client side:
Sending: AT+BLEGATTCWR=0,4,1,1,2\r\n 
Receiving: >
Sending: '2''0' (note that '2''0' are integers and not ASCII values)

#### Server side:
Receiving: +WRITE:0,2,1,1,2,'2''0'\r\n

### Actual Behavior
#### Client side:
Sending: AT+BLEGATTCWR=0,4,1,1,2\r\n 
Receiving: >
Sending: '2''0'

#### Server side:
Receiving: +WRITE:0,2,1,1,2,'1''0'\r\n
### Steps to repropduce

1.step : init server and client
2.step: start services on server side
3.step: connect divices
4.step: write to CCCD 

## Other items if possible
[BleAllMightyService.csv](https://github.com/espressif/esp-at/files/8040665/BleAllMightyService.csv)

- [ ] Service csv file

Luxin812, 2022-04-14:
Whether the Service characteristic descriptor you want to write only supports the notify.

Luxin812, 2022-04-14:
If your service feature description only support the notify, it will automatically modify the value you write to 0x0001.

closed 2023-06-05 by ustccw