espressif/esp-at #609: Commands that give wifi connection information do not work properly

keeturan, 2022-02-11:
AT+CWSTATUS & +CIPSTATUS Commands that give wifi connection information do not work properly.
I am using latest fw esp8266 at bin 2.2.1.
i disconnect wifi it shows still tcp socket created status (some wifi drops during uplink operations*).

keeturan, 2022-02-11:
CWSTATE & CIPSTATUS

ustccw, 2022-02-14:
As the [docs](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html?highlight=CIPSTATUS#id4) said, It is recommended to use [AT+CWSTATE](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-wstate) command to query Wi-Fi state and [AT+CIPSTATE](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html?highlight=CIPSTATUS#cmd-ipstate) command to query TCP/UDP/SSL state.

`AT+CIPSTATUS` command is deprecated.

closed 2023-06-05 by ustccw