espressif/esp-at #610: esp32-s2-mini-1 AT+HTTPCLIENT command - can't send any json data

ortros1107, 2022-02-17:
Greetings!

I have a problem. Whenever I'm trying to send a data through httpclient at command using json, I get an error (sending with the other content types seems to work - I've send data using urlencoded and xml content-types). The only thing that can be send is an empty json object. I've also tried (among other things) to send json data using the examples from the official documentation but to no avail. Otherwise I'm using esp32-s2-mini-1 version 2.1.0.0. Thank You in advance for answers!

ustccw, 2022-03-10:
@ortros1107 have you escaped the following special characters:

[image]

ustccw, 2022-03-10:
For example, if you wanna send json string like `"{"array": [1,2,3]}"`, so in AT commands, they should be like `"\"{\"array\": [1\,2\,3]}\""`.

ortros1107, 2022-03-10:
First I want to thank You for replying. The problem was (actually) because
I used only one set of double quotation marks (when I should've used two)
when writing json. However, thank You again for your kindness and have a
nice day.

ustccw, 2022-03-11:
no probs.

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

ortros1107, 2022-06-02:
I haven't tested it yet, but as soon as I will, I will tell You about it.

ustccw, 2023-06-05:
We assume this worked for you, and will close it. If that's not the case, please open another github issue.

closed 2023-06-05 by ustccw

ortros1107, 2023-06-05:
Yes it did, thank you very much. I'm sorry for not answering sooner. You
can close it whenever You want.