espressif/esp-at #611: ESP-WROOM-02 is not responding to AT commands as expected Firmware v2.2.1.0

bmalbusca, 2022-02-17:
I using the ESP-WROOM-02 (esp8266 core) as WIFI module and the samd21g18 as main processing unit. To test, I´m using the ESP-WROOM-02 DEV board kitc and an Arduino nano IOT which has also a samd21g18 MCU. I flashed the ESP-WROOM-02 via the ESP flash download tool with the [AT Firmware v2.2.1.0][1] . The user guide highlights that the default ESP8266 RTOS AT firmware for ESP-WROOM-02 swaps RX/TX with CTS/RTS.

[![pin connections][2]][2]

For testing the UART, I used the Arduino Serial monitor and the UART connected to the on-board FTDI. To link the new pins (IO15 and IO13) , I connected the pin RX0 to IO13 and the TX to IO15 as a bypass.
When I sent any AT command (or any thing that I wrote on serial) the ESP always returned an empty line and a new line saying ready. Fist I tough that was the CR and line ending, I tried all combinations a didn´t worked.

[![serial com][3]][3]

To understand was happening I connected a logic analyser to RX0 in order to check if was the same as what I was sending via serial monitor. And it was sending right.

Then, I connected the logic to the IO15 (ESP TX). But something appeared wrong. The first message is missing a bit.

[![MISO reading][4]][4]
[![First word wrong][5]][5]

  [1]: https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/AT_Binary_Lists/ESP8266_AT_binaries.html
  [2]: https://i.stack.imgur.com/icxrH.png
  [3]: https://i.stack.imgur.com/c98Oc.png
  [4]: https://i.stack.imgur.com/SJdXL.png
  [5]: https://i.stack.imgur.com/Vjwew.png

jandrassy, 2022-02-17:
I don't understand why you bring it here. did you try the AT commands on Serial1 of the Nano?

ustccw, 2022-03-10:
@bmalbusca i dont think i understand this totally: `To link the new pins (IO15 and IO13) , I connected the pin RX0 to IO13 and the TX to IO15 as a bypass.` GPIO13 is ESP8266's RX, so should we connect it to peer TX, so does GPIO15.

Could you please double check your hardware connection according to https://espressif-docs.readthedocs-hosted.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Get_Started/Hardware_connection.html#id16