espressif/esp-at #612: 透传模式下遇到 0x00 则接收数据结束，造成数据丢失。

xuhongv, 2022-02-23:
- 分支：release/v2.2.0.0_esp8266
- commitId: 248fbd060cc2aadfa15312108fd838a2c6e093ba
- 问题描述：在自定义AT指令实现透传时候，成功进去透传模式后，发送Hex数组带有 00 元素时候，则自动丢弃后面的数据，比如发送 [20,63,F2,00,FC,69] ，模块则只收到 [20,63,F2] 。
- Code:
```

 static esp_at_cmd_struct at_custom_cmd[] = {
        {"+XU", NULL, NULL, NULL, at_setup_cmd_smqttpub},

    };
    esp_at_custom_cmd_array_regist(at_custom_cmd, sizeof(at_custom_cmd) / sizeof(at_custom_cmd[0]));
   

#define BUFFER_LEN (2048)
static xSemaphoreHandle at_sync_sema = NULL;
void wait_data_callback(void)
{
    xSemaphoreGive(at_sync_sema);
}

uint8_t at_setup_cmd_smqttpub(uint8_t *cmd_name)
{

    int32_t received_len = 0;
    int32_t remain_len = 0;
    uint8_t *buf = NULL;
    uint8_t buffer[64] = {0};

    buf = (uint8_t *)malloc(BUFFER_LEN);
    if (buf == NULL)
    {
        memset(buffer, 0, 64);
        snprintf((char *)buffer, 64, "malloc failed\r\n");
        esp_at_port_write_data(buffer, strlen((char *)buffer));
    }

    // 示例代码
    // 不必在此处创建 semaphores
    if (!at_sync_sema)
    {
        at_sync_sema = xSemaphoreCreateBinary();
        assert(at_sync_sema != NULL);
    }

    // 返回输入数据提示符 ">"
    esp_at_port_write_data((uint8_t *)">", strlen(">"));

    // 设置回调函数，在接收到输入数据后由 AT 端口调用
    esp_at_port_enter_specific(wait_data_callback);

    // 接收输入的数据
    while (xSemaphoreTake(at_sync_sema, portMAX_DELAY))
    {
        memset(buf, 0, BUFFER_LEN);

        received_len = esp_at_port_read_data(buf, BUFFER_LEN);
        // 检查是否退出该模式
        // 退出条件是接收到 "+++" 字符串
        if ((received_len == 3) && (strncmp((const char *)buf, "+++", 3)) == 0)
        {
            esp_at_port_exit_specific();

            // 示例代码
            // 如果剩余数据长度 > 0，说明缓冲区内仍有数据需要处理
            // 可自定义如何处理剩余数据
            // 此处只是简单打印出剩余数据
            remain_len = esp_at_port_get_data_length();

            if (remain_len > 0)
            {
                esp_at_port_recv_data_notify(remain_len, portMAX_DELAY);
            }

            break;
        }
        else if (received_len > 0)
        {
            ESP_LOG_BUFFER_HEX(TAG, buf, received_len);
            ESP_LOGI(TAG, "[APP] Free memory: %d bytes [%d]", esp_get_free_heap_size(), strlen((char *)buf));
        }
    }

    free(buf);

    return ESP_AT_RESULT_CODE_OK;
}
```

Alson-tang, 2022-02-23:
[image]
[image]

您好，我使用你的代码进行测试（仅仅加了数据打印），没有发现有丢数据的情况。我附上我 AT 的测试指令序列，左边是系统 log 口，右边是 AT 口。请您看下是否我哪里测试有问题？

xuhongv, 2022-03-05:
已解决，谢谢

closed 2022-03-05 by xuhongv