espressif/esp-at #613: SPI AT 速率ceshi

wangli631, 2022-02-23:
在测试SPI-AT固件的速率时，手册上提到将streambuffer 的大小设置为8192 字节，但是我没有找到streambuffer的位置。此外，在不修改的情况下，请问streambuffer的大小是多少？
[image]

Jacques-zhao, 2022-03-10:
HI,默认的streambuffer是4096  https://github.com/espressif/esp-at/blob/master/main/interface/hspi/Kconfig#L83

wangli631, 2022-03-10:
> HI,默认的streambuffer是4096 https://github.com/espressif/esp-at/blob/master/main/interface/hspi/Kconfig#L83

谢谢你的回复，如果我想要加大streambuffer，我只需要将这里的4096改为8192即可，其他任何地方都不需要改动，我这样理解对吗？
[image]

Kevincoooool, 2022-03-14:
你好，你的spi at固件是esp32的吗，是直接编译出来的吗，我这编译出错了能否请教一下

wangli631, 2022-03-15:
> 你好，你的spi at固件是esp32的吗，是直接编译出来的吗，我这编译出错了能否请教一下

Please download the firmware from GitHub Actions: https://github.com/espressif/esp-at/actions/runs/1822235111#artifacts

You can take a look if there is any download issue: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_download_the_latest_temporary_version_of_AT_from_github.html

wangli631, 2022-03-15:
> 你好，你的spi at固件是esp32的吗，是直接编译出来的吗，我这编译出错了能否请教一下

我用的是esp32-c3,固件是直接采用官方发布的，我也尝试过自己编译，但是也失败了。

Kevincoooool, 2022-03-15:
> Please download the firmware from GitHub Actions: https://github.com/espressif/esp-at/actions/runs/1822235111#artifacts
>
> You can take a look if there is any download issue: [...]

这里全都没有关于ESP32这个型号可用的SPI_AT固件，只有ESP32C3的SPI_AT固件

wangli631, 2022-03-15:
> 这里全都没有关于ESP32这个型号可用的SPI_AT固件，只有ESP32C3的SPI_AT固件

官方不推荐在ESP32上采用SPI接口，可以考虑使用SDIO接口的固件

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

wangli631, 2022-05-27:
> May I know if there is any update? If the issue still exists, please let us know.

you can close this question

closed 2022-05-27 by wangli631