espressif/esp-at #615: ESP32C3 USB-JTAG

GiulioDallaVecchia, 2022-03-04:
Hi team,

I flashed official AT firmware to esp32-C3-mini1 module through USB-JTAG build-in interface. After that I'm not able to connect to the module anymore. Does AT firmware configure or use PINs 18-19?

Thanks a lot!

ustccw, 2022-03-10:
Actually not. The default uart pins esp-at used defined here: https://docs.espressif.com/projects/esp-at/en/latest/Get_Started/Hardware_connection.html#id11
i am not sure what is your real jtag pins, i found:
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html#id1
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#id1

the first one used the same pins as esp-at uart1, i doubt possible pins conflict, please double check.

GiulioDallaVecchia, 2022-03-10:
Hi @ustccw,

I'm using build-in JTAG-USB interface that is on pin IO18/IO19:

-  [https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html](url)

After flashing the official AT command firmware to the module I can't re-flash any other firmware using USB-JTAG build-in. When I try to connect with openocd I always receive an error.

I've already tried to do that with the "hello world" project and I don't get any error when I try to connect the board with openocd after flashing the code. It seems that the AT command firmware reconfigure the GPIO that are used for USB-JTAG connection. Is it possibile?

Thanks a lot in andvance.

ustccw, 2022-03-11:
@GiulioDallaVecchia i dont think AT command firmware reconfigures the GPIO that are used for USB-JTAG connection.
Would you mind drawing a sketch of your hardware connection, and the steps you did, and all logs. I'd be happy to ask my folks to help this issue.

xcguang, 2022-03-11:
it may also be confirmed by enabling driver log

GiulioDallaVecchia, 2022-03-11:
Hi @ustccw,

Below you can see the connections. I'm using USB cable connected to D+/D-. Below you can also see the log as I try to reconnect to the module to flash a new image after flashing the AT command firmware using the same cable and connection.

> C:\Users\gdvecchia\Desktop\esp-idf>openocd -d3 -f board/esp32c3-builtin.cfg -c "init; halt; program_esp AT_Command.bin 0x00000 verify exit"
> Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> User : 13 0 options.c:57 configuration_output_handler(): debug_level: 3
> User : 14 0 options.c:57 configuration_output_handler():
> Debug: 15 10 options.c:181 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
> Debug: 16 10 options.c:182 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
> Debug: 17 20 options.c:183 add_default_dirs(): exepath=C:/Users/gdvecchia/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin
> Debug: 18 20 options.c:184 add_default_dirs(): bin2data=../share/openocd
> Debug: 19 30 configuration.c:42 add_script_search_dir(): adding C:\Users\gdvecchia\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\share\openocd\scripts
> Debug: 20 30 configuration.c:42 add_script_search_dir(): adding C:\Users\gdvecchia\AppData\Roaming/OpenOCD
> Debug: 21 40 configuration.c:42 add_script_search_dir(): adding C:/Users/gdvecchia/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin/../share/openocd/site
> Debug: 22 40 configuration.c:42 add_script_search_dir(): adding C:/Users/gdvecchia/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin/../share/openocd/scripts
> Debug: 23 60 configuration.c:97 find_file(): found C:\Users\gdvecchia\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\share\openocd\scripts/board/esp32c3-builtin.cfg
> Debug: 24 70 configuration.c:97 find_file(): found C:\Users\gdvecchia\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\share\openocd\scripts/interface/esp_usb_jtag.cfg
> Debug: 25 80 command.c:143 script_debug(): command - interface interface esp_usb_jtag
> Debug: 27 80 command.c:355 register_command_handler(): registering 'espusbjtag'...
> Debug: 28 80 command.c:355 register_command_handler(): registering 'espusbjtag'...
> Debug: 29 90 command.c:355 register_command_handler(): registering 'espusbjtag'...
> Debug: 30 90 command.c:355 register_command_handler(): registering 'espusbjtag'...
> Info : 31 90 transport.c:117 allow_transports(): only one transport option; autoselect 'jtag'
> Debug: 32 100 command.c:355 register_command_handler(): registering 'jtag_flush_queue_sleep'...
> Debug: 33 100 command.c:355 register_command_handler(): registering 'jtag_rclk'...
> Debug: 34 110 command.c:355 register_command_handler(): registering 'jtag_ntrst_delay'...
> Debug: 35 110 command.c:355 register_command_handler(): registering 'jtag_ntrst_assert_width'...
> Debug: 36 110 command.c:355 register_command_handler(): registering 'scan_chain'...
> Debug: 37 120 command.c:355 register_command_handler(): registering 'jtag_reset'...
> Debug: 38 120 command.c:355 register_command_handler(): registering 'runtest'...
> Debug: 39 120 command.c:355 register_command_handler(): registering 'irscan'...
> Debug: 40 130 command.c:355 register_command_handler(): registering 'verify_ircapture'...
> Debug: 41 130 command.c:355 register_command_handler(): registering 'verify_jtag'...
> Debug: 42 130 command.c:355 register_command_handler(): registering 'tms_sequence'...
> Debug: 43 140 command.c:355 register_command_handler(): registering 'wait_srst_deassert'...
> Debug: 44 140 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 45 140 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 46 150 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 47 150 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 48 150 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 49 160 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 50 160 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 51 160 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 52 170 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 53 170 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 54 170 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 55 180 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 56 180 command.c:355 register_command_handler(): registering 'jtag'...
> Debug: 57 180 command.c:355 register_command_handler(): registering 'svf'...
> Debug: 58 190 command.c:355 register_command_handler(): registering 'xsvf'...
> Debug: 59 190 command.c:143 script_debug(): command - adapter_khz adapter_khz 40000
> Debug: 61 190 core.c:1704 jtag_config_khz(): handle jtag khz
> Debug: 62 200 core.c:1667 adapter_khz_to_speed(): convert khz to interface specific speed value
> Debug: 63 200 core.c:1667 adapter_khz_to_speed(): convert khz to interface specific speed value
> Debug: 64 210 configuration.c:97 find_file(): found C:\Users\gdvecchia\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\share\openocd\scripts/target/esp32c3.cfg
[... cut ...]
> Error: 269 990 libusb1_common.c:351 jtag_libusb_get_serial(): libusb_get_string_descriptor_ascii() failed with -9
> Debug: 270 1005 command.c:630 run_command(): Command 'init' failed with error code -4
> User : 271 1005 command.c:695 command_run_line():
> Debug: 272 1005 riscv.c:487 riscv_deinit_target(): riscv_deinit_target()
> Debug: 273 1005 riscv-013.c:1460 deinit_target(): riscv_deinit_target()
> Debug: 274 1015 target.c:2012 target_free_all_working_areas_restore(): freeing all working areas

[image]

ustccw, 2022-03-11:
got it, will update issue if any progress.

igrr, 2022-03-11:
@GiulioDallaVecchia as a debugging step, could you try the following?

1. Put ESP32-C3 into download mode by bringing IO9 (boot) pin low, then toggling the EN pin.
2. Verify that you see "USB JTAG/serial debug unit" in Device Manager.
3. Run OpenOCD again

If OpenOCD connects successfully, then we will be sure that the issue is related to the application (in this case ESP-AT) loaded into the module. If it still fails to connect, it might be some hardware issue.

GiulioDallaVecchia, 2022-03-11:
Hi @igrr,

I've already tried that. When I put the module into download mode OpenOCD connects successfully.

igrr, 2022-03-11:
Thanks for checking this! I think the issue might be caused by the following.

USB and Wi-Fi peripherals in ESP32-C3 share some common hardware (PLL), and in the older versions of ESP-IDF there was a bug that enabling Wi-Fi disabled USB interface. Seems that AT firmware enables Wi-Fi by default.
This issue was fixed in more recent release/v4.3 branch versions  than the one the AT firmware is now using.

You can try to make a simple change in ESP-IDF code to  check if that's indeed the issue.

1. Find this file: https://github.com/espressif/esp-idf/blob/87f8a0d5f11fea174f8a77bf230bb9caf6caf38e/components/esp_wifi/src/phy_init.c#L615-L617
2. Remove the ifdef around `phy_bbpll_en_usb(true);`, making sure that this function is called.
3. Build the application and flash it to the board.
4. Restart the board and check if the USB device is listed in the Device Manager
5. Try running OpenOCD again

GiulioDallaVecchia, 2022-03-11:
Hi @igrr,

There is a missing header file. When I try to compile after modified the file I get the error below.

> i.dir/src/phy_init.c.obj   -c ../esp-idf/components/esp_wifi/src/phy_init.c
../esp-idf/components/esp_wifi/src/phy_init.c: In function 'esp_phy_load_cal_and_init':
../esp-idf/components/esp_wifi/src/phy_init.c:616:5: error: implicit declaration of function 'phy_bbpll_en_usb' [-Werror=implicit-function-declaration]
     phy_bbpll_en_usb(true);
     ^~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[166/1230] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/if_api.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed

igrr, 2022-03-11:
My bad, I didn't notice the declaration is also wrapped with an ifdef. Please remove the ifdef here and try again: https://github.com/espressif/esp-idf/blob/87f8a0d5f11fea174f8a77bf230bb9caf6caf38e/components/esp_wifi/include/phy.h#L97-L102
(this file is already included in phy_init.c)

igrr, 2022-03-11:
Thanks, that's good to know!
The fix is already merged into release/v4.3 branch of ESP-IDF and will be part of the next 4.3.3 bugfix release. I'm not sure when the fix will be picked up by esp-at project, though. Regarding esp-at release plan, I'll leave it for @ustccw to confirm.

It might also be possible to make this work with the current commit of IDF which esp-at is using, without making the changes I have suggested above. To do this, enable `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` in menuconfig. This has the effect of enabling `CONFIG_ESP_PHY_ENABLE_USB` option. That said, I am not sure if enabling `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` will have any side effect in esp-at, so can't guarantee this option works.

ustccw, 2022-03-14:
@GiulioDallaVecchia @igrr thanks for your information and for being so quick!
esp-at will update esp-idf version to release/v4.3 latest in next week, and the next ESP32-C3 AT version (v2.4.0.0) is scheduled to be released in early April.

About `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` and `CONFIG_ESP_PHY_ENABLE_USB`, i'm afraid esp-at will not configure it by default. Because working with USB, the RF performance may be affected. If users wanna debug by JTAG, please enable them in menuconfig.

closed 2022-03-22 by igrr

QGB, 2022-05-08:
```
C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: Unable to execute JTAG queue
Error: Failed to write ESP32C3_TIMG1WDT_PROTECT (-4)!
Error: Failed to disable WDTs (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: Failed to enable EBREAKS handling (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x10, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read dmstatus!
Error: Failed to resume core (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
```

QGB, 2022-05-09:
jtag write_flash success !
```
C:\>cd C:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\blink

C:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\blink>openocd -f board/esp32c3-builtin.cfg -c "program_esp build/bootloader/bootloader.bin 0x0 verify exit"
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [0] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 30 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 73 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Info : PROF: Data transferred in 374.808 ms @ 53.3607 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
shutdown command invoked

C:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\blink>
```

jeremyherbert, 2022-07-17:
Hi @igrr @ustccw - I think this bug is still present somehow. I flashed the ESP32-C3-MINI-1 AT firmware version 2.4.0.0 via native USB on a blank WROOM module (using the binary with modified time: 20 May 2022, from the espressif website - I didn't build it myself). 

Here is the command I used to flash the firmware originally via the USB interface:
```
esptool.py --chip esp32c3 write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0xf000 phy_multiple_init_data.bin 0x0 bootloader/bootloader.bin 0x60000 esp-at.bin 0x1e000 at_customize.bin 0x25000 customized_partitions/server_cert.bin 0x3a000 customized_partitions/mqtt_key.bin 0x27000 customized_partitions/server_key.bin 0x29000 customized_partitions/server_ca.bin 0x2f000 customized_partitions/client_ca.bin 0x31000 customized_partitions/factory_param.bin 0x1F000 customized_partitions/ble_data.bin 0x3c000 customized_partitions/mqtt_ca.bin 0x38000 customized_partitions/mqtt_cert.bin 0x2b000 customized_partitions/client_cert.bin 0x2d000 customized_partitions/client_key.bin
```

After resetting/power cycling the ESP32-C3, when attempting to (for example) read the flash id over native USB, I get the following error:

```
$ esptool.py flash_id
esptool.py v4.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
/dev/ttyACM0 failed to connect: [Errno 71] Protocol error

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

Here is the `lsusb` output while the device is in the bad state (ie after flashing + reset):
```
$ lsusb -v -d 303a:1001
Bus 001 Device 008: ID 303a:1001 Espressif USB JTAG/serial debug unit
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x303a
  idProduct          0x1001
  bcdDevice            1.01
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0
      iInterface              0
      CDC Header:
        bcdCDC               1.10
[... 96 lines cut ...]

I have a USB protocol analyser (Total Phase), and when I hook it up and run `esptool.py` I get the following trace (running esptool corresponds to packet index 189 in this trace):

[image]

(same screenshot but with expanded failed SETUP packets):

[image]

[esp32-c3-dump.csv](https://github.com/espressif/esp-at/files/9127058/esp32-c3-dump.csv)

I can provide the protocol dump in tdc format also if it would help. Also, my system details:

`uname -a`:
```
Linux goose 5.4.0-120-generic #136-Ubuntu SMP Fri Jun 10 13:40:48 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

`lsb_release -a`:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
```

jeremyherbert, 2022-07-17:
I can also replicate this problem on a brand new ESP32-C3-DevKitC-02 by doing the following:

1. Flash AT firmware via CP2102 USB to serial converter
2. Connect USB DP/DM (GPIO19/18) and GND to a different USB port
3. Run `esptool.py flash_id` with the native USB CDC port => **it fails**
4. Run `esptool.py flash_id` with the CP2102 serial port => it works
5. Run `esptool.py erase_flash` with the CP2102 serial port
6. Run `esptool.py flash_id` with the native USB CDC port => **it works again**
7. Run `esptool.py flash_id` with the CP2102 serial port => it works

So the AT firmware is definitely breaking the USB interface.

Enumeration succeeds for the native USB port but esptool cannot communicate with it (nor can it force a reset). Any operation via the CP2102 works fine from my testing.