espressif/esp-at #616: unknown cmake command: __add_uf2_targets (IDFGH-6911)

agarwal-radha, 2022-03-09:
Link: 
"https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html" 
I have followed each step of the above link. I am able to successfully build hello_world example project. 
But, at this step: 'Run the project configuration utility menuconfig to configure.', I am getting the attached output.

ragarwal@ragarwal:~/esp/esp-at$ ./build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
choose(range[1,2]):1

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. MINI-1 (description: ESP32-U4WDH chip inside)
6. ESP32-SDIO
7. ESP32-D2WD (description: 2MB flash, No OTA)
8. ESP32_QCLOUD (description: QCLOUD TX:17 RX:16)
choose(range[1,8]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP32,module_name=WROOM-32
WARNING: IDF_PATH environment variable is set to /home/ragarwal/esp/esp-idf but idf.py path indicates IDF directory /home/ragarwal/esp/esp-at/esp-idf. Using the environment variable directory, but results may be unexpected...
Executing action: menuconfig
Running cmake in directory /home/ragarwal/esp/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/ragarwal/esp/esp-at"...
silence:0 sdkconfig:/home/ragarwal/esp/esp-at/module_config/module_esp32_default/sdkconfig.defaults
CMake Warning at esp-idf/tools/cmake/idf.cmake:18 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build:
                              /home/ragarwal/esp/esp-idf
Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:53 (include)

-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at esp-idf/tools/cmake/idf.cmake:47 (include):
  include could not find load file:

    uf2
Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:53 (include)

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ragarwal/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/ragarwal/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/ragarwal/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/ragarwal/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/ragarwal/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/ragarwal/esp/esp-at/sdkconfig
Loading defaults file /home/ragarwal/esp/esp-at/module_config/module_esp32_default/sdkconfig.defaults...
-- Found PythonInterp: /home/ragarwal/.espressif/python_env/idf4.2_py3.8_env/bin/python (found version "3.8.10") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "esp-at" version: v2.3.0.0_esp32c3-32-g0067e16
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-nano.ld
-- Adding linker script /home/ragarwal/esp/esp-at/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/ragarwal/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
silence:0, ld core:/home/ragarwal/esp/esp-at/components/at/lib/libesp32_at_core.a
Parsing CSV input...
Verifying table...
CMake Error at esp-idf/tools/cmake/project.cmake:503 (__add_uf2_targets):
  Unknown CMake command "__add_uf2_targets".
Call Stack (most recent call first):
  CMakeLists.txt:55 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ragarwal/esp/esp-at/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
idf.py build ret: 2
idf.py build failed

igrr, 2022-03-09:
@agarwal-radha I think the issue may be as indicated in the displayed warning:

Warning at esp-idf/tools/cmake/idf.cmake:18 (message):
IDF_PATH environment variable is different from inferred IDF_PATH.

You may need to unset IDF_PATH before running the esp-at build script.

agarwal-radha, 2022-03-09:
@igrr Thanks for your response. I am able to build firmware to enable ESP AT Classic Bluetooth.
I enabled the commands in the component config mentioned in the below link
Link: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/how_to_enable_at_classic_bluetooth.html

But, I am getting the following output on launching the monitor, seems like the image size doesn't fit. What should I do now?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7664
load:0x40078000,len:16424
load:0x40080400,len:3704
0x40080400: _init at ??:?

entry 0x4008067c
I (27) boot: ESP-IDF v4.3.2-214-g87f8a0d5f1 2nd stage bootloader
I (27) boot: compile time 14:24:23
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (90) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (97) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (113) boot: End of partition tableA app          00 11 00280000 00180000
I (117) boot: No factory image, trying OTA 0
I (122) boot_comm: chip revision: 1, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=2d208h (184840) map
I (204) esp_image: segment 1: paddr=0012d230 vaddr=3ffbdb60 size=02de8h ( 11752) load
I (209) esp_image: segment 2: paddr=00130020 vaddr=400d0020 size=140730h (1312560) map
I (685) esp_image: segment 3: paddr=00270758 vaddr=3ffc0948 size=0174ch (  5964) load
I (688) esp_image: segment 4: paddr=00271eac vaddr=40080000 size=1dc80h (121984) load
I (741) esp_image: segment 5: paddr=0028fb34 vaddr=400c0000 size=00064h (   100) load
I (741) esp_image: segment 6: paddr=0028fba0 vaddr=50000000 size=00010h (    16) load
E (747) esp_image: Image length 1637344 doesn't fit in partition length 1572864
E (755) boot: OTA app partition slot 0 is not bootable
E (761) esp_image: image at 0x280000 has invalid magic byte (nothing flashed here?)
E (769) boot: OTA app partition slot 1 is not bootable
E (775) boot: No bootable app partitions in the partition table

igrr, 2022-03-09:
I'm sorry, I'm not sure about this second issue. I'll leave it to the folks here in esp-at project to comment. (cc @ustccw)

ustccw, 2022-03-10:
@agarwal-radha the log said that `Image length 1637344 (0x18fbe0) doesn't fit in partition length 1572864 (0x180000)`. Which means the firmware you generated size is 0x18fbe0, cannot fit in the default partition table: https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/partitions_at.csv#L6
so you can:
- Disable features you would not use on `./build.py menuconfig`, such as MQTT (`./build.py menuconfig` -> `Component config` -> `AT` -> `AT MQTT command support`)
- config silence mode (step 3: Enable silence mode to remove some logs and reduce the firmware size?) when compile firmware

if both of them still cannot meet your needs. so you have to consider to modify the default partition table: https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/partitions_at.csv according to your needs.

closed 2023-06-05 by ustccw