espressif/esp-at #617: 我怎样控制esp32主动发送RTS帧和CTS帧呢？(How to control esp32 to actively send RTS frame and CTS frame)

ogleede, 2022-03-10:
我选择一块ESP32作为AP，另一块ESP32作为STA，STA连接到了AP，我是否可以利用AT+UART_CUR，另一方主动发出RTS帧和CTS帧呢？esp32是否能控制RTS帧和CTS帧的发送呢？

ustccw, 2022-03-11:
@ogleede emm, what does your CTS/RTS frame mean, 802.11 control frame or UART hardware flow control?   
and would you mind explaining why you wanna do that.

ogleede, 2022-03-11:
是802.11的RTS帧，我想让STA发送RTS帧，AP回复CTS帧作为控制帧，当STA收到CTS帧后，触发一些其他逻辑。
我想的是，AP和STA每次通信前，都有RTS和CTS的交互。所以需要esp32可以主动发送这两个帧。

ustccw, 2022-03-11:
About sending 802.11 frame, you can try:
  ```
  esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);
  ```
About receiving 802.11 frame, you can try (device should be in promiscuous mode):
  ```
  esp_err_t esp_wifi_set_promiscuous_rx_cb(wifi_promiscuous_cb_t cb);
  ```
About how to add AT commands, you can refer to https://docs.espressif.com/projects/esp-at/zh_CN/latest/Compile_and_Develop/How_to_add_user-defined_AT_commands.html

ustccw, 2022-03-18:
@ogleede does that work?

ogleede, 2022-03-18:
@ustccw Thanks for your advice, I did not verify the correctness of this method later. Because I considered that even though RTS/CTS frames coule be sent, I did not find a way to identify the two frames.

ustccw, 2022-03-18:
just a possible help. the subtype (in 802.11 mac header)  is different for CTS (0xC0) and RTS (0xB0). you can try to parse the 802.11 frame by that way.

closed 2022-05-27 by ogleede