espressif/esp-at #618: HTTP command triggers the watchdog timer

ChiaWeiChien, 2022-03-15:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [none]
- Module or chip used:  [ESP8285H16]
- IDF version: 
    `v3.4-43-ge9516e4c`
- AT version
    `v2.2.1.0_esp8266-42-g248fbd0`
- Build System:         [CMake]
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description
The HTTP POST command triggers the watchdog timer if HTTPS was enabled.
I trierd to disable the non-task watchdog in the menuconfig but it didn't work.
if I use http instead of https, it works normal.

Also, I tried the HTTP Get (refer to [esp-at user guid](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Examples/http_at_examples.html#the-head-method-of-http-client))
`AT+HTTPCLIENT=2,0,"https://www.espressif.com/sites/all/themes/espressif/images/about-us/solution-platform.jpg",,,2`
The wdt was triggered also.

### Expected Behavior

```
AT+HTTPCPOST="https://httpbin.org/post",10,2,"connection: keep-alive","content-type: text/plain"

OK

>
+HTTPCPOST:282,{
                   "args": {},
                                 "data": "1234567890",
                                                         "files": {},
                                                                        "form": {},
                                                                                      "headers": {
                                                                                                      "Content-Length": "10",
                                                                                                                                  "Content-Type": "text/plain",
         "Host": "httpbin.org",
                                    "User-Agent": "ESP32 HTTP Client/1.0",
                                                                               "X-Amzn-Trace-Id": "Root=1-62305ac0-1c02c0454900a7eb1b
+HTTPCPOST:102,c67079"
                        },
                             "json": 1234567890,
                                                   "origin": "114.25.34.8",
                                                                              "url": "http://httpbin.org/post"
                                                                                                              }

SEND OK
```

### Actual Behavior

```
AT+HTTPCPOST="https://httpbin.org/post",10,2,"connection: keep-alive","content-type: text/plain"

OK

>
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 6960, room 16
tail 0
chksum 0xa5
load 0x3ffe8008, len 24, room 8
tail 0
chksum 0x2c
load 0x3ffe8020, len 3196, room 8
tail 4
chksum 0xeb
csum 0xeb

ready
WIFI CONNECTED
WIFI GOT IP
```

### Steps to repropduce

1. Create connection to AP
2. Wait DHCP to assign the IP address

```
WIFI CONNECTED
WIFI GOT IP
```

3. send AT command `AT+HTTPCPOST="https://httpbin.org/post",10,2,"connection: keep-alive","content-type: text/plain"`
4. enter 10 characters for the http body `1234567890`
5. the wdt will be triggered.

ustccw, 2022-03-18:
@ChiaWeiChien Hi, please try to enlarge AT_PROCESS_TASK_STACK_SIZE to 4096 (`./build.py menuconfig` -> `Component config` -> `AT` -> `The stack size of the AT process task in AT library..` ).

Maybe our prompt is not that conspicuous, when user enables `AT http command support` on esp8266, there is a help on that help menu.
> Please enlarge AT_PROCESS_TASK_STACK_SIZE to be more than 4096 if HTTPS link need to establish, otherwise, unexpected crash probably occur due to AT process task overflow.

I'll re-document that information to docs.espressif.com, thanks for your feedback!

ChiaWeiChien, 2022-03-18:
Hi, It works normal after I changed the task stack size. Thank you very much.

closed 2022-03-18 by ChiaWeiChien