espressif/esp-at #619: Ethernet - esp_at_eth_cmd_regist takes a lot of time to execute

ubichsel, 2022-03-15:
I have a WROOM32 board with a LAN8720 phy chip. It works well with the ethernet commands. Just the start-up time is much longer with ethernet (approximately 4 seconds). The log level is "error"
I see the function "esp_at_eth_cmd_regist" takes exactly this 4 seconds to execute (called in app_main.c). I can't debug deeper since the code is in the AT-Core library. Do you have any idea why it takes so long or if I can do something to make it faster? Is it normal that it will take such a long time?

Below are the logs in verbose level between the ethernet initialization:
**Ethernet cmd regist start**
D (3268) nvs: nvs_open_from_partition NVM_011 0
V (3268) esp_netif_objects: esp_netif_next_unsafe 0
D (3268) esp_netif_lwip: check: remote, if=0x3ffbd7d0 fn=0x40149d9c
D (3275) esp_netif_lwip: esp_netif_dhcpc_stop_api esp_netif:0x3ffbd7d0
D (3281) esp_netif_lwip: dhcp client stop successfully
D (3287) esp_netif_lwip: call api in lwip: ret=0x0, give sem
V (3292) esp_netif_objects: esp_netif_next_unsafe 0
D (3297) nvs: nvs_open_from_partition NVM_010 0
D (3302) nvs: nvs_open_from_partition NVM_006 0
V (3306) esp_netif_objects: esp_netif_next_unsafe 0
D (3311) esp_netif_lwip: check: remote, if=0x3ffbd7d0 fn=0x40149e68
D (3318) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffbd7d0
D (3324) esp_netif_lwip: dhcp client re init
D (3329) esp_netif_lwip: call api in lwip: ret=0x0, give sem
V (3335) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (3341) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (3349) intr_alloc: Connected src 38 to int 17 (cpu 0)
V (3394) esp_netif_objects: esp_netif_next_unsafe 0
<27>[0;32mI (3394) esp_eth.netif.glue: 4c:11:ae:ad:a0:03
<27>[0;32mI (3394) esp_eth.netif.glue: ethernet attached to netif
D (7399) event: running post ETH_EVENT:0 with handler 0x40149534 and context 0x3ffbd9cc on loop 0x3ffb7df0
D (7399) esp_netif_handlers: esp_netif action has started with netif0x3ffbd7d0 from event_id=0
D (7406) esp_netif_lwip: check: remote, if=0x3ffbd7d0 fn=0x4014b294
D (7413) esp_netif_lwip: esp_netif_start_api 0x3ffbd7d0
D (7418) esp_netif_lwip: esp_netif_get_hostname esp_netif:0x3ffbd7d0
D (7424) esp_netif_lwip: check: local, if=0x3ffbd7d0 fn=0x4014ad64
D (7431) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbd7d0
V (7437) esp_netif_objects: esp_netif_next_unsafe 0
V (7442) esp_netif_objects: esp_netif_next_unsafe 0x3ffbd7d0
V (7448) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbd5e8
V (7454) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbd7d0
D (7460) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (7466) event: running post ETH_EVENT:0 with handler 0x400e63cc and context 0x3ffb5b94 on loop 0x3ffb7df0
**ethernet init done**

Many thanks for your help / and information.

ustccw, 2022-03-18:
@ubichsel `esp_at_eth_cmd_regitst()` called the following possible time-consuming APIs, could you please add more debug info into their implement, see what happened.
- esp_event_handler_instance_register
- esp_eth_driver_install
- esp_netif_attach
- esp_eth_new_netif_glue
- esp_eth_start
- `esp-at/components/at/src/at_eth_init.c`

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

ubichsel, 2022-08-24:
Sorry I had other projects in the meantime. 
At the end it was in the function esp_eth_start->phy->negotiate(phy) -> lan8720_negotiate -> autonego_timeout_ms. This timeout seems to be used only there at startup to wait for a connection (and default is 4 seconds). I set the phy_config.autonego_timeout_ms = 0; in at_eth_init.c to 0. In this case it starts-up immediately. If you see any problems with this solution please let me know. Otherways it is solved for me.

closed 2023-06-05 by ustccw