espressif/esp-at #620: Ethernet - DHCP issues

KindarConrath, 2022-03-17:
I am using a WT32-ETH01

ESP-IDF 4.4
ESP-AT built from master

I have configured the firmware to communicate with the 8720 chip on the ETH01. It is running in Station mode.
It will not connect with the Router with either DHCP or a static IP address set.
In DHCP it will not get an IP address on the ethernet, Static it will set its IP but not appear on the network.

AT+CWDHCP=0,1 or AT+CWDHCP=1,1 
will not set the Ethernet DHCP bit

Once it is set to a static IP address, I can not switch it back to DHCP (though it doesn't appear to be doing a DHCP request anyways.

KindarConrath, 2022-03-17:
The ESP-AT documentation is slightly incomplete. 
AT+CWDHCP=1,4 WILL set the Ethernet DHCP bit, but it is missing from the documentation. (Show in in status but not mode)

However, with DHCP enabled, the device still will not get an IP address from the Router it is connected to even though the Link light is on.

ustccw, 2022-03-18:
@KindarConrath Seems you modified the ESP-IDF version to v4.4, right?  
what if you use our default esp-at repository with default ESP-IDF version.  

If the issue still exists, please enable debug log, and provide the steps you did, all the logs, and possible useful clues for us to reproduce it.

KindarConrath, 2022-03-18:
I have now rebuilt the project with the current version of the repo & using the ESP-IDF version that comes in the project folder.
It behaves the exact same, everything appears to work but it will not talk on the network.
I'm not sure how to enable the debug log, but I will include what I have so far.
I have also tried it with a second Ethernet cable & a second WT32-ETH01 board.

[Monitor Log](https://pastebin.com/dghiap0b)
[AT UART Log](https://pastebin.com/jGjyJ8Gd)

Build Changes - 

components->at->src->at_eth_init.c
between line 33 & 34 added
mac_config.sw_reset_timeout_ms = 1000; (this made it more stable)

components->customized_partitions->factory_param->factory_param_data.csv
-uart moved to 2&4 as per [this](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Ethernet_AT_Commands.html#cmd-ethpre) cts & rts pins set to -1 to disable
changed first line (WROOM-32)
PLATFORM_ESP32,WROOM-32,,0xfcfc,3,0,78,1,1,13,CN,115200,2,4,-1,-1,-1,-1

in menuconfig I changed
Component Config->Ethernet->ESP32 internal controller->RMII clock mode = Output from Internal
Component Config->Ethernet->ESP32 internal controller->Output RMII clock from GPIO0 = true
Component Config->AT->AT command terminator support = true
Component Config->AT->AT ethernet support = true
Component Config->AT->AT ethernet support->Ethernet PHY = Microchip LAN8720 PHY

(turned off bluetooth & OTA)
Component Config->AT->AT BLE = false
Component Config->AT->AT BLE HID = false
Component Config->AT->AT blufi = false
Component Config->AT->AT BT = false
Component Config->AT->AT OTA = false

ustccw, 2022-03-21:
sorry that i don't have a WT32-ETH01 board on hand.  
But from the logs you provided, it seems hardware is ready for your dhcp issues.

- For static IP address (Disable Ethernet DHCP)
there is one log said `+ETH_GOT_IP:192.168.0.55`, so ESP32 should obtain this static IP address.
But `AT+CIPSTART="TCP","www.google.com",80` command returns `unknown host: www.google.com`. This issue should be a DNS problem (dns server cannot return a valid ip address of `www.google.com`). You can set dns server by [AT+CIPDNS](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#at-cipdns-query-set-dns-server-information) command. If this doesn't work, make sure your router can access the internet, or connect to another router (LAN port) to test.

- For dynamic IP address (Enable Ethernet DHCP)
As you done, `AT+CWDHCP=7` should be the best choice. There is one log said `+ETH_CONNECTED`, but no a valid IP address got, like `+ETH_GOT_IP:192.168.0.55`. i am not sure what happened. So i would like to suggest you run [ethernet demo](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/ethernet/basic) first, and see if it can [get a valid ip address](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/ethernet/basic/main/ethernet_example_main.c#L61) here. If not, try to enable all logs (`idf.py menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` -> `Verbose`) and provide all possible useful info, I'd be happy to ask my workmate to help this issue.

btw, a few weeks ago, I tested the Ethernet feature with our ESP32-Ethernet-Kit board, and it seems work well.

[image]

KindarConrath, 2022-03-21:
Ok it continues to get interesting, I pulled out a fresh one, and it comes with firmware on it already. This firmware seems to work flawlessly (though I can not figure out the default AT terminator...) you can see the boot log [here](https://pastebin.com/L9TDY2d9)
Interestingly, it appears to be running an older version of the same ethernet_basic project with AT commands added.

I think I have figured out that this board uses a 50Hz oscillator signal on GIPO0 to sync the two chips, which is enabled with GPIO16. The older ESP-IDF had an option to enable & set the reset pin, but it is not in the configuration anymore.

So I need to set the timing signal to GPIO0_IN and configure the reset pin to be GPIO16. That should make it work.
setting the sync signal to GPIO0_IN is easy, that is the default and it is in the configurator, but the reset pin, do you know where I would configure that?

[Here](https://pastebin.com/0kNxnLJG) are my last logs of the day, with it configured as best I can figure, with logging fully enabled.

I can use the built in firmware, I assume that the "AT end" is an EN pin. but I would rather have a newer build of the ESP-AT if I can get it working! 

Thanks again for all your help!

ustccw, 2022-03-22:
happy to hear that.
From the [first log](https://pastebin.com/L9TDY2d9), it looks like there is a unmatched ESP-IDF version for esp-at project.
in your logs, it printed `ESP-IDF 5d27367-dirty`, but esp-at didn't use that ESP-IDF version. so there are some potential risks when using unmatched ESP-IDF. More details are here: https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_update_IDF.html

About `The older ESP-IDF had an option to enable & set the reset pin, but it is not in the configuration anymore.` so could you please describe more details, such as which ESP-IDF version, which configuration, etc.

KindarConrath, 2022-03-22:
To be clear, the first log is from the firmware that comes STOCK on the device, it was built by the manufacturer and I can not replicate their success.

As you can see from this issue someone posted a couple of years ago, #366 there used to be an option "use reset pin of PHY chip" and "PHY rst gpio number" I need to set this pin to 16 for this device.

KindarConrath, 2022-03-22:
I have finally got the Ethernet demo working, I was correct, the reset_gpio_num needs to be set to 16 for this device.

ustccw, 2022-04-01:
seems you are using eco1 board from `chip revision: 1`. i am not sure if this is an possible cause.
Could you please try to run esp-at/esp-idf/examples/ethernet/basic fistly, to make sure that your hardware is ready.

Further more, you can enable debug log in menuconfig, and add more debug info into ethernet trace, see what happened.

- esp_event_handler_instance_register
- esp_eth_driver_install
- esp_netif_attach
- esp_eth_new_netif_glue
- esp_eth_start
- esp-at/components/at/src/at_eth_init.c

closed 2023-06-05 by ustccw