espressif/esp-at #621: spp  - Service uuid 0xA002--> Write Char uuid 0xC304  sending 1 byte only (from nrfconnect)

RenyV, 2022-03-22:
AT BIN Version

AT+GMR
AT version:2.1.0.0(883f7f2 - Jul 24 2020 11:50:07)
SDK version:v4.0.1-193-ge7ac221
compile time(0ad6331):Jul 28 2020 02:46:20
Bin version:2.1.0(SOLO-1)

I run ESP32 with AT Commands from another Cortex M 32 bit microcontroller

ESP32: Server
nrfConnect App: Client

My requirement is : 
1. I need to Write data from ESP32 to nrfConnect App 
2. I need to Read data in ESP32 (sent from nrfConnect App)

I used the followinf example from esp32 link
https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/bluetooth_le_at_examples.html#establish-spp-connection-between-esp32-and-mobile-phone-and-transmit-data-in-uart-bluetooth-le-passthrough-mode

I do the following sequence for initialization

ATE1
OK

AT BIN Version
AT+GMR
AT version:2.1.0.0(883f7f2 - Jul 24 2020 11:50:07)
SDK version:v4.0.1-193-ge7ac221
compile time(0ad6331):Jul 28 2020 02:46:20
Bin version:2.1.0(SOLO-1)
OK

BLE INIT
AT+BLEINIT=2
OK

ESP32 Bluetooth LE server creates services 
AT+BLEGATTSSRVCRE
OK
WIFI GOT IP

ESP32 Bluetooth LE server starts services
AT+BLEGATTSSRVSTART
OK

ESP32 Bluetooth LE server discovers local Services.
AT+BLEGATTSSRV?
+BLEGATTSSRV:1,1,0xA002,1
+BLEGATTSSRV:2,1,0xA003,1
OK

ESP32 Bluetooth LE server discovers local characteristics.
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300,0x02
+BLEGATTSCHAR:"desc",1,1,1,0x2901
+BLEGATTSCHAR:"char",1,2,0xC301,0x02
+BLEGATTSCHAR:"desc",1,2,1,0x2901
+BLEGATTSCHAR:"char",1,3,0xC302,0x08
+BLEGATTSCHAR:"desc",1,3,1,0x2901
+BLEGATTSCHAR:"char",1,4,0xC303,0x04
+BLEGATTSCHAR:"desc",1,4,1,0x2901
+BLEGATTSCHAR:"char",1,5,0xC304,0x08
+BLEGATTSCHAR:"char",1,6,0xC305,0x10
+BLEGATTSCHAR:"desc",1,6,1,0x2902
+BLEGATTSCHAR:"char",1,7,0xC306,0x20
+BLEGATTSCHAR:"desc",1,7,1,0x2902
+BLEGATTSCHAR:"char",1,8,0xC307,0x02
+BLEGATTSCHAR:"desc",1,8,1,0x2901
+BLEGATTSCHAR:"char",2,1,0xC400,0x02
+BLEGATTSCHAR:"desc",2,1,1,0x2901
+BLEGATTSCHAR:"char",2,2,0xC401,0x02
+BLEGATTSCHAR:"desc",2,2,1,0x2901
OK

ESP32 Bluetooth LE server gets its MAC address 
AT+BLEADDR?
+BLEADDR:"24:a1:60:e7:b8:d6"
OK

Set Bluetooth LE advertising data 
AT+BLEADVDATA="0201060A09457370726573736966030302A0"
OK

ESP32 Bluetooth LE server starts advertising.
AT+BLEADVSTART
OK

**_In nrfconnect App I Scan for the Espressif device and Connection made_**

ESP32 Bluetooth LE server check for connection status
AT+BLECONN?
+BLECONN:0,"66:eb:8c:ef:2f:5f"
OK

set the parameters of Bluetooth LE connection
AT+BLECONNPARAM=0,12,14,1,500
OK
+BLECONNPARAM:0,12,14,12,1,500

ESP32 Bluetooth LE server configures Bluetooth LE SPP.
AT+BLESPPCFG=1,1,7,1,5
OK

ESP32 Bluetooth LE server enables Bluetooth LE SPP 
AT+BLESPP
OK
>

t

### When I try to Write a text data "test" in Service 0xA002-->Char uuid 0xC304, i get only the first byte written 't'### 
[image]

Please see above. 

Is this something to do with Char uuid 0xC304? How do I solve this issue?

I want to write multiple bytes @ char 0xC304 and notify/indicate multiple bytes @ char 0xC306. Please help.

RenyV, 2022-03-23:
Any help please?

Luxin812, 2022-04-14:
Maybe you can test it with the latest version.
```
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADDR?
+BLEADDR:"fc:f5:c4:3b:fc:3a"
OK
AT+BLEADVDATA="0201060A09457370726573736966030302A0"

OK
AT+BLEADVSTART

OK
+BLECONN:0,"75:db:91:5c:4d:a6"
AT+BLEGATTSSRV?
+BLEGATTSSRV:1,1,0xA002,1
+BLEGATTSSRV:2,1,0xA003,1

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300,0x02
+BLEGATTSCHAR:"desc",1,1,1,0x2901
+BLEGATTSCHAR:"char",1,2,0xC301,0x02
+BLEGATTSCHAR:"desc",1,2,1,0x2901
+BLEGATTSCHAR:"char",1,3,0xC302,0x08
+BLEGATTSCHAR:"desc",1,3,1,0x2901
+BLEGATTSCHAR:"char",1,4,0xC303,0x04
+BLEGATTSCHAR:"desc",1,4,1,0x2901
+BLEGATTSCHAR:"char",1,5,0xC304,0x08
+BLEGATTSCHAR:"char",1,6,0xC305,0x10
+BLEGATTSCHAR:"desc",1,6,1,0x2902
+BLEGATTSCHAR:"char",1,7,0xC306,0x20
+BLEGATTSCHAR:"desc",1,7,1,0x2902
+BLEGATTSCHAR:"char",1,8,0xC307,0x02
+BLEGATTSCHAR:"desc",1,8,1,0x2901
+BLEGATTSCHAR:"char",2,1,0xC400,0x02
+BLEGATTSCHAR:"desc",2,1,1,0x2901
+BLEGATTSCHAR:"char",2,2,0xC401,0x02
+BLEGATTSCHAR:"desc",2,2,1,0x2901

OK
AT+BLESPPCFG=1,1,7,1,5

OK
AT+BLESPP

ERROR

ready

ready

ready

ready
AT+RESTORE

OK

ready
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADDR?
+BLEADDR:"fc:f5:c4:3b:fc:3a"
OK
AT+BLEADVDATA="0201060A09457370726573736966030302A0"

OK
AT+BLEADVSTART

OK
+BLECONN:0,"79:55:dc:ad:78:22"
+BLECONNPARAM:0,0,0,6,0,500
+BLECONNPARAM:0,0,0,36,0,500
AT+BLECONNPARAM=0,12,14,1,500

OK
+BLECONNPARAM:0,12,14,12,1,500
AT+BLESPPCFG=1,1,7,1,5

OK
+WRITE:0,1,6,1,2,\0x01\0x00
AT+BLESPP

ERROR
+WRITE:0,1,7,1,2,\0x02\0x00
AT+BLESPP

OK

>testAT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADDR?
+BLEADDR:"fc:f5:c4:3b:fc:3a"
OK
AT+BLEADVDATA="0201060A09457370726573736966030302A0"

OK
AT+BLEADVSTART

OK
+BLECONN:0,"79:55:dc:ad:78:22"
+BLECONNPARAM:0,0,0,6,0,500
+BLECONNPARAM:0,0,0,36,0,500
+WRITE:0,1,7,1,2,\0x02\0x00
AT+BLESPPCFG=1,1,7,1,5

OK
AT+BLESPP

OK

>testAT+GMR
AT version:2.3.0.0-dev(bb67465 - ESP32 - Mar 25 2022 12:36:11)
SDK version:v4.3.2-450-g8ffddf5
compile time(80c5529):Mar 25 2022 13:27:20
Bin version:2.2.0(WROOM-32)

OK
```

closed 2023-06-05 by ustccw