espressif/esp-at #622: wifi easy connect

sumanclimote, 2022-03-25:
Does this lib support the "wifi easy connect", I am interested to make a IOT system with this:
https://www.wi-fi.org/discover-wi-fi/wi-fi-easy-connect
https://www.wi-fi.org/download.php?file=/sites/default/files/private/Wi-Fi_Easy_Connect_Specification_v2.0.pdf

ustccw, 2022-03-28:
@sumanclimote Sorry that esp-at has no plan about **wifi easy connect** for AT commands.

I would like to suggest you add your AT commands  according to your needs.
Refer to:
- https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_add_user-defined_AT_commands.html
- esp-at/esp-idf/examples/wifi/wifi_easy_connect

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

ayoububilite, 2022-08-17:
After compiling the ESP IDF Project, Wifi Easy Connect isn't shown, is it still supported?

aayoub13, 2022-08-17:
Trying to connect Easy Connect using Oneplus 0Pro, 7T, Samsung Note10, no Response

closed 2023-06-05 by ustccw