espressif/esp-at #623: ESP32-AT HTTPS connection

Nesrine352, 2022-03-29:
Hi,

I am trying to create SSL connection using ESP32-AT commands on ESP32-WROOM. i have release /v2.2.0 AT firmware version.

[image]

My setps are:

AT+CWMODE=1
AT+CWJAP="<SSID>","<password>"
AT+CIPSNTPCFG=1,1245,"0.pool.ntp.org","time.google.com"
AT+CIPSTART="SSL","iot.espressif.cn",8443
AT+CIPSEND=42
for the last command i'm getting the following message.
>+IPD,61:{"status": 400, "message": "io error, may be idle too long"}                                                                                
CLOSED
  
Am i missing some setps?
i'd be very gratful for any king of help

ustccw, 2022-03-30:
@Nesrine352 i think the steps you did are right, esp-at also documented it at https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/TCP-IP_AT_Examples.html#esp-device-as-an-ssl-client-in-single-connection

the web server (http://iot.espressif.cn/#/) said that "the IOT platform will stop maintenance on January 30, 2022".
Further more, you can enable ssl debug log by:
- `./build.py menuconfig` -> `Component config` -> `mbedTLS` -> `Enable mbedTLS debugging` -> `Verbose`
- `./build.py menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` -> `Debug`

See what happened.

Nesrine352, 2022-03-31:
Hi @ustccw ,

it works! thanks for your reply :)