espressif/esp-at #624: Why was SDIO  AT-command support removed for ESP8266?

pete-onecare, 2022-03-30:
We have ESP8285 1MB Flash that uses SDIO-SPI for AT commands (using pins 9-13), and an older version of firmware (v1.4). We want to upgrade to latest version, but apparently according to release notes, SDIO AT support was removed for ESP8266 in release v2.2.0.0. 

Why was this done, and what work-around can we do to get SPI working in the latest firmware?

ustccw, 2022-03-31:
@pete-onecare ESP8266-SDIO is no longer supported for the following reasons.
1. ESP8266 SDIO is hard to use, it need to be guided to boot.
2. ESP8266 heap size is too small, so the throughput is low.

We recommend using ESP32 SDIO instead.

On other hand, we do not intend to support ESP8266 SPI AT. If you want use SPI AT, we recommend using ESP32-C3.

cc @xcguang @ESP-Coco

pete-onecare, 2022-03-31:
We do not have an option to use another chip at this point. We already have
ESP8285 installed in our product and we have older v1.4 version of firmware
that DOES SUPPORT SPI or SDIO SPI

Is it all possible for me to compile newer firmware that would support SDIO
SPI on ESP8266/ESP8285 1MB Flash?

Otherwise we are stuck with this very old firmware from 2017 that does not
support SSL.

xcguang, 2022-04-01:
You can continue using the old SDK for your existed products: https://github.com/espressif/ESP8266_NONOS_SDK

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

pete-onecare, 2022-05-27:
The old SDK did not work for me either. I needed to have a <1024 kb
configuration and using SDIO AT commands, which was custom (Nano + SDIO).
The software did not work on our chip and I can't figure out why. It is on
hold until I have time to resume this.