espressif/esp-at #628: esp32c3 BLEGATTSNTFY requiring messages larger than 26 bytes to be sent out twice

ptqmedina, 2022-04-15:
Issue:

I am using esp-at SPI.

I am acting as a BLE server and have a connected android client sending commands to retrieve data from the device. When using BLEGATTSNTFY to send data to the client, any packet that is larger than 26 bytes is requiring me to send the data twice. The phone is also receiving this data twice. I have tried with multiple android devices, all behaving in the same way.

On the first send, the esp32 sends the '>' symbol and i then send the data i want, it then echoes the BLEGATTSNTFY command again and proceeds to request data once again.

Overall the issue i am having with BLEGATTSNTFY seems to be:
if the payload is greater than 26 bytes, esp32c3 requests that it is sent twice.
if the payload is less than 26 bytes, works as expected.
if the payload is 26 bytes exactly, esp32c3 may be stuck in infinite loop asking for data

Any help/guidance would be appreciated. 

I've attached relevant information below
================================

esp32c3:
ESP-AT SPI is being used
AT version:2.4.0.0-dev(s-a1a1bb5 - ESP32C3 - Feb 25 2022 09:34:23)
SDK version:v4.3.2-214-g87f8a0d5f1\r\ncompile time(0067e16):Mar 10 2022 21:29:20
Bin version:2.3.0(MINI-1)

Client:
Samsung Galaxy S22 (Android)

AT Messages:
Connected android client sends command
"+WRITE:0,1,1,,12,�\177~\f"

Query for configured MTU
AT+BLECFGMTU?
"AT+BLECFGMTU?\r\n" (echo)
"+BLECFGMTU:0,512\r\n"
"\r\nOK\r\n"

Send a test packet of 35 bytes long to service 1, characteristic 1
AT+BLEGATTSNTFY=0,1,1,35
"AT+BLEGATTSNTFY=0,1,1,35\r\n" (echo)

esp32c3 requests data to be sent
>

send data 35 bytes of data
··Ik····UUUUUUUUUUUUUUUUUUUUUUUUUUU

esp32c3 sends echo again ?
"AT+BLEGATTSNTFY=0,1,1,35\r\n"

busy pending from esp32
"\r\nbusy p...\r\n" 

esp32c3 says OK to last command and gives another > in same message that requires me to send data again
"\r\nOK\r\n>"

send 35 bytes of data
··Ik····UUUUUUUUUUUUUUUUUUUUUUUUUUU

Luxin812, 2022-04-19:
Can you test with ESPAT UART to see if you have the same issue?

ptqmedina, 2022-04-20:
I don't have the ability to test with ESP-AT UART as this prototype is built around the SPI library.

This is the only AT command that I have experienced this issue with.

ptqmedina, 2022-05-09:
Any suggestions for what I could look at to better debug this issue? 

I am also seeing this with BLEGATTSIND. Instead of 26 bytes being the threshold, 25 bytes is the threshold for this commands issue.

if the payload is greater than 25 bytes, esp32c3 requests that it is sent twice.
if the payload is less than 25 bytes, works as expected.
if the payload is 25 bytes exactly, esp32c3 infinite loop.

I did try different BLE test apps (nrf connect, ble scanner, etc), multiple mobile devices, and a PC as a client which resulted in the same issues. These are currently the only two esp-at commands I am experiencing issues with.

Jacques-zhao, 2022-05-11:
@ptqmedina Can you test it with another phone (such as Iphone)?
We reproduced it the way you did without a problem, we don't have the phone model you said.

ptqmedina, 2022-05-11:
@Jacques-zhao 
I will try to get an iphone to test with. As an update I was able to locate the source of my issue:

For some reason esp_at_port_get_data_length() always returns a value showing that there is data remaining to be read. Since there is remaining data seen on the port, esp_at_port_recv_data_notify() causes the repeat of the AT command that was sent, this is where the repeat echo and repeat '>' come from. However, it has already read all of the data sent using esp_at_port_read_data() right before that and the data gets sent out correctly over BLE.

I don't have this issue when using CIPSEND when using wifi which also sends a '>' to receive data, although it does not seem to use the exact same method as BLEGATTSNTFY or BLEGATTSIND.

I experience this with any command that uses this same method (BLEGATTSNTFY, BLEGATTSIND etc.) and was also able to write a custom AT command that also does this (without transmitting BLE).

In my custom AT command i tried to replicate what BLEGATTSNTFY was doing and I came up with what you see below. Removing esp_at_port_recv_data_notify() fixes the issue I am seeing. Do you have any suggestions in what I could look at in my SPI setup?
```
int32_t length = 0; 
if (esp_at_get_para_as_digit(3, &length) != ESP_AT_PARA_PARSE_RESULT_OK)
{ 
    return ESP_AT_RESULT_CODE_ERROR; 
}
uint8_t *buf = (uint8_t *)malloc(length);
if(!at_ntfy_ble)
{
    at_ntfy_ble = xSemaphoreCreateBinary();
    assert(at_ntfy_ble != NULL);
}
esp_at_port_write_data((uint8_t*)">", strlen(">"));
esp_at_port_enter_specific(wait_data_callback);
int32_t received_len = 0;
int32_t ret = 0;
while(received_len != length)
{
    ret = esp_at_port_read_data(buf+received_len,length-received_len);
    received_len = received_len + ret;
}
esp_at_port_exit_specific();
int32_t remainder = esp_at_port_get_data_length();
if (remainder > 0) 
{
    esp_at_port_recv_data_notify(remainder, portMAX_DELAY); // removing this works
}
// send ble data out here... not related to the problem
free(buf);
```

Jacques-zhao, 2022-05-12:
Hi @ptqmedina  This looks like a problem I solved recently, you can try it again after apply this patch
[0001-fix-SPI-fix-C3-SPI-notify-block.zip](https://github.com/espressif/esp-at/files/8674818/0001-fix-SPI-fix-C3-SPI-notify-block.zip)

Jacques-zhao, 2022-05-17:
@ptqmedina  This fix has been merged , the commit is https://github.com/espressif/esp-at/commit/17ec85270f8e1b447482926df2843961b57498e6 , the upcoming version will include this patch.

closed 2023-06-05 by ustccw