espressif/esp-at #629: ESP32 SPP master crashes on AT+BTSPPSEND command

wjklimek1, 2022-04-18:
Environment:
Two ESP32 flashed with AT firmware with Bluetooth Classic enabled, connected to STM32 microcontrollers with UART on 115200 baudrate. Configured in SPP profile with following commands:

Master:
AT+BTINIT=1\r\n
AT+BTSCANMODE=1\r\n
AT+BTSPPINIT=1\r\n

Slave:
AT+BTINIT=1\r\n
AT+BTSCANMODE=2\r\n
AT+BTSPPINIT=2\r\n
AT+BTSPPSTART\r\n

Procedure:
After initialization master connects to slave with AT+BTSPPCONN command and connection is correctly established. Then master sends 9 byte long message to slave and slave responds to this mesaage with another 9 byte message. This works fine for roughly 1290 messages, then master crashes and ESP32 reboots, sending "ready" string to UART 1s after crash.

After crash master ESP32 sends to USB UART following crash report:
``` [18:43:16:407] ␍␊
[18:43:16:408] assert failed: xQueueSemaphoreTake queue.c:1447 (( pxQueue ))␍␊
[18:43:16:412] ␍␊
[18:43:16:413] Backtrace:0x400814db:0x3ffcf7f0 0x40091e89:0x3ffcf810 0x400979b1:0x3ffcf830 0x40092ef2:0x3ffcf950 0x400f5586:0x3ffcf990 0x400e83b9:0x3ffcf9c0 0x400f48e6:0x3ffcf9e0 0x400953ad:0x3ffcfa20␍␊
[18:43:16:429] ␍␊
[18:43:16:429] ␍␊
[18:43:16:429] ELF file SHA256: f1babf7b1c6a5842␍␊
[18:43:16:432] ␍␊
[18:43:16:432] Rebooting...␍␊
[18:43:16:434] ets Jul 29 2019 12:21:46␍␊
[18:43:16:436] ␍␊
[18:43:16:436] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[18:43:16:441] configsip: 0, SPIWP:0xee␍␊
[18:43:16:444] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[18:43:16:450] mode:DIO, clock div:2␍␊
[18:43:16:452] load:0x3fff0030,len:7648␍␊
[18:43:16:457] load:0x40078000,len:16432␍␊
[18:43:16:460] load:0x40080400,len:3704␍␊
[18:43:16:460] entry 0x4008067c␍␊
[18:43:16:462] <0x1b>[0;32mI (27) boot: ESP-IDF v4.3.2-450-g8ffddf53bc 2nd stage bootloader<0x1b>[0m␍␊
[18:43:16:469] <0x1b>[0;32mI (27) boot: compile time 14:09:58<0x1b>[0m␍␊
[18:43:16:473] <0x1b>[0;32mI (27) boot: chip revision: 3<0x1b>[0m␍␊
[18:43:16:478] <0x1b>[0;32mI (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0<0x1b>[0m␍␊
[18:43:16:484] <0x1b>[0;32mI (38) boot.esp32: SPI Speed      : 40MHz<0x1b>[0m␍␊
[18:43:16:488] <0x1b>[0;32mI (43) boot.esp32: SPI Mode       : DIO<0x1b>[0m␍␊
[18:43:16:493] <0x1b>[0;32mI (48) boot.esp32: SPI Flash Size : 4MB<0x1b>[0m␍␊
[18:43:16:497] <0x1b>[0;32mI (52) boot: Enabling RNG early entropy source...<0x1b>[0m␍␊
[18:43:16:503] <0x1b>[0;32mI (58) boot: Partition Table:<0x1b>[0m␍␊
[18:43:16:506] <0x1b>[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m␍␊
[18:43:16:514] <0x1b>[0;32mI (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000<0x1b>[0m␍␊
[18:43:16:521] <0x1b>[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000<0x1b>[0m␍␊
[18:43:16:529] <0x1b>[0;32mI (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000<0x1b>[0m␍␊
[18:43:16:536] <0x1b>[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000<0x1b>[0m␍␊
[18:43:16:546] <0x1b>[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00100000 00180000<0x1b>[0m␍␊
[18:43:16:551] <0x1b>[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000<0x1b>[0m␍␊
[18:43:16:561] <0x1b>[0;32mI (113) boot: End of partition table<0x1b>[0m␍␊
[18:43:16:563] <0x1b>[0;32mI (118) boot_comm: chip revision: 3, min. application chip revision: 0<0x1b>[0m␍␊
[18:43:16:570] <0x1b>[0;32mI (125) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=282c0h (164544) map<0x1b>[0m␍␊
[18:43:16:627] <0x1b>[0;32mI (193) esp_image: segment 1: paddr=001282e8 vaddr=3ffbdb60 size=040e4h ( 16612) load<0x1b>[0m␍␊
[18:43:16:635] <0x1b>[0;32mI (200) esp_image: segment 2: paddr=0012c3d4 vaddr=40080000 size=03c44h ( 15428) load<0x1b>[0m␍␊
[18:43:16:644] <0x1b>[0;32mI (206) esp_image: segment 3: paddr=00130020 vaddr=400d0020 size=115864h (1136740) map<0x1b>[0m␍␊
[18:43:17:052] <0x1b>[0;32mI (619) esp_image: segment 4: paddr=0024588c vaddr=40083c44 size=16184h ( 90500) load<0x1b>[0m␍␊
[18:43:17:090] <0x1b>[0;32mI (656) esp_image: segment 5: paddr=0025ba18 vaddr=400c0000 size=00064h (   100) load<0x1b>[0m␍␊
[18:43:17:098] <0x1b>[0;32mI (656) esp_image: segment 6: paddr=0025ba84 vaddr=50000000 size=00010h (    16) load<0x1b>[0m␍␊
[18:43:17:110] <0x1b>[0;32mI (674) boot: Loaded app from partition at offset 0x100000<0x1b>[0m␍␊
[18:43:17:115] <0x1b>[0;32mI (675) boot: Disabling RNG early entropy source...<0x1b>[0m␍␊
[18:43:17:325] module_name:WROOM-32␍␍␊
[18:43:17:327] 2.2.0␍␍␊
```

System crash on an RTOS semaphore's assert is definitely a bug in a ESP-AT firmware, but how can i fix it? This problem occurres always, even if I only transmit data one way, from slave to master or from master to slave.

Luxin812, 2022-04-20:
Can you provide your version information (queried by AT+GMR command) and provide the complete AT port and log port logs?

Luxin812, 2022-04-20:
Maybe you can test with the latest version.

wjklimek1, 2022-04-20:
AT+GMR command returns following data:
```
AT+GMR␍␊
AT version:2.3.0.0-dev(bb67465 - ESP32 - Mar 25 2022 12:36:11)␍␊
SDK version:v4.3.2-450-g8ffddf53bc␍␊
compile time(19b48fe):Apr  1 2022 14:16:19␍␊
Bin version:2.2.0(WROOM-32)␍␊
```

If you need it, I can provide full logs of AT commands entering/exiting both ESPs within a day or so.

On each ESP32 module RX2 and TX2 are connected to STM32 microcontroller, and debug messages from AT ESP firmware are printed to embedded UART-USB converter. Debug UARTs don't send anything between boot and crash. Boot message is exactly the same as the message on rebooting after crash, only boot cause is different (POWERON_RESET)