espressif/esp-at #633: Flash Write Clarification

campbellyoung, 2022-05-09:
Hi,

I'm just wanting to clarify the behavior for flash writes so I don't accidentally wear out the flash. [AT+CWAUTOCONN](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-autoc) says "The configuration changes will be saved in the NVS area."

Should I expect this command to write to flash every time it is executed, or does it only save the setting when it changes? The [Zephyr driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi/esp_at/esp.c#L956) writes this value each time the device reboots, so I will need to change the behavior if the write is unconditional

ustccw, 2022-05-09:
@campbellyoung Hi, thanks for your observation! All nvs write operations will not write to flash unconditionally.
ESP-IDF merged this feature on this PR: https://github.com/espressif/esp-idf/pull/3310, and ESP-AT got this after v2.0.0.0 version. 
Please know.

ustccw, 2022-05-09:
I'll update the esp-at documentation to make it clearer of flash write.

campbellyoung, 2022-05-09:
Great, thanks for the clarification! I thought that would be the case but it’s nice to have it confirmed. Appreciate the detailed response 🙂

closed 2022-05-09 by campbellyoung