espressif/esp-at #634: ESP32-C3-MINI-1_AT_Bin_V2.4.0.0 : Frequent disconnection in BLE, connection not stable.

gkg-7, 2022-05-23:
I have configured one ESP32C3-Mini as the server and one on the DevKit as the client. When I try to connect it gets connected but disconnects in less than 5 seconds. But sometimes it doesn't disconnect and allows me to do my project. Due to the frequent disconnection and the unstable behavior, I am not able to proceed to the next step.

Luxin812, 2022-05-23:
Is the client using a mobile phone, or an ESP32C3-MINI device?

gkg-7, 2022-05-23:
The client is using the ESP32C3-Mini DevKit.

gkg-7, 2022-05-23:
I am executing this example : https://docs.espressif.com/projects/esp-at/en/release-v2.4.0.0/esp32c3/AT_Command_Examples/bluetooth_le_at_examples.html#establish-spp-connection-between-two-esp32-c3-development-boards-and-transmit-data-in-uart-bluetooth-le-passthrough-mode

gkg-7, 2022-05-23:
This issue occurs almost 90% of the time.

Luxin812, 2022-05-23:
At what step does a BLE disconnect usually occur?

Luxin812, 2022-05-23:
I have tested 10 times and still can't reproduce your problem, can you provide the AT log with timestamp?

gkg-7, 2022-05-23:
Server side Steps:

1. AT+BLEINIT=2
2. AT+BLEGATTSSRVCRE
3. AT+BLEGATTSSRVSTART
4. AT+BLEADVDATA="0201060A09457370726573736966030302A0"
5. AT+BLEGATTSSRV?
6. AT+BLEGATTSCHAR?
7. AT+BLEADVSTART

Client side steps:

1. AT+BLEINIT=1
2. AT+BLESCAN=1,3
3. AT+BLECONN=0,"34:b4:72:b2:68:52"

As soon as I give the At command for connection, it takes almost 5 seconds and it gets disconnected.

gkg-7, 2022-05-23:
> I have tested 10 times and still can't reproduce your problem, can you provide the AT log with timestamp?
[ESP32C3MINIClient_LOG
[ESP32C3MINIServer_LOG.txt](https://github.com/espressif/esp-at/files/8754580/ESP32C3MINIServer_LOG.txt)
.txt](https://github.com/espressif/esp-at/files/8754563/ESP32C3MINIClient_LOG.txt)

gkg-7, 2022-05-23:
I am sorry, don't have the server log with timestamp. I can get that for you in sometime.

Luxin812, 2022-05-23:
Is your BLE environment very complicated? From the log, the scanned BLE server rssi is -67. Frequent disconnections should be related to this
[image]
.

gkg-7, 2022-05-23:
This is the server log with timestamp.
[ESP32C3MINIServer_LOG_Timestamp.txt](https://github.com/espressif/esp-at/files/8754660/ESP32C3MINIServer_LOG_Timestamp.txt)

gkg-7, 2022-05-23:
> Is your BLE environment very complicated? From the log, the scanned BLE server rssi is -67. Frequent disconnections should be related to this [image] .

Can I share my BLE environment schematic? if you could go through that and let me know, it would be really helpful.

Luxin812, 2022-05-23:
You can change to a place with less BLE devices (if it is a shielded box, it is better) to test.

gkg-7, 2022-05-23:
Please let me know if I can share my BLE environment schematic and the gerber file, also I can show you the location where it is placed on our board.

Few weeks back we had done a schematic and gerber review from espressif, I have the report for that as well.

gkg-7, 2022-05-23:
> You can change to a place with less BLE devices (if it is a shielded box, it is better) to test.

Sure I will try that as well.

gkg-7, 2022-05-23:
Yes it is an official module from espressif, we have mounted that on our target board. I will share you the schematic and a picture of our board as well.

Schematic:
[SNM476M_Espressif review_21032022.pdf](https://github.com/espressif/esp-at/files/8754823/SNM476M_Espressif.review_21032022.pdf)

Placement:
[image]

my personal mail id: geoffreykgeorge@gmail.com.

Could you please share your mail ID to that mail? I could share more info regarding this issue.

Luxin812, 2022-05-24:
Maybe you can set the AT+CWMODE=0 command before both devices make a BLE connection to see if the situation improves.

gkg-7, 2022-05-24:
I have done that but the server RSSI still remains somewhere around -65 to -70 dB. Can we increase the gain with AT+RFPOWER?

gkg-7, 2022-05-24:
We are desperately in need of a solution here. We have another part ESP32­C3­MINI­1U with us also, what antenna should we use for this module? We are just trying different options as we are in a very critical stage of product development.

Luxin812, 2022-05-24:
Can you help to download the firmware from the attachment and burn it to the device and test if the same disconnection issue exists? I want to see if it's a software problem. [ESP32-C3-MINI-1_AT_Bin_V2.3.0.0 (4).zip](https://github.com/espressif/esp-at/files/8759980/ESP32-C3-MINI-1_AT_Bin_V2.3.0.0.4.zip)

gkg-7, 2022-05-24:
Okay sure I will try that now.

gkg-7, 2022-05-24:
5/24/2022 12:48:58 [RX] - AT+BLESCAN=1,1<CR><LF>
<CR><LF>
OK<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-57,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-57,,,0<CR><LF>
+BLESCAN:"e5:9c:a8:d8:bb:bb",-62,07ff4c0012020001,,1<CR><LF>
+BLESCAN:"3c:f3:c2:3c:04:df",-93,1eff060001092002911e6b000b3a0461896af2450d6ae72629bbfca2344358,,1<CR><LF>
+BLESCAN:"0c:36:a2:42:53:2d",-84,02011a0bff4c000906032c0a2886bb,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-60,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-60,,,0<CR><LF>
+BLESCAN:"25:80:87:33:21:60",-90,1eff0600010920021108a33f9827bff1f953ec1a3f840271c0e68814e01810,,1<CR><LF>
+BLESCAN:"18:eb:83:62:d9:34",-87,1eff0600010920026f1a3cc9858fbf0b91c9928ccefbfb3889720eebba6127,,1<CR><LF>
+BLESCAN:"49:d8:7e:f5:48:c3",-80,1eff0600010920020724004497c99c41986b8e6d869cbd47a11cb5976f796b,,1<CR><LF>
+BLESCAN:"1e:c1:29:02:a4:fb",-71,1eff06000109200242fa2a42ddf24c6ec01a989ef0fd9971f9295dd54fff7d,,1<CR><LF>
+BLESCAN:"43:e9:14:1c:6c:ab",-85,1eff060001092002202928b6b3b9fa239d5e8af32b48fbda4394288bf38860,,1<CR><LF>
+BLESCAN:"2e:a1:e2:00:06:b1",-83,1eff060001092002a45174413133323aec690c4478fff5a2b5955890693a3a,,1<CR><LF>
+BLESCAN:"2f:46:06:61:2c:a3",-88,1eff0600010920022059c031324071cfbc983b28542f34df442b7e39933249,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-57,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-57,,,0<CR><LF>
+BLESCAN:"65:86:14:38:5e:f4",-66,1eff0600010920027a3587c96f5e18df38d7378e297752a788e87305741e55,,1<CR><LF>
+BLESCAN:"19:68:ea:77:96:da",-72,1eff06000109200265e9a82c87a1c47d71d7e29ae362026cb0bc652ac5b8cd,,1<CR><LF>
+BLESCAN:"3c:f3:c2:3c:04:df",-91,1eff060001092002911e6b000b3a0461896af2450d6ae72629bbfca2344358,,1<CR><LF>
+BLESCAN:"31:7f:52:92:3e:ac",-89,1eff060001092002f7032a211af453939cf28f838c46da33d364a076d7722c,,1<CR><LF>
+BLESCAN:"49:5a:1a:7f:6c:72",-85,02011a1aff4c000c0e00e84fbb12c7ed634f3b1c2e314d1005431c748541,,1<CR><LF>
+BLESCAN:"49:5a:1a:7f:6c:72",-86,,,1<CR><LF>
+BLESCAN:"2b:f6:85:12:c6:bc",-61,1eff060001092002570a3afced5e9544b200e79080854b7b7f3ac3b351d14d,,1<CR><LF>
+BLESCAN:"1e:c1:29:02:a4:fb",-74,1eff06000109200242fa2a42ddf24c6ec01a989ef0fd9971f9295dd54fff7d,,1<CR><LF>
+BLESCAN:"18:eb:83:62:d9:34",-78,1eff0600010920026f1a3cc9858fbf0b91c9928ccefbfb3889720eebba6127,,1<CR><LF>
+BLESCAN:"ff:38:f5:0a:65:ad",-70,0201061009534e4d3437362d41314c3237323033,,1<CR><LF>
+BLESCAN:"ff:38:f5:0a:65:ad",-69,,11077c16a55eba11cb920c497fb801119a56,1<CR><LF>
+BLESCAN:"2f:46:06:61:2c:a3",-89,1eff0600010920022059c031324071cfbc983b28542f34df442b7e39933249,,1<CR><LF>
+BLESCAN:"65:86:14:38:5e:f4",-65,1eff0600010920027a3587c96f5e18df38d7378e297752a788e87305741e55,,1<CR><LF>
+BLESCAN:"19:68:ea:77:96:da",-72,1eff06000109200265e9a82c87a1c47d71d7e29ae362026cb0bc652ac5b8cd,,1<CR><LF>
+BLESCAN:"5f:ff:63:a9:d9:59",-86,1eff0600010920023c1e61765f9d80a4616cfb2f6c223edfad9824f60e1282,,1<CR><LF>
+BLESCAN:"3c:25:7a:da:cf:38",-86,1eff06000109200270ee0b839d6cbcf55f89c32a9e8257df29112a54abb5ad,,1<CR><LF>
+BLESCAN:"35:83:05:7c:53:5f",-83,1eff060001092002be6faa67e20acc1bb4190d045ba98c81bf0e6e0aeefba5,,1<CR><LF>
+BLESCAN:"44:c0:98:3b:f7:d4",-80,1eff060001092002dcd1bcf0782f8ba2c18cebfa82e5b7b002a564ef554259,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-59,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-59,,,0<CR><LF>
+BLESCAN:"eb:aa:52:7b:
5/24/2022 12:48:59 [RX] - 5a:c5",-59,0201061209534e4d3437362d4232304c333330303235,,1<CR><LF>
+BLESCAN:"eb:aa:52:7b:5a:c5",-59,,11077c16a55eba11cb920c497fb801119a56,1<CR><LF>
+BLESCAN:"49:d8:7e:f5:48:c3",-78,1eff0600010920020724004497c99c41986b8e6d869cbd47a11cb5976f796b,,1<CR><LF>
+BLESCAN:"1e:c1:29:02:a4:fb",-72,1eff06000109200242fa2a42ddf24c6ec01a989ef0fd9971f9295dd54fff7d,,1<CR><LF>
+BLESCAN:"18:eb:83:62:d9:34",-79,1eff0600010920026f1a3cc9858fbf0b91c9928ccefbfb3889720eebba6127,,1<CR><LF>

5/24/2022 12:49:02 [TX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/24/2022 12:49:02 [RX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/24/2022 12:49:06 [RX] - +BLECONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

5/24/2022 12:49:12 [RX] - +BLEDISCONN:0,"34:b4:72:b2:68:52"<CR><LF>

It is still disconnecting with your shared software.

Luxin812, 2022-05-24:
> 
I changed some configuration in the attached temporary test firmware, can you help me test this firmware again?
[factory.zip](https://github.com/espressif/esp-at/files/8760667/factory.zip)

gkg-7, 2022-05-25:
Yes, sorry for the late reply. I am trying this firmware in my end now.

gkg-7, 2022-05-25:
I have tried this as well.

5/25/2022 15:44:54 [TX] - AT+BLESCAN=1,1<CR><LF>

5/25/2022 15:44:54 [RX] - AT+BLESCAN=1,1<CR><LF>
<CR><LF>
OK<CR><LF>
+BLESCAN:"44:b3:1b:c1:0c:cb",-76,02011a020a070aff4c001005031c4853a7,,1<CR><LF>
+BLESCAN:"44:b3:1b:c1:0c:cb",-78,,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-88,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-86,,,0<CR><LF>
+BLESCAN:"0d:b3:1e:26:d1:c7",-82,1eff0600010920022362b95c10621f16291727881eb1f825b09afa7ad36887,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-90,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-95,,,0<CR><LF>
+BLESCAN:"0d:b3:1e:26:d1:c7",-83,1eff0600010920022362b95c10621f16291727881eb1f825b09afa7ad36887,,1<CR><LF>
+BLESCAN:"44:b3:1b:c1:0c:cb",-75,02011a020a070aff4c001005031c4853a7,,1<CR><LF>
+BLESCAN:"44:b3:1b:c1:0c:cb",-75,,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-88,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-84,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-88,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-87,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-86,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-85,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-80,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-80,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-79,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-80,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-79,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-78,,,0<CR><LF>
+BLESCAN:"d6:4c:61:be:07:14",-81,07ff4c0012020000,,1<CR><LF>
+BLESCAN:"0d:b3:1e:26:d1:c7",-78,1eff0600010920022362b95c10621f16291727881eb1f825b09afa7ad36887,,1<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-84,02011a020a080aff4c0010052b18383ddb,,1<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-84,,,1<CR><LF>

5/25/2022 15:44:57 [TX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/25/2022 15:44:57 [RX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/25/2022 15:45:00 [RX] - +BLECONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

5/25/2022 15:45:10 [TX] - AT+BLEGATTCPRIMSRV=0<CR><LF>

5/25/2022 15:45:10 [RX] - AT+BLEGATTCPRIMSRV=0<CR><LF>

5/25/2022 15:45:15 [RX] - +BLEDISCONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

I have tried this from my home where less BLE devices are there, it still disconnects.

gkg-7, 2022-05-25:
5/25/2022 15:55:18 [RX] - AT+BLESCAN=1,1<CR><LF>
<CR><LF>
OK<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,,,0<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-76,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,02011a020a070bff4c0010062a196c1f3478,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,,,1<CR><LF>
+BLESCAN:"41:ef:a7:b6:41:47",-61,02011a020a0c0bff4c001006101978073f58,,1<CR><LF>
+BLESCAN:"41:ef:a7:b6:41:47",-61,,,1<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-77,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,,,0<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-58,02011a020a070bff4c0010062a196c1f3478,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>
+BLESCAN:"44:b3:1b:c1:0c:cb",-96,02011a020a070aff4c0010050b1c4853a7,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-67,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-67,,,0<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,02011a020a070bff4c0010062a196c1f3478,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-68,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,,,0<CR><LF>
+BLESCAN:"41:ef:a7:b6:41:47",-55,02011a020a0c0bff4c001006101978073f58,,1<CR><LF>
+BLESCAN:"41:ef:a7:b6:41:47",-55,,,1<CR><LF>
+BLESCAN:"f6:56:92:09:41:0a",-81,07ff4c0012020000,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-68,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-67,,,0<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-77,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-75,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,02011a020a070bff4c0010062a196c1f3478,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-66,,,0<CR><LF>

5/25/2022 15:55:21 [TX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/25/2022 15:55:21 [RX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>
+BLECONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

5/25/2022 15:55:24 [TX] - AT+BLEGATTCPRIMSRV=0<CR><LF>

5/25/2022 15:55:24 [RX] - AT+BLEGATTCPRIMSRV=0<CR><LF>

5/25/2022 15:55:26 [RX] - +BLEDISCONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

5/25/2022 15:57:33 [TX] - AT+BLESCAN=1,1<CR><LF>

5/25/2022 15:57:33 [RX] - AT+BLESCAN=1,1<CR><LF>
<CR><LF>
OK<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-91,02011a020a080aff4c0010052b18383ddb,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,02011a020a070bff4c0010062a196c1f3448,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-59,,,1<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-80,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-86,02011a020a080aff4c0010052b18383ddb,,1<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-86,,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,02011a020a070bff4c0010062a196c1f3448,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,,,1<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-79,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-67,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-67,,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>
+BLESCAN:"5f:92:fe:fe:1a:7f",-81,02011a020a070aff4c0010050f1c84d170,,1<CR><LF>
+BLESCAN:"5f:92:fe:fe:1a:7f",-81,,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,02011a020a070bff4c0010062a196c1f3448,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,,,1<CR><LF>
+BLESCAN:"cf:d6:08:69:b1:89",-83,07ff4c0012020003,,1<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-78,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"5f:92:fe:fe:1a:7f",-84,02011a020a070aff4c0010050f1c84d170,,1<CR><LF>
+BLESCAN:"5f:92:fe:fe:1a:7f",-85,,,1<CR><LF>
+BLESCAN:"64:a1:24:3e:17:43",-83,02011a020a080aff4c0010052b18383ddb,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,02011a020a070bff4c0010062a196c1f3448,,1<CR><LF>
+BLESCAN:"5a:ae:ec:fc:8a:1d",-60,,,1<CR><LF>
+BLESCAN:"28:d0:43:af:d5:ff",-80,1eff060001092002c7da2ee162239384276f1cd9e31e160cebab5367ce0251,,1<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-64,0201060a09457370726573736966030302a0,,0<CR><LF>
+BLESCAN:"34:b4:72:b2:68:52",-65,,,0<CR><LF>

5/25/2022 15:57:36 [TX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/25/2022 15:57:36 [RX] - AT+BLECONN=0,"34:b4:72:b2:68:52"<CR><LF>

5/25/2022 15:57:38 [RX] - +BLECONN:0,"34:b4:72:b2:68:52"<CR><LF>
<CR><LF>
OK<CR><LF>

5/25/2022 15:57:44 [RX] - +BLEDISCONN:0,"34:b4:72:b2:68:52"<CR><LF>

The Server RSSI keeps changing and it disconnects as soon as I connect or send any commad.

Luxin812, 2022-05-26:
Hi, I am very sorry, I would like to confirm one thing again, regarding your pcb design you have gone through this channel (https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design- review ) Has it been reviewed at Espressif?
I would like to ask you abruptly, are you currently working in China? If so, can you send us a board? After the issue is solved, it will be returned to you.

gkg-7, 2022-05-26:
Yes we had done a review by espressif but not through the channel you mentioned here. We had sent our files to your Indian vendor RaByte and they had forwarded the things to you for review. 
We are from Trimble, India. We are not from China.
Yes we can send the board to you. 

I am sorry just to confirm again, did your hardware team receive any documents from your Indian Vendor RaByte?
I am attaching the review report by espressif here:
[#2295_【Trimble Mobility】Hardware Review Report_202202330_EN.pdf](https://github.com/espressif/esp-at/files/8776382/2295_.Trimble.Mobility.Hardware.Review.Report_202202330_EN.pdf)

Luxin812, 2022-05-26:
> Yes we had done a review by espressif but not through the channel you mentioned here. We had sent our files to your Indian vendor RaByte and they had forwarded the things to you for review. We are [...]

I'm not sure about this, maybe you can confirm this with your agent. If you are not in China, mailing boards back and forth can take a lot of time. You can purchase a packet capture network card online, and then view the packet capture tutorial. Then provide BLE disconnected packets?

gkg-7, 2022-05-26:
We can send a board for you to test and provide a solution. Meanwhile could you help us with doing a review of our hardware design and the module placement? So that we know that the performance issue is not caused because of that. We at this point can't do a re-spin, is there any way espressif can tune or customize a firmware for good performance with the current design? In mass production we have a huge target. Please help us here.

On the other we haven't used the Packet capture network card, so its actually new to us.

Luxin812, 2022-05-26:
> We can send a board for you to test and provide a solution. Meanwhile could you help us with doing a review of our hardware design and the module placement? So that we know that the performance issue is not caused because of that. We at this point can't do a re-spin, is there any way espressif can tune or customize a firmware for good performance with the current design? In production we have a huge target. Please help us here.
> 
> On the other hand, we haven't used the Packet capture network card, so its actually new to us.

Can you help to download the BLE demo
([ble_connect_demo.zip](https://github.com/espressif/esp-at/files/8778169/ble_connect_demo.zip))
 test in the attachment to see if the BLE connection will be disconnected? For the burning method, please refer to the flash_args file in the attachment.

Luxin812, 2022-05-27:
Maybe you have tested the connection function of ESPAT BLE with ESP32-C3-DevKitM-1 development board, what is the result?
[image]

gkg-7, 2022-05-27:
Yes I have tested ESP-AT with the Dev Kit. It doesn't disconnect at all. It works perfectly.

gkg-7, 2022-05-30:
> Can you help to download the BLE demo ([ble_connect_demo.zip](https://github.com/espressif/esp-at/files/8778169/ble_connect_demo.zip)) test in the attachment to see if the BLE connection will be [...]

I have flashed this BLE demo firmware into the board. But its is not discoverable when BLE scanning is done on the client side.

gkg-7, 2022-05-30:
I have tested once more with your  AT version:2.4.0.0(4c6eb5e - ESP32C3 - May 20 2022 03:11:59), SDK version:qa-test-v4.3.3-20220423,compile time(5641e0a):May 20 2022 11:13:44, Bin version:2.4.0(MINI-1) firmware.

I have attached the log here. The server gets connected and operates well and then later has disconnection issue in a similar RSSI level. As you can see in this log the server gets connected to the client in -45 dB and operates well exchanging data but gets disconnected in the same RSSI level for another instance.
[ESP32MINIClient_LOG_Timestamp.txt](https://github.com/espressif/esp-at/files/8796355/ESP32MINIClient_LOG_Timestamp.txt)

gkg-7, 2022-05-30:
I think I have found out the reason why these disconnections happen. 
Scenario:
1. When its just basic connection and disconnection, there is no issue.
2. When its SPP connection between the two modules and if we shutdown or reset the modules without ending the SPP connection the server or the client goes into a scenario where it doesn't retain a connection. If possible please try this at your end with your latest software.

Steps to test:
1. Initialize one ESP32C3 as your server and one as your client. 
2. Make the connection.
3. Enable the SPP connection and transfer some data and then immediately reset or shutdown any of those modules.

Please let me know if you see a similar issue here.

Luxin812, 2022-05-30:
Can you send me instructions to reproduce the issue? When the device is disconnected without exiting SPP, it will automatically reconnect，the relevant configuration see AT+BLESPPCFG command (https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/BLE_AT_Commands.html#at-blesppcfg-query-set-bluetooth-le-spp-parameter)

gkg-7, 2022-05-30:
So to end the SPP mode, I should send "+++"?

gkg-7, 2022-05-30:
> Can you send me the AT command sequence to reproduce the issue? When the device is disconnected without exiting SPP, it will automatically reconnect，the relevant configuration see AT+BLESPPCFG command (https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/BLE_AT_Commands.html#at-blesppcfg-query-set-bluetooth-le-spp-parameter)

I have mentioned the steps to reproduce the issue above, I am putting it here once again,
Steps to test:

1. Initialize one ESP32C3 as your server and one as your client.
2. Make the connection.
3. Enable the SPP connection and transfer some data and then immediately reset or shutdown any of those modules.

Luxin812, 2022-05-31:
> So to end the SPP mode, I should send "+++"?

When sending a packet containing only +++, the device exits SPP mode and returns to normal command mode. Please wait at least one second before sending the next AT command.

Luxin812, 2022-05-31:
> I have mentioned the steps to reproduce the issue above, I am putting it here once again, Steps to test:
>
> 1. Initialize one ESP32C3 as your server and one as your client.
> 2. Make the connection.
> 3. [...]

Hi, I have seen your reproduction steps, I am not sure if I understand exactly what you are trying to express.
If I didn't understand wrong you meant to express that resetting or shutting down the module (either server or client) while the server or client is not out of SPP mode will cause the BLE connection to be disconnected right?
If my understanding is correct, probably this BLE disconnect is a normal behavior. If I'm wrong, could you please tell me more about what steps you took and what you expected in the end, but what would actually happen?

gkg-7, 2022-05-31:
> > So to end the SPP mode, I should send "+++"?
> 
> When sending a packet containing only +++, the device exits SPP mode and returns to normal command mode. Please wait at least one second before sending the next AT command.

I tried sending +++\r\n and it did not exit from the SPP, I send this command from the client side.

Luxin812, 2022-05-31:
> > > So to end the SPP mode, I should send "+++"?
> > 
> > 
> > When sending a packet containing only +++, the device exits SPP mode and returns to normal command mode. Please wait at least one second before sending the next AT command.
> 
> I tried sending +++\r\n and it did not exit from the SPP, I sent this command from the client side. Anything else is required while sending the "+++"?

You should send '+++' instead of '+++\r\n'.

gkg-7, 2022-05-31:
> Hi, I have seen your reproduction steps, I am not sure if I understand exactly what you are trying to express. If I didn't understand wrong you meant to express that resetting or shutting down the [...]

Yes you understood it right. The BLE disconnects when shutdown or reset during the SPP mode. The issue I am facing is that, it disconnects in every seconds of connection attempts after it disconnected once from the reset or shutdown. I mean to say that once it gets exited from the SPP mode with a abrupt shutdown or reset, it is not able to connect in subsequent attempts. 

> You should send '+++' instead of '+++\r\n'(https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#data-mode-only-exit-from-data-mode).

This worked. Thanks.

gkg-7, 2022-05-31:
What is the ideal RSSI value for a module with both Wifi and BLE ON?

Luxin812, 2022-05-31:
> What is the ideal RSSI value for a module with both Wifi and BLE ON?

Maybe you can open an issue again

gkg-7, 2022-05-31:
Sure, I'll do that. But have you tested my scenario at your end?

Luxin812, 2022-06-01:
> Sure, I'll do that. But have you tested my scenario at your end?

Yes, I tested it with ESP32C3 v2.4 firmware on two ESP32C3-MINI-Devkit boards and it did not reproduce your issue, you can see how my reproduction steps compare to yours here different? Maybe your issue has been tried on the ESP32C3-MINI-Devkit development board? My steps to reproduce are as follows:
server:
```
AT+BLEINIT=2
AT+BLEGATTSSRVCRE
AT+BLEGATTSSRVSTART
AT+BLEADVDATA="0201060A09457370726573736966030302A0"
AT+BLEADVSTART
```
client:
```
AT+BLEINIT=1
AT+BLESCAN=1,3
AT+BLECONN=0,<mac>
```
So far, the BLE connection is successfully established
server:
```
AT+BLEGATTSSRV?
AT+BLEGATTSCHAR?
```
client:
```
AT+BLEGATTCPRIMSRV=0
AT+BLEGATTCCHAR=0,3
AT+BLESPPCFG=1,3,5,3,7
AT+BLESPP
```
server:
```
AT+BLESPPCFG=1,1,7,1,5
AT+BLESPP
```
At this time, after the client sends the data 'ABCDEF', directly press the RST key on the client development board to restart the device hardware, the client will print 'ready', and then the client sends the following command:
```
AT+BLEINIT=1
AT+BLESCAN=1,3
AT+BLECONN=0,"34:b4:72:b2:68:52"
```
At this time, the BLE connection is successfully established, and the connection will not be disconnected after 5s.

gkg-7, 2022-06-02:
We tried these steps and we are not receiving that disconnection anymore. I re-flashed the V2.4 software and it started to work. If something is corrupted will we get a similar issue?

gkg-7, 2022-06-02:
UART4 RX = AT+CWMODE=3
UART4 RX = 
UART4 RX = OK
UART4 RX = AT+BLEINIT=2
UART4 RX = 
UART4 RX = OK
UART4 RX = AT+BLENAME="SNM476M-123456789"
UART4 RX = 
UART4 RX = OK
UART4 RX = AT+BLEGATTSSRVCRE
UART4 RX = 
UART4 RX = OK
UART4 RX = AT+BLEGATTSSRVSTART
UART4 RX = 
UART4 RX = OK
UART4 RX = AT+BLEADVSTART
UART4 RX = OK

I tried these steps to set name and I am not able to see this name on my phone.

Luxin812, 2022-06-02:
Hi,

Just came across a post with the same question as yours, you may be able to find the answer here, I hope it helps you.

closed 2023-06-05 by ustccw