espressif/esp-at #636: limitation on the length of the URL in the AT+HTTPCLIENT command

heatsiphon, 2022-05-25:
I am trying to pass a long querystring (actually a long string that will be around 330 characters long, but the AT+HTTPCLIENT generates an "ERROR" when the string is anything longer than ~250 characters.

Since you all won't release the source code for the AT+HTTPCLIENT command, can you please tell me why there is a string length limitation and PREFERABLY make the limit longer, like 500 characters on the URL??

ustccw, 2022-05-26:
@heatsiphon Hi, as the [AT+HTTPCLIENT command first note](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#notes) mentioned: If the length of the entire command containing the URL exceeds 256 bytes, please use the [AT+HTTPURLCFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#cmd-httpurlcfg) command to preset the URL first, and then set the <”url”> parameter of this command to "".

btw, you also can try the [AT+HTTPCGET](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#http-at) and [AT+HTTPCPOST](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#http-at) commands.

heatsiphon, 2022-05-26:
I also tried a different method:

AT+HTTPURLCFG=341

OK

>**(SAME STRING AS ABOVE)**
SET OK

AT+HTTPCGET=""

ERROR

ustccw, 2022-05-27:
@heatsiphon An example to use long URL like the following:
the URL is 341 bytes, Could you please have a test. maybe i should add more examples to AT docs.
```
https://download.espressif.com/esp_share/temp/1234567890123456789012345678901234567890/esp32-c5-11111111111111111112222222222222222223333333333333333333333344444444444444444444444455555555555555555555555555666666666666666666666666666677777777777777777777777777778888888888888888888888888899999999999999999990000000000000000000000000aaaaa.jpg
```

[image]

Beside this, there are more things i have to say.
1. Send `AT+HTTPURLCFG?` to query current URL length and the URL before `AT+HTTPCGET` or `AT+HTTPCLIENT`.
2. If received the ERROR like the following, there should be more characters than expected (double check that if `\r\n` is appended to the tail by mistake).
```
    SET OK

    ERROR
```
3. `AT+HTTPCLIENT=2,3,"","","",2` command itself is okey, to know why it returns `ERROR`, could you please provide the real URL for me, i'll check it.

4. May i know what does this mean: "It is not accepting the URL"

ShaneRolin, 2022-09-12:
AT+HTTPURLCFG=380

https://remote.heatsiphon.com/hs/warning-b.php?err=999&s=21A333&data=23|761|988|522|1|0|0|0|0|0|1|1|24|761|988|522|1|0|0|0|0|0|1|1|25|761|988|522|1|0|0|0|0|0|1|1|26|761|988|522|1|0|0|0|0|0|1|1|27|761|988|522|1|0|0|0|0|0|1|1|28|761|988|522|1|0|0|0|0|0|1|1|29|761|988|522|1|0|0|0|0|0|1|1|30|761|988|522|1|0|0|0|0|0|1|1|31|761|988|522|1|0|0|0|0|0|1|1|32|761|988|522|1|0|0|0|0|0|1|1|

AT+HTTPURLCFG=379

OK

>
SET OK
AT+HTTPCGET=""

ERROR

It will not send the data to the URL.  Please show me the correct command syntax.

heatsiphon, 2022-09-12:
AT+HTTPURLCFG=380

https://remote.heatsiphon.com/hs/warning-b.php?err=999&s=21A333&data=23|761|988|522|1|0|0|0|0|0|1|1|24|761|988|522|1|0|0|0|0|0|1|1|25|761|988|522|1|0|0|0|0|0|1|1|26|761|988|522|1|0|0|0|0|0|1|1|27|761|988|522|1|0|0|0|0|0|1|1|28|761|988|522|1|0|0|0|0|0|1|1|29|761|988|522|1|0|0|0|0|0|1|1|30|761|988|522|1|0|0|0|0|0|1|1|31|761|988|522|1|0|0|0|0|0|1|1|32|761|988|522|1|0|0|0|0|0|1|1|

OK

SET OK
AT+HTTPCGET=""

ERROR

It will not send the data to the URL. Please show me the correct command syntax.  I cannot get this to work at all.

ustccw, 2022-09-13:
[image]

@heatsiphon the url length seems to be 379 bytes, not 380 bytes. could you please check it again.

heatsiphon, 2022-09-13:
This is what I get.  I have a very hard time getting that command to ever give me an OK without ERROR.

AT+HTTPURLCFG=379

OK

>https://remote.heatsiphon.com/hs/warning-b.php?err=999&s=21A333&data=23|761|988|522|1|0|0|0|0|0|1|1|24|761|988|522|1|0|0|0|0|0|1|1|25|761|988|522|1|0|0|0|0|0|1|1|26|761|988|522|1|0|0|0|0|0|1|1|27|761|988|522|1|0|0|0|0|0|1|1|28|761|988|522|1|0|0|0|0|0|1|1|29|761|988|522|1|0|0|0|0|0|1|1|30|761|988|522|1|0|0|0|0|0|1|1|31|761|988|522|1|0|0|0|0|0|1|1|32|761|988|522|1|0|0|0|0|0|1|1|

SET OK

ERROR
AT+HTTPCGET=""
+HTTPCGET:12,Mail Sent.

OK

heatsiphon, 2022-09-13:
@ustccw is there a reason this command MUST have the number of bytes submitted as part of AT+HTTPURLCFG?  Why can't it just take the submitted string and do a LEN() on it and use that without us having to send a # of bytes?

On the one above, I used 379 and then copied that exact line - 379 bytes, and pasted it into the serial terminal then pressed SEND, and I still get SET OK    ERROR.  Why is this??

ustccw, 2022-09-14:
could you please check the tail part of this comment: https://github.com/espressif/esp-at/issues/636#issuecomment-1139422143

closed 2023-06-05 by ustccw