espressif/esp-at #638: How to Enable

Nagarajan2507, 2022-05-30:
Hi

I am following "[ESP32C3-AT][v2.4.0.0]User-Guide.pdf" Chapter 5 How to Compile and Develop Your Own AT Project.

Module: ESP32-C3-WROOM-2
IDF VER: 4.3.2

Component config -> AT -> AT FS Command Support is checked,

[image]

Facing compile errors as shown below:

D:\Espressif\proj\esp-at>python build.py build
Platform name:ESP32C3   Module name:MINI-1      Silence:0
Ready to set up environment variables..
PATH is D:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin;D:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin;D:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\bin;D:\Espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin;D:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\Espressif\tools\cmake\3.16.4\bin;D:\Espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\bin;D:\Espressif\tools\ninja\1.10.2\;D:\Espressif\tools\idf-exe\1.0.1\;D:\Espressif\tools\ccache\3.7\;D:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;D:\Espressif\esp-idf\tools;D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\;D:\Espressif\tools\idf-git\2.34.2\cmd\;D:\Espressif;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\VSShell\Common7\IDE\;C:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\;C:\Program Files\Microsoft SQL Server\100\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\100\DTS\Binn\;C:\Program Files\Microsoft SQL Server\100\DTS\Binn\;C:\Users\Nagarajan\AppData\Local\Android\Sdk\platform-tools;C:\Users\Nagarajan\AppData\Local\Android\Sdk\tools;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\FTDI\Vinculum II Toolchain\Tools\bin;C:\Users\Nagarajan\AppData\Local\Programs\Python\Python37;C:\Program Files (x86)\Common Files\PerkinElmer\;C:\Program Files\OpenSSL-Win64\bin;C:\Users\Nagarajan\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Sophos\Sophos SSL VPN Client\bin;C:\Users\Nagarajan\AppData\Local\Programs\Microsoft VS Code\bin
IDF_PYTHON_ENV_PATH is D:\Espressif\python_env\idf4.3_py3.8_env
sys.platform is win32
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2021r2-8.4.0. To use it, run 'D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool xtensa-esp32s2-elf found in PATH: esp-2021r2-8.4.0. To use it, run 'D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool xtensa-esp32s3-elf found in PATH: esp-2021r2-8.4.0. To use it, run 'D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool riscv32-esp-elf found in PATH: esp-2021r2-8.4.0. To use it, run 'D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool openocd-esp32 found in PATH: v0.10.0-esp32-20211111. To use it, run 'D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe esp-idf\tools\idf_tools.py export --prefer-system'
PATH is D:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;D:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;D:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;D:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;D:\Espressif\proj\esp-at\esp-idf\tools;%PATH%;D:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin;D:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin;D:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\bin;D:\Espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin;D:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\Espressif\tools\cmake\3.16.4\bin;D:\Espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\bin;D:\Espressif\tools\ninja\1.10.2\;D:\Espressif\tools\idf-exe\1.0.1\;D:\Espressif\tools\ccache\3.7\;D:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;D:\Espressif\esp-idf\tools;D:\Espressif\python_env\idf4.3_py3.8_env\Scripts\;D:\Espressif\tools\idf-git\2.34.2\cmd\;D:\Espressif;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\VSShell\Common7\IDE\;C:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\;C:\Program Files\Microsoft SQL Server\100\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\100\DTS\Binn\;C:\Program Files\Microsoft SQL Server\100\DTS\Binn\;C:\Users\Nagarajan\AppData\Local\Android\Sdk\platform-tools;C:\Users\Nagarajan\AppData\Local\Android\Sdk\tools;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\FTDI\Vinculum II Toolchain\Tools\bin;C:\Users\Nagarajan\AppData\Local\Programs\Python\Python37;C:\Program Files (x86)\Common Files\PerkinElmer\;C:\Program Files\OpenSSL-Win64\bin;C:\Users\Nagarajan\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Sophos\Sophos SSL VPN Client\bin;C:\Users\Nagarajan\AppData\Local\Programs\Microsoft VS Code\bin
IDF_PYTHON_ENV_PATH is D:\Espressif\python_env\idf4.3_py3.8_env
Executing action: all (aliases: build)
Running ninja in directory d:\espressif\proj\esp-at\build
Executing "ninja all"...
[0/1] Re-running CMake...
silence:0  sdkconfig:D:/Espressif/proj/esp-at/module_config/module_esp32c3_default/sdkconfig.defaults
esp-at branch: master
esp-at docs version: latest
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file D:/Espressif/proj/esp-at/sdkconfig
Loading defaults file D:/Espressif/proj/esp-at/module_config/module_esp32c3_default/sdkconfig.defaults...
-- App "esp-at" version: v2.4.0.0-2-g40ae56a
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-nano.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script D:/Espressif/proj/esp-at/build/esp-idf/esp32c3/esp32c3_out.ld
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp32c3/ld/esp32c3.project.ld.in
-- Adding linker script D:/Espressif/proj/esp-at/esp-idf/components/esp32c3/ld/esp32c3.peripherals.ld
silence:0 , ld core:D:/Espressif/proj/esp-at/components/at/lib/libesp32c3_at_core.a
silence:0 , ld core:D:/Espressif/proj/esp-at/components/clouds/qcloud/lib/libesp32c3_qcloud_core.a
Parsing CSV input...
Verifying table...
-- Components: app_trace app_update asio at bootloader bootloader_support bt cbor clouds cmock coap console customized_partitions cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb unity vfs wear_levelling wifi_provisioning wpa_supplicant
[... cut ...]
  File "D:/Espressif/proj/esp-at/tools/esp_at_factory_bin_combine.py", line 47, in esp32_at_combine_bin
    with open(os.path.join(build_dir, bin_file), 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'D:/Espressif/proj/esp-at/build\\customized_partitions/fatfs.bin'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
A fatal error occurred: idf.py build failed

I am struggling to success more than one week.
Any help is very much appreciated.

Regards
V.Nagarajan

ustccw, 2022-06-06:
@Nagarajan2507 Thanks for the report! will fix it in two weeks.

Nagarajan2507, 2022-06-08:
Thanks for looking into this.
Waiting for fix.

ustccw, 2022-06-15:
@Nagarajan2507 
Could you please unzip the following zip and copy it to esp-at/build/customized_partitions/fatfs.bin, then try to compile again.   
[fatfs.zip](https://github.com/espressif/esp-at/files/8907716/fatfs.zip)

### Reason for compilation failure
In the logs you provided, it mentioned that “ 'make' is not recognized as an internal or external command”.
it means that your windows OS is lack of gcc and g++ compiler. But our [fatfs partition](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32c3_default/at_customize.csv#L16) is generated by [mkfatfs](https://github.com/espressif/esp-at/tree/master/tools/mkfatfs) component, this component is from https://github.com/jkearins/ESP32_mkfatfs, and it need gcc and g++ compiler.
So esp-at project compiled failed due to lack of such compiler.

### Possible workaround for you
1. Use the generated `fatfs.bin` directly, as i mentioned at the beginning.
2. If you have a linux os, just run `sudo apt install gcc g++`, this would be the easier way.
3. Download and install gcc and g++ tools directly into your windows OS.
4. Set up a windows environment where installing such additional tools is possible. Eg. https://chocolatey.org/.

### Actions for ESP-AT project
We're porting ESP-IDF release/v5.0 to esp-at project, once it‘s done, esp-at will use cross-platform [FatFs Partition Generator](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/fatfs.html#fatfs-partition-generator) instead of deprecated  [mkfatfs](https://github.com/espressif/esp-at/tree/master/tools/mkfatfs) tool.

Nagarajan2507, 2022-06-20:
Thanks for your support.
Workaround 1 is OK
Workaround 3 is not able to succeed. Any help for this.

closed 2023-06-05 by ustccw