espressif/esp-at #642: Failed to start ESP13 (ESP-WROOM-02) after installing AT Firmware

kbeaugrand, 2022-06-04:
Hi, 

I have a Iduino ESP 13 (with an ESP 8266 chip). 
I'd like to upgrade the original AT firmware to the last provided by ESP-IDF (at that time ESP8266-AT-V2.2.1.0). 

Uploading the firmware works fine with this command line : 
```
esptool --chip auto --port COM4 --baud 115200 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x9000 ota_data_initial.bin 0x0 bootloader/bootloader.bin 0x10000 esp-at.bin 0xF0000 at_customize.bin 0xFC000 customized_partitions/client_ca.bin 0x106000 customized_partitions/mqtt_key.bin 0x104000 customized_partitions/mqtt_cert.bin 0x108000 customized_partitions/mqtt_ca.bin 0xF1000 customized_partitions/factory_param.bin 0xF8000 customized_partitions/client_cert.bin 0xFA000 customized_partitions/client_key.bin
```

Download logs : 

```
esptool.py v4.1
Serial port COM4
Connecting................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:bf:c0:e2:3c:dc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Flash will be erased from 0x00000000 to 0x00002fff...
Flash will be erased from 0x00010000 to 0x000defff...
Flash will be erased from 0x000f0000 to 0x000f0fff...
Flash will be erased from 0x000fc000 to 0x000fcfff...
Flash will be erased from 0x00106000 to 0x00106fff...
Flash will be erased from 0x00104000 to 0x00104fff...
Flash will be erased from 0x00108000 to 0x00108fff...
Flash will be erased from 0x000f1000 to 0x000f1fff...
Flash will be erased from 0x000f8000 to 0x000f8fff...
Flash will be erased from 0x000fa000 to 0x000fafff...
Compressed 3072 bytes to 112...
Wrote 3072 bytes (112 compressed) at 0x00008000 in 0.1 seconds (effective 309.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.0 seconds (effective 2036.0 kbit/s)...
Hash of data verified.
Flash params set to 0x0230
Compressed 10224 bytes to 6843...
Wrote 10224 bytes (6843 compressed) at 0x00000000 in 0.8 seconds (effective 100.2 kbit/s)...
Hash of data verified.
Compressed 844288 bytes to 529252...
Wrote 844288 bytes (529252 compressed) at 0x00010000 in 46.9 seconds (effective 143.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 204...
Wrote 3072 bytes (204 compressed) at 0x000f0000 in 0.1 seconds (effective 256.9 kbit/s)...
Hash of data verified.
Compressed 2344 bytes to 1499...
Wrote 2344 bytes (1499 compressed) at 0x000fc000 in 0.2 seconds (effective 90.0 kbit/s)...
Hash of data verified.
Compressed 1692 bytes to 1322...
Wrote 1692 bytes (1322 compressed) at 0x00106000 in 0.2 seconds (effective 84.2 kbit/s)...
Hash of data verified.
Compressed 1168 bytes to 895...
Wrote 1168 bytes (895 compressed) at 0x00104000 in 0.1 seconds (effective 73.3 kbit/s)...
Hash of data verified.
Compressed 1172 bytes to 914...
Wrote 1172 bytes (914 compressed) at 0x00108000 in 0.1 seconds (effective 73.8 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 83...
Wrote 4096 bytes (83 compressed) at 0x000f1000 in 0.1 seconds (effective 510.4 kbit/s)...
Hash of data verified.
Compressed 2344 bytes to 1487...
Wrote 2344 bytes (1487 compressed) at 0x000f8000 in 0.2 seconds (effective 106.3 kbit/s)...
Hash of data verified.
Compressed 3368 bytes to 2526...
Wrote 3368 bytes (2526 compressed) at 0x000fa000 in 0.3 seconds (effective 88.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I get the arguments from the documentation and download.config file from the firmware.

But after starting the device, I cannot access to AT command on Serial with 115200 baud. After changing my console to 78440 bauds I can see some errors : 

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7332, room 16 
tail 4
chksum 0x14
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0x9d
load 0x3ffe8420, len 3532, room 4 
tail 8
chksum 0x69
csum 0x69
[00]
```

Could someone tell me what I'm doing wrong ?

ustccw, 2022-07-04:
@kbeaugrand  That module seems to come from [Ai-Thinker company](http://www.ai-thinker.com/product-3.html), not espressif. For getting a better technical support, i would like to recommend you seek help from [Ai-Thinker](http://www.ai-thinker.com/contact-1.html).

Of course, you also can consider to buy the module from [espressif](https://products.espressif.com/#/product-selector?language=en&names=), and read the [roadmap](https://github.com/espressif/esp-at#project-roadmap) of esp-at project firstly.

kbeaugrand, 2022-07-17:
@ustccw , 

You're right, the module was provided by Ai-Thinker company. However I assumed that this should works since it might be the "same" kind of hardware. But I understand your point.

Thank you for your answer. 

Regards,

closed 2022-07-17 by kbeaugrand

darkgirl, 2022-09-18:
> @kbeaugrand That module seems to come from [Ai-Thinker company](http://www.ai-thinker.com/product-3.html), not espressif. For getting a better technical support, i would like to recommend you seek [...]

我记得安信可不也是用的你们的芯片吗? 同样的芯片为什用你们的固件连启动都有问题?