espressif/esp-at #643: ESP-AT supports the existence of Bluetooth LE server and Bluetooth LE client at the same time?

bolotkalil, 2022-06-08:
Hello, 

Anyone know how to use GATT server and client at the same time with esp-at? 
I found on the documentation is it possible: [Look here](https://drive.google.com/file/d/1by1QR_-Ax-hyGIck8Kivr6ykIDjHTy9a/view?usp=sharing)
[Link to document](https://github.com/espressif/esp-at/blob/v2.4.0.0/docs/en/AT_Command_Examples/bluetooth_le_at_examples.rst)

I couldn't find any example how to use it..

talon75, 2022-07-27:
Hi,
I'm facing the very same issue. 
Unfortunately, the cited example keeps the client/server roles separated. Moreover, the AT+BLEINIT notes state that "After the initialization, the Bluetooth LE role cannot be changed unless you run AT+RST to restart the system first and then re-initialize the Bluetooth LE role."

Could you please clarify this?

Luxin812, 2022-07-28:
Hi,

This document should be revised. In fact, after initializing the BLE role, if you want to change the BLE role, you can execute the AT+BLEINIT=0 command to cancel the BLE role, and then switch the BLE role. It is not necessary to restart the system to change the BLE role.
In fact, whether you are initializing the client or the server, you can use the GATT server and client at the same time with esp-at. You can synchronize your usage scenario here. Maybe I can provide you with the AT command sequence of this scenario.

talon75, 2022-07-28:
Hi, thank you.
This seems to conflict with another note:

> After the initialization, the Bluetooth LE role cannot be changed unless you run [AT+RST](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-rst) to restart the system first and then re-initialize the Bluetooth LE role.

Anyway, what I'd like to achieve is to have two concurrent sessions, so that I can create a daisy-chain between nodes, where every node is a server for the previous and a client for the next. This doesn't seem to be possible, right? Your statement suggests that client and server coexist in terms of function availability, but their radio activity is mutually exclusive.

Thanks again.

Luxin812, 2022-09-20:
> Hi, thank you. This seems to conflict with another note:
>
>
> Anyway, what I'd like to achieve is to have two concurrent sessions, so that I can create a daisy-chain between nodes, where every node is [...]

Hi,

Yes, unfortunately espat ble does not support the scenario where each node is the server of the previous node and the client of the next node.

closed 2023-06-05 by ustccw