espressif/esp-at #646: Issue with ESP32AT SPI Firmware

NaitikSanas, 2022-07-01:
Hello we have been developing driver for esp32-AT SPI for k210 MCU.  We have compiled version Release/v2.2.0.0_esp32. it works but its very unstable. for instance many times Handshake line is just pulled low then never pulled high again. any further communication to esp32 causes it to crash.  from source code we found that when Handshake line goes low Master needs to wait but many times Handshake line never pulled high back.

also sometimes what happens is if any command say CIPSEND is being in progress, and if i receive any data from network the flow for CIPSEND Command is disrupted causing it to crash. for both of these instances esp32 crashes after sending junk data to Host MCU. 

Documentation for AT-SPI seems like incorrect since it does not even provide correct SPI transaction format which we figured out by running ESP32-AT SPI master slave examples on two esp32 boards.

# ESP32 Backtrace : 
```
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
E (29317) at_spi_slave: RD_BUF error status 4
uru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4000921a  PS      : 0x00060034  A0      : 0x80007d16  A1      : 0x3ffbe2a0
A2      : 0x00800000  A3      : 0x00006e43  A4      : 0x00000000  A5      : 0x3ffcac70
A6      : 0x3ffbefd8  A7      : 0x00000001  A8      : 0x3ff40000  A9      : 0x00000033
A10     : 0x00800000  A11     : 0x3ff4001c  A12     : 0x00060021  A13     : 0x3ffbe370
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4008d844  LEND    : 0x4008d871  LCOUNT  : 0x00000000
0x4008d844: i2c_write_master at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:126 (discriminator 3)

0x4008d871: i2c_write_master at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:126 (discriminator 3)

Core  0 was running in ISR context:
EPC1    : 0x401d724f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4000921a
0x401d724f: uart_hal_write_txfifo at c:\users\sanas\esp\esp-at\build/../esp-idf/components/soc/src/hal/uart_hal_iram.c:35

Backtrace:0x40009217:0x3ffbe2a0 0x40007d13:0x3ffbe2c0 0x40007c69:0x3ffbe2e0 0x40008148:0x3ffbe300 0x40083071:0x3ffbe390 0x400828f2:0x3ffbe3c0 0x401c4510:0x3ffcad20 0x40157d5c:0x3ffcad50 0x40157e54:0x3ffcad80 0x401a38f9:0x3ffcadb0 0x40098a4d:0x3ffcadd0 0x401b44ff:0x3ffcae00 0x401b6e39:0x3ffcae50 0x401b6ead:0x3ffcae70 0x4009729e:0x3ffcae90 0x4008f781:0x3ffcaec0
0x40083071: at_spi_intr at c:\users\sanas\esp\esp-at\build/../components/spi_driver/at_spi_driver.c:535 (discriminator 1)

0x400828f2: _xt_lowint1 at c:\users\sanas\esp\esp-at\build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x401c4510: register_chipv7_phy at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:3699

0x40157d5c: esp_phy_rf_init at c:\users\sanas\esp\esp-at\build/../esp-idf/components/esp_wifi/src/phy_init.c:260

0x40157e54: esp_modem_sleep_exit at c:\users\sanas\esp\esp-at\build/../esp-idf/components/esp_wifi/src/phy_init.c:411

0x401a38f9: wifi_rf_phy_enable at ??:?

0x40098a4d: pm_dream at ??:?

0x401b44ff: pm_tbtt_process at ??:?

0x401b6e39: ppGetTaskHdl at ??:?

0x401b6ead: pp_timer_do_process at ??:?

0x4009729e: ppTask at ??:?

0x4008f781: vPortTaskWrapper at c:\users\sanas\esp\esp-at\build/../esp-idf/components/freertos/xtensa/port.c:143

```
Wave-forms Before crashing : 

[image]

ustccw, 2022-07-04:
@ESP-Coco please take a look.

Jacques-zhao, 2022-07-04:
Hi @NaitikSanas ,  since the ESP32 SPI has some problems that cannot be solved, we do not recommend you to use a pure SPI for development. We recommend using the SDIO SPI (https://github.com/espressif/esp-at/tree/master/examples/at_spi_master/sdspi) for the same functionality.

Pratiksha0102, 2022-07-23:
HELLO!
@ESP-Coco, I am having same issue, what are your thoughts on that?

Jacques-zhao, 2022-08-01:
Hi @Pratiksha0102  We recommend using the SDIO SPI (https://github.com/espressif/esp-at/tree/master/examples/at_spi_master/sdspi) for the same functionality. It can get a higher rate (over 10Mbps for TCP), and it's more stable(Many customers are already in mass production ). The original SPI driver had some problems that were very difficult to deal with.

NaitikSanas, 2022-08-05:
@Jacques-zhao  That's understandable we will move to SDIO since it looks more promising. thanks for your response.

closed 2022-08-05 by NaitikSanas