espressif/esp-at #647: Cant use comma in +MQTTPUB data field.

PontusO, 2022-07-01:
Using AT version:2.3.0.0(s-e98993f - ESP32C3 - Dec 23 2021 09:03:59), SDK version:v4.3.1-354-g98d34e5f6d, compile time(638cf57):May 25 2022 09:02:46, Bin version:2.3.0(MINI-1) I have a problem with sending the comma sign in the data field.
For instance the following line: 
AT+MQTTPUB=0,"messages/log","765933, 94875095",0,0
Produces the following response:
ERR CODE:0x01080002
ERR CODE:0x010a6044

ERROR
Whereas the following line 
AT+MQTTPUB=0,"messages/log","765933 94875095",0,0
works perfectly fine.

ustccw, 2022-07-04:
please use `\,` instead of `,`, since `,` should be escaped. See [Escape character syntax is needed if a string contains special characters, such as ,, ", or \\](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#at-command-types) for more details.

PontusO, 2022-07-05:
Excellent, thank you. Didn't note that in the documentation.