espressif/esp-at #649: When I use AT+CIPSTART="SSL" to make an https request here, I use passive reception mode. Why does the last packet of data wait for two minutes and then report it bit by bit?

doudouladou, 2022-07-06:


doudouladou, 2022-07-06:
Environment:
hardware：esp8266
software VERSION：AT   2.2.1(WROOM-02)

Problem Description:
I am using at+cipstart make a https request.
When i use Active reporting mode，everything is normal.
When switching to passive receiving mode, some data will be received first, and then it will be stuck for two minutes before the remaining data will be reported.

doudouladou, 2022-07-08:
Ok, I tried the firmware compiled with the latest code and it works

doudouladou, 2022-07-08:
[image]

I have tried several times, but sometimes the problem still occurs

What kind of information do I need to provide here

ustccw, 2022-07-22:
@doudouladou please see https://github.com/espressif/esp-at/issues/598~