espressif/esp-at #653: ESP32-PICO-D4, v2.4.0.0, change uart pins, only garbage charactors displays on serial tool window, while v2.2.0.0-esp32 succeeds

picospuch, 2022-07-22:
Having read the doc carefully, changed pin configuration in the csv file, and did full recompilation the source of branch release/v2.4.0.0, I found idf-at seems setting wrong baud rate other than the default 115200.

When I revert to v2.2.0.0-esp32, everything works smoothly according to the documents.

ustccw, 2022-07-25:
@picospuch could you please provide more details, includes your .csv file, your 4MB bin, AT command port logs, AT log port logs.  
i would like to look into it if it can be reproduced.
~

picospuch, 2022-07-25:
- .csv
[factory_param_data.csv](https://github.com/espressif/esp-at/files/9180223/factory_param_data.csv)

- 4MB bin
[factory_PICO-D4.bin.gz](https://github.com/espressif/esp-at/files/9180239/factory_PICO-D4.bin.gz)

- AT log port logs
[minicom.zip](https://github.com/espressif/esp-at/files/9180312/minicom.zip)

- AT command port logs
nothing but 2 or 3 chars of garbage after hard-reset.

ustccw, 2022-07-26:
@picospuch Thanks for your information.

The reason is that GPIO32 (UART Tx you configurated) is used for external 32kHz crystal (RTC clock), for BLE light sleep function, so that it can be used for UART TX any more.

There are two workarounds:
1. Change UART TX from GPIO32 to other available GPIOs. 
2. Disable BLE light sleep on menuconfig
`./build.py menuconfig` -> `Component config` -> `ESP32-specific` -> `RTC clock source` -> Choose `Internal 150kHz RC oscillator`

Before v2.4.0.0, esp-at firmware does not support BLE light sleep function, so it works well.

picospuch, 2022-07-26:
> @picospuch Thanks for your information.
>
> The reason is that GPIO32 (UART Tx you configurated) is used for external 32kHz crystal (RTC clock), for BLE light sleep function, so that it can be used for [...]

@ustccw thanks for your quick response. You solved the issue! I'm switching to latest version now! btw, I will delete the logs I posted, because they contain some strings of my private information.

closed 2022-07-26 by picospuch

ustccw, 2022-07-27:
No worries.  
btw, i added a workaround for it internally, so that you do not need to do anything, just `git pull` esp-at repository.
this change should auto sync to GitHub in one week after CI passed.