espressif/esp-at #655: SSL TCP server with passive receive mode reports wrong length in +IPD

JAndrassy, 2022-07-23:
Passive receive mode and a client (web browser) of a secure server started with `AT+CIPSERVER=1,443,"SSL"`:
```
AT+GMR
AT version:2.5.0.0-dev(s-5d850d5 - ESP32C3 - Jul 15 2022 03:28:53)
SDK version:v4.3.2-553-g588436db47
compile time(d1ac5ae):Jul 23 2022 16:22:21
Bin version:2.4.0(MINI-1)

OK

AT+CIPRECVMODE=1

OK

AT+CIPSERVER=1,443,"SSL"

OK

0,CONNECT

+IPD,0,450

AT+CIPRECVDATA=0,450
+CIPRECVDATA:421,GET / HTTP/1.1
Host: 192.168.1.114
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:102.0) Gecko/20100101 Firefox/102.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
Upgrade-Insecure-Requests: 1
Sec-Fetch-Dest: document
Sec-Fetch-Mode: navigate
Sec-Fetch-Site: cross-site

OK

AT+CIPRECVDATA=0,29

ERROR

AT+CIPRECVLEN?
+CIPRECVLEN:0,-1,-1,-1,-1

OK
```
browser sent 421, but +IPD reported 450 so the AT+CIPRECVDATA to retrieve the expected 29 bytes more returned ERROR.

ustccw, 2022-07-25:
@JAndrassy Hi, here is one note about it. 
> For SSL connections, ESP-AT will return the length of the encrypted data, so the returned length will be larger than the real data length.

On MCU side, please call [AT+CIPRECVLEN](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#at-ciprecvlen-obtain-socket-data-length-in-passive-receiving-mode) to query if there are more data after `AT+CIPRECVDATA` command.
~

JAndrassy, 2022-07-25:
> On MCU side, please call [AT+CIPRECVLEN](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#at-ciprecvlen-obtain-socket-data-length-in-passive-receiving-mode) to query if there are more data after AT+CIPRECVDATA command.

yes I know. But similar problem was already fixed for outgoing TCP connection.

ustccw, 2022-07-25:
For SSL connections, ESP-AT will return **the length of the encrypted data**, so the returned length will be larger than the real data length.

JAndrassy, 2022-07-25:
> For SSL connections, ESP-AT will return **the length of the encrypted data**, so the returned length will be larger than the real data length.

I don't see that in 2.4 anymore. (I imagine Ivan fixed it)

ustccw, 2022-07-26:
`+IPD` and "+CIPRECVLEN" return same result, here is the [note](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#id138).

JAndrassy, 2022-07-26:
> `+IPD` and "+CIPRECVLEN" return same result, here is the [note](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#id138).

that is not true in 2.4.0

ustccw, 2022-07-27:
i am so sorry that i am confused about it :crying_cat_face: . could you please provide more details, how can it not work on v2.4.0.

JAndrassy, 2022-07-27:
> i am so sorry that i am confused about it crying_cat_face . could you please provide more details, how can it not work on v2.4.0.

2.4.0. reports the right size with +IPD in passive mode (not a larger size) for SSL connection started with CIPSTART

JAndrassy, 2022-07-27:
sorry. yes it reports larger size, but if the data are received with multiple AT+CIPRECVDATA, then it reports the right size after the first AT+CIPRECVDATA. and this applies for CIPSTARTED connection and for connection over a CIPSERVER.
it is acceptable behavior. sorry for bothering you.

closed 2022-07-27 by JAndrassy