espressif/esp-at #656: ESP32 WROOM Bluetooth Classic SPP Loses BTNAME when coming back from light sleep

hchang4, 2022-07-25:
To reproduce
disable, wifi, bluetooth, and bluetooth classic.
AT+CWMODE=0
AT+BLEINIT=0
AT+BTINIT=0
config light sleep to wake up with gpio 27 high 
AT+SLEEPWKCFG=2,27,1
externally drive gpio 27 low
go into light sleep mode.
AT+SLEEP=2
wake up by driving gpio 27 high
Check bluetooth name
AT+BTNAME?
bluetooth name will be blank.

Luxin812, 2022-07-29:
Because the BT name will be cleared after BT deinit, you can reset the BT name after each BT initialization.

hchang4, 2022-07-29:
If this is the correct  operation why is BTNAME retained during deep sleep?
Also I tested light sleep using arduino firmware and it retains BTNAME.

Luxin812, 2022-09-20:
> If this is the correct operation why is BTNAME retained during deep sleep? Also I tested light sleep using arduino firmware and it retains BTNAME.
> […](#)

Hi, 

I tried to reproduce the problem by release ESP32AT v2.4, but it didn't reproduce. Can you provide your AT version information (AT+GMR)? Or can you use the released version of ESP32AT v2.4 to test if your scenario reproduces this issue?
release ESP32AT v2.4: https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Binary_Lists/ESP32_AT_binaries.html

```
[2022-09-20_10:41:05:557]AT+GMR
[2022-09-20_10:41:05:557]AT version:2.4.0.0(s-4c6eb5e - ESP32 - May 20 2022 03:12:58)
[2022-09-20_10:41:05:557]SDK version:qa-test-v4.3.3-20220423
[2022-09-20_10:41:05:557]compile time(5641e0a):May 20 2022 11:11:43
[2022-09-20_10:41:05:557]Bin version:2.4.0(WROVER-32)

[2022-09-20_10:41:05:557]OK
[2022-09-20_10:41:09:634]AT+bTINIT?
[2022-09-20_10:41:09:634]+BTINIT:0

[2022-09-20_10:41:09:634]OK
[2022-09-20_10:41:25:965]AT+BTNAME?
[2022-09-20_10:41:25:965]+BTNAME:ESP32_AT

[2022-09-20_10:41:25:965]OK
[2022-09-20_10:41:56:414]AT+CWMODE=0

[2022-09-20_10:41:56:414]OK
[2022-09-20_10:42:02:227]AT+SLEEPWKCFG=2,27,1

[2022-09-20_10:42:02:227]OK
[2022-09-20_10:42:21:841]at+sleep=2

[2022-09-20_10:42:21:841]OK
[2022-09-20_10:42:39:190]AT+BTNAME?
[2022-09-20_10:42:39:190]+BTNAME:ESP32_AT

[2022-09-20_10:42:39:190]OK
[2022-09-20_10:42:40:296]AT+BTNAME?
[2022-09-20_10:42:40:296]+BTNAME:ESP32_AT

[2022-09-20_10:42:40:296]OK
[2022-09-20_10:42:43:691]AT+bTINIT?
[2022-09-20_10:42:43:691]+BTINIT:0

[2022-09-20_10:42:43:691]OK
[2022-09-20_10:42:45:860]AT+BTNAME?
[2022-09-20_10:42:45:860]+BTNAME:ESP32_AT

[2022-09-20_10:42:45:860]OK
```

hchang4, 2022-09-20:
ESP32_AT is the default bluetooth name. Try changing the bluetooth name to
something other than the default before going to sleep.

Luxin812, 2022-11-15:
> ESP32_AT is the default bluetooth name. Try changing the bluetooth name to something other than the default before going to sleep.
> […](#)

Hi，The issue is not that the bt name will be cleared due to entering light sleep. The actual reason is that the code of the AT+BTINIT=0 command contains logic to clear the bt name parameter. This behavior is reasonable at the application level. Because bt is disabled, some parameters related to bt are cleared

closed 2023-06-05 by ustccw