espressif/esp-at #657: issue compiling AT-firmware

DavidMR91, 2022-08-03:
When I try to make the step "python3 build.py build" the process ends with an error:

[1233/1248] Building C object esp-idf/at/CMakeFiles/__idf_at.dir/src/at_ota_cmd.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
A fatal error occurred: idf.py build failed

This is all the shell:
`
david@ubuntu:~/esp/esp-at$ python3 build.py build
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
3. PLATFORM_ESP32C2
choose(range[1,3]):2

Module name:
1. MINI-1 (description: TX:7 RX:6)
2. ESP32C3-SPI
3. ESP32C3_QCLOUD (description: QCLOUD TX:7 RX:6)
4. ESP32C3_12F (description: TX:22 RX:21)
choose(range[1,4]):4

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):1
Platform name:ESP32C3	Module name:ESP32C3_12F	Silence:1
submodules check completed for updates.
Ready to set up environment variables..
PATH is /home/david/esp/esp-at/esp-idf/components/esptool_py/esptool:/home/david/esp/esp-at/esp-idf/components/espcoredump:/home/david/esp/esp-at/esp-idf/components/partition_table:/home/david/esp/esp-at/esp-idf/components/app_update:/home/david/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/david/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/david/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/david/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/david/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/david/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/david/.espressif/python_env/idf4.3_py3.6_env/bin:/home/david/esp/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /home/david/.espressif/python_env/idf4.3_py3.6_env
sys.platform is linux
export str is 
PATH is /home/david/esp/esp-at/esp-idf/components/esptool_py/esptool:/home/david/esp/esp-at/esp-idf/components/espcoredump:/home/david/esp/esp-at/esp-idf/components/partition_table:/home/david/esp/esp-at/esp-idf/components/app_update:/home/david/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/david/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/david/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/david/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/david/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/david/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/david/.espressif/python_env/idf4.3_py3.6_env/bin:/home/david/esp/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /home/david/.espressif/python_env/idf4.3_py3.6_env
Executing action: all (aliases: build)
Running cmake in directory /home/david/esp/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=0 /home/david/esp/esp-at"...
silence:1 sdkconfig:/home/david/esp/esp-at/module_config/module_esp32c3_default/sdkconfig_silence.defaults
esp-at branch: master
esp-at docs version: latest
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++
-- Check for working CXX compiler: /home/david/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /home/david/esp/esp-at/sdkconfig
Loading defaults file /home/david/esp/esp-at/module_config/module_esp32c3_default/sdkconfig_silence.defaults...
-- Found PythonInterp: /home/david/.espressif/python_env/idf4.3_py3.6_env/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp-at" version: v2.4.0.0-57-g9e71ee6-dirty
-- Adding linker script /home/david/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/david/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/david/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/david/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
[... 164 lines cut ...]
../components/at/src/at_default_config.c:72:24: error: 'CONFIG_ESP_AT_OTA_TOKEN_ESP32C3' undeclared here (not in a function); did you mean 'CONFIG_ESP_AT_OTA_TOKEN_ESP32C3_12F'?
     {"ESP32C3-12F",    CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F,         CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3-12F },
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CONFIG_ESP_AT_OTA_TOKEN_ESP32C3_12F
../components/at/src/at_default_config.c:72:56: error: invalid suffix "F" on integer constant
     {"ESP32C3-12F",    CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F,         CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3-12F },
                                                        ^~~
../components/at/src/at_default_config.c:72:69: error: 'CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3' undeclared here (not in a function); did you mean 'CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3_12F'?
     {"ESP32C3-12F",    CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F,         CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3-12F },
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                     CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3_12F
../components/at/src/at_default_config.c:72:105: error: invalid suffix "F" on integer constant
     {"ESP32C3-12F",    CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F,         CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3-12F },
                                                                                                         ^~~
../components/at/src/at_default_config.c:72:5: warning: missing initializer for field 'ota_token' of 'esp_at_module_info_t' {aka 'const struct <anonymous>'} [-Wmissing-field-initializers]
     {"ESP32C3-12F",    CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F,         CONFIG_ESP_AT_OTA_SSL_TOKEN_ESP32C3-12F },
     ^
../components/at/src/at_default_config.c:52:11: note: 'ota_token' declared here
     char* ota_token;
           ^~~~~~~~~
[1233/1248] Building C object esp-idf/at/CMakeFiles/__idf_at.dir/src/at_ota_cmd.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
A fatal error occurred: idf.py build failed
`

ustccw, 2022-08-04:
@DavidMR91 it looks like you are adding a new module support. You can refer to [How to Add Support for a Module](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_add_support_for_a_module.html#how-to-add-support-for-a-module) for more details.

btw, AFAIK, variable names in C language are made up of letters (upper and lower case) and digits. The underscore character ("_") is also permitted. Names must not begin with a digit.

so It is not allowed that a variable includes **`-`** character in your case. could you try `CONFIG_ESP_AT_OTA_TOKEN_ESP32C3_12F` instead of `CONFIG_ESP_AT_OTA_TOKEN_ESP32C3-12F`.

DavidMR91, 2022-08-04:
Thank you very much for the reply. Changing "-" to "_" has managed to finish the compilation.

closed 2022-08-04 by DavidMR91