espressif/esp-at #662: ESP32-C3 SPI Slow transfer

AlexAri1, 2022-08-25:
Hi, 
I'm using the est-at (branch release/2.4.0.0 | hash: f26b306d58d4ec31261e39680fc3bafef997d292) on an ESP32-C3 (custom board).
I'm connected to an iPad on a **local network** (with another esp32 as softAP).

I followed the "How to Improve ESP-AT Throughput Performance" (https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Compile_and_Develop/How_to_optimize_throughput.html#how-to-improve-esp-at-throughput-performance)

I connect to the ipad with: 
- AT+CIPSTART="TCP,"<ip>",<port>

Setup: 
- AT+CIPSENDLCFG=9216,2920
- AT+CIPSENDL=1032288

Then I send 3kB packets, and every ~3 packets I wait for "+CIPSENDL" to be sure there were no data loss.

**What I am experiencing**:
The first 70kB are quick enough, but then it gets slower and slower, sometimes the +CIPSENDL takes a few seconds (sometimes more than 5). It takes around 30 to 40 seconds to send 1MB.
I attached the logs with the at commands I send with timestamps, and the sdkconfig I use.

Is there anything I'm missing ?
Thanks
[YAT-Log-20220825-160703.log](https://github.com/espressif/esp-at/files/9425542/YAT-Log-20220825-160703.log)
[sdkconfig.txt](https://github.com/espressif/esp-at/files/9425544/sdkconfig.txt)

AlexAri1, 2022-08-25:
The RF shield was faulty on the device, looks OK on another one.

closed 2022-08-25 by AlexAri1

ustccw, 2022-09-05:
@AlexAri1 Could you please take a look at this one: https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Compile_and_Develop/How_to_implement_SPI_AT.html

ustccw, 2022-09-05:
one more possible useful doc here: https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_optimize_throughput.html

if this is still not that effective, maybe @ESP-Coco could give more suggestion.

ustccw, 2022-09-05:
@ESP-Coco could you please take a look.

Jacques-zhao, 2022-09-06:
@AlexAri1 The rate of 0.8Mbps is too low, we can test rates up to 10Mbps (https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Compile_and_Develop/How_to_implement_SPI_AT.html#reference-results). This rate requires you to make sure that you perform as follows:

1. Using  [Passthrough Mode](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/index_of_abbreviations.html#term-Passthrough-Mode)
2. Discard the UART to type commands, and automatically input the command directly into the code, because UART is very slow, which will slow down the SPI transmission rate
3. [Configure system, LWIP, Wi-Fi parameters](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Compile_and_Develop/How_to_optimize_throughput.html#simple-quick-configuration)
4. Test in a shielded box, or find a space with low WiFi interference

Jacques-zhao, 2022-09-07:
@AlexAri1  I think I have found the reason for your slow transmission speed, and the stuck problem needs to be solved. The design strategy of SPI AT is that the slave decides whether to receive or send, and the master does not have the right to transmit actively. I think the reason for the stuck is that this strategy is not strictly followed. 
We provide a reference [SPI master example](https://github.com/espressif/esp-at/tree/master/examples/at_spi_master/spi/esp32_c_series) using  ESP32-C3 , and I suggest you modify this example to test the rate

Jacques-zhao, 2022-09-08:
Hi, @AlexAri1 , your understanding is correct.
But your screenshot is too fuzzy, could you please send the logic analyzer to me as an attachment?

Jacques-zhao, 2022-09-09:
Hi, @AlexAri1 , The logic analyzer you sent is completely normal , I suspect there is a problem with the command AT +CIPSENDL, I am on vacation now, I will repeat this problem after the end of the vacation. Maybe you also can try sending data in passthrough mode instead of using this command

Jacques-zhao, 2022-10-21:
Hi @AlexAri1  I guess this is because the at_spi_slave_task has not been executed. The priority of at_spi_slave_task is 10, and the priority of LWIP task and WiFi task is much higher than this. The more TCP data, the later the at_spi_slave_task will be executed