espressif/esp-at #667: Compiling issue ESP-AT Windows

Polyteres, 2022-09-16:
Hi everyone. I have a problem compiling the ESP-AT project. I am following the steps from here https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp-at-get-at-sdk for Windows. In the step 3 "python build.py install", I receive this error:

Ready to set up ESP-IDF tools..
ERROR: This script was called from a virtual environment, can not create a virtual environment again
A fatal error occurred: set up ESP-IDF python-env failed

I do not undestand why. Please I need help.

Best regards.

ustccw, 2022-09-22:
@Polyteres Could you please provide detailed logs after you cloned ESP-AT.

BPaillard, 2022-10-01:
Yes, I have the same problem. But it looks like the subsequent steps are OK, and the firmware seems to be compiling OK.