espressif/esp-at #668: v2.2.0.0_esp8266 无法启动

darkgirl, 2022-09-18:
按照官网教程编译 ESP8266_1M AT 固件, 烧写后 log 只输出一部分就挂了

`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7496, room 16 
tail 8
chksum 0xc6
load 0x3ffe8008, len 24, room 0 
tail 8
chksum 0x7f
load 0x3ffe8020, len 3340, room 0 
tail 12
chksum 0x0d
csum 0x0d

`

ustccw, 2022-09-22:
Log output should come from **GPIO2**, not the download port.  
Could you please double check your [hardware connection](https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Get_Started/Hardware_connection.html#esp8266-series).

techrah, 2023-02-05:
I have the same issue. I downloaded `esp8285-1MB-at` from [here](https://github.com/espressif/esp-at/actions/runs/3592445311#artifacts). I also tried building from source (both 2.2.0 and 2.1.0) with the same result.

The log output from GPIO2 starts at baud rate 74880 and shows this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7464, room 16

tail 8

chksum 0x60

load 0x3ffe8008, len 24, room 0

tail 8

chksum 0xd9

load 0x3ffe8020, len 3340, room 0

tail 12

chksum 0xae

csum 0xae
```

Then, if I switch to baud rate 115200, I get this:
```
(45) boot: ESP-IDF v3.4-63-ge3348ac7 2nd stage bootloader
                                                                            I (45) boot: compile time 14:36:34
                                                                                                              I (45) boot: SPI Speed      : 26.7MHz
   I (49) boot: SPI Mode       : DOUT
                                     I (53) boot: SPI Flash Size : 1MB
                                                                      I (57) boot: Partition Table:
                                                                                                   I (61) boot: ## Label            Usage          Type ST Offset   Length
                          I (68) boot:  0 otadata          OTA data         01 00 00009000 00002000
                                                                                                   I (75) boot:  1 nvs              WiFi data        01 02 00010000 00008000
                            I (83) boot:  2 at_customize     unknown          40 00 00018000 00008000
                                                                                                     I (90) boot:  3 factory          factory app      00 00 00020000 000e0000
                              I (97) boot: End of partition table
                                                                 I (101) boot: Defaulting to factory image
                                                                                                          I (106) esp_image: segment 0: paddr=0x00020010 vaddr=0x40220010 size=0xb34cc (734412) map
                                                   I (115) esp_image: segment 1: paddr=0x000d34e4 vaddr=0x402d34dc size=0x1bec8 (114376) map
                                                                                                                                            I (123) esp_image: segment 2: paddr=0x000ef3b4 vaddr=0x3ffe8000 size=0x00768 (  1896) load
                                                                                      I (132) esp_image: segment 3: paddr=0x000efb24 vaddr=0x40100000 size=0x00080 (   128) load
                                I (141) esp_image: segment 4: paddr=0x000efbac vaddr=0x40100080 size=0x059f0 ( 23024) load
                                                                                                                          I (150) boot: Loaded app from partition at offset 0x20000
                                   phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new

module_name:ESP8266_1MB

max tx power=78,ret=0

2.2.1
```

which, if reformatted, looks like this:
```
(45) boot: ESP-IDF v3.4-63-ge3348ac7 2nd stage bootloader
I (45) boot: compile time 14:36:34
I (45) boot: SPI Speed      : 26.7MHz
I (49) boot: SPI Mode       : DOUT
I (53) boot: SPI Flash Size : 1MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (75) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (83) boot:  2 at_customize     unknown          40 00 00018000 00008000
I (90) boot:  3 factory          factory app      00 00 00020000 000e0000
I (97) boot: End of partition table
I (101) boot: Defaulting to factory image
I (106) esp_image: segment 0: paddr=0x00020010 vaddr=0x40220010 size=0xb34cc (734412) map
I (115) esp_image: segment 1: paddr=0x000d34e4 vaddr=0x402d34dc size=0x1bec8 (114376) map
I (123) esp_image: segment 2: paddr=0x000ef3b4 vaddr=0x3ffe8000 size=0x00768 (  1896) load
I (132) esp_image: segment 3: paddr=0x000efb24 vaddr=0x40100000 size=0x00080 (   128) load
I (141) esp_image: segment 4: paddr=0x000efbac vaddr=0x40100080 size=0x059f0 ( 23024) load
I (150) boot: Loaded app from partition at offset 0x20000
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new

module_name:ESP8266_1MB

max tx power=78,ret=0

2.2.1
```

Meanwhile, from the UART0_TXD pin (GPIO1), at baud rate 74880, I see this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7464, room 16

tail 8

chksum 0x60

load 0x3ffe8008, len 24, room 0

tail 8

chksum 0xd9

load 0x3ffe8020, len 3340, room 0

tail 12

chksum 0xae

csum 0xae
```

It seems that the main app is not being executed. From what I understand, it should switch to baud rate 115200 and accept AT commands but it doesn't echo anything back to my terminal.

My device is an ESP-01M, ESP8285 chip, 1MB Flash.

Any advice is appreciated. Thank you in advance. 🙏

techrah, 2023-02-06:
Thanks. After carefully reading how the UART0 TX/RX-CTS/RTS swap works, I got it to work. I'll post my two solutions here in case someone else has the same issue. For solution 2, I have one more question.

### Solution 1

When connected like this:

UART0|USB-to-Serial Converter
-|-
GPIO13 RX|TX
GPIO15 TX|RX

the key to getting this to work is that GPIO15 needs to be pulled LOW during boot-up.

### Solution 2

I also got it to work without swapping the pins:

UART0|USB-to-Serial Converter
-|-
GPIO1 TX|RX
GPIO3 RX|TX

by modifying `components/customized_partitions/raw_data/factory_param/factory_param_data.csv`, changing this line

    PLATFORM_ESP8266,ESP8266_1MB,No OTA,0xfcfc,3,0,78,0,1,13,CN,115200,15,13,3,1,5,-1

to

    PLATFORM_ESP8266,ESP8266_1MB,No OTA,0xfcfc,3,0,78,0,1,13,CN,115200,1,3,3,1,5,-1

**QUESTION:** Is there a way to do this without changing this file? The only thing I could find in menuconfig that seems to be related to this is `CONFIG_ESP_UART0_SWAP_IO` but that is currently NOT set anyway.

Thanks!

closed 2023-06-05 by ustccw