espressif/esp-at #670: HTTPS Problem

charon22, 2022-09-27:
## Environment
Development Kit: ESP32-LyraT V4.3
Module: ESP32-WROVER-E
IDF version: v4.4
ADF version: v2.4

## Problem
We must use ESP-IDF 4.4 to do our current project. 
When we use ESP-AT's HTTP command, we mainly use HTTPS, and each time we use it, the cpu halt phenomenon occurs. 
When I checked this, I found out that the esp_http_client_config_t structure of esp_http_client.h of ESP-IDFV4.4 is a problem caused by the different structures of ESP-IDFV4.3. 
I would like to know if it is possible to support IDF4.4 structures in ESP-AT.

[image]

Thanks

xcguang, 2022-09-28:
Hello,
Sorry, we have no plan to use IDF v4.4 on AT project, and after released C2 AT, we will use IDF release/v5.0 for ESP32 and ESP32C3 AT.

BTW, ADF is https://github.com/espressif/esp-adf, is not AT.

xcguang, 2022-09-29:
emm, ESP32 AT may work on IDF v5.0 in October or in November, and if everything goes well, maybe released in December.

What's your project schedule, if I may ask?

charon22, 2022-09-30:
We are currently testing the feature implementation. The PCB will be manufactured next month, and production will probably start early next year.

charon22, 2022-10-06:
Can you close this issue

closed 2022-10-06 by charon22