espressif/esp-at #671: AT+CIPSEND The transparent transmission interval is 20ms

funful, 2022-10-10:
How do I change the 20ms interval in AT firmware Transparent mode？

ustccw, 2022-10-17:
it can't be changed till now. May i ask why you need to change this value?

funful, 2022-10-27:
It doesn't work when I'm passing through at 120HZ at high speed

LimaBR, 2023-03-15:
Hi. Any updates? Is it defined in the closed source libraries?

2fdesign, 2023-03-16:
It's probably ties to BLE spec of 20bytes per 20ms.  Even though CIPSEND can be used in SPP, I wouldn't be surprised if it is still ticked off 20ms.  How big is the data packet?  Maybe you can increase the packet size to increase throughput?

ustccw, 2023-03-16:
has added an AT command: `AT+TRANSINTV=<ms>` to support configurable interval in passthrough mode.  
Wait the internal CI passed and then push to GitHub.

closed 2023-03-17 by igrr