espressif/esp-at #672: ESP32-C3 SPI Read Problem

andrew7m, 2022-10-12:
I'm seeing a problem where I get a handshake, read the status and it's telling me that there is almost 4KB of data to read from the esp32.

This is unexpected because what I was expecting to read is very short, like an OK or SEND OK. And the contents of what I read looks like its just history of what's in a circular buffer on the esp32.

I'm using passive receive mode and never ask for more than 512 bytes at a time, however the server I'm connecting to never sends anywhere near that amount of data to me anyway.

I am establishing a TCP connection with a server and sending chucks of data from a file and it responds with short replies.

Attached in the zip are two examples of some transactions that end in my application reading nonsense from the esp32. Eventually my application crashes before it can even read everything.

This problem is repeatable. After a power cycle, this will always happen within a minute or two.

My setup of the esp32 is (not shown in the saleae logic analyzer traces):

ATE0
AT+CWMODE=0,0
AT+SYSSTORE=0
AT+CIPMUX=1
AT+CIPRECVMODE=1
AT+CWAPPROTO=3
AT+CWSTAPROTO=3

Followed by switching to station mode, scanning, connecting to an access point to which I get the WIFI CONNECTED and WIFI GOT IP messages and then I connect to the server and get the 0,CONNECT message and then start sending and receiving.

Version info is
AT version:2.5.0.0-dev(7060263 - ESP32C3 - Sep  7 2022 07:02:12)
SDK version:v4.3.2-553-g588436db47
compile time(dc7abb3):Sep 12 2022 11:41:43
Bin version:2.4.0(ESP32C3-SPI)
[Sessions.zip](https://github.com/espressif/esp-at/files/9760529/Sessions.zip)

ustccw, 2022-10-17:
@ESP-Coco could you please help to take a look~

Jacques-zhao, 2022-10-17:
Hi @andrew7m   I'm sorry that I didn't find the problem in your logic analyzer data. As for the problem you mentioned, I think we need to do the following tests.
1. To determine if this data was sent by AT core (Print in [here](https://github.com/espressif/esp-at/blob/master/main/interface/hspi/at_hspi_task_esp32_series.c#L184)), you can change the LOG output(Component config → Log output → Default log verbosity ) from Error to Info
2. To determine if the data length  was transmitted by ESP32, you can add log [here](https://github.com/espressif/esp-at/blob/master/main/interface/hspi/at_hspi_task_esp32_series.c#L286) and print the value of send_len

After adding the above two logs, recompile the AT firmware and test it, and send me the logs and logic analyzer data, please help to mark the time of the error found in the logic analyzer data, and I will analyze it together

andrew7m, 2022-10-18:
Hi @Jacques-zhao thanks for the diagnostic advice.

I followed it and have attached some results in the zip file.

The problem manifests differently now. Instead of there being an unexpected large number of bytes to read, bytes are now going missing.

In the attached zip file is the esp32 log file and the logic analyzer trace.

It shows what should be a "+IPD,0,10\r\n" coming through as just "10\r\n".

The relevant portion of the log file is this, from line 16710:

```
I (43238) HSPI-AT: Write data len: 13
I (43238) HSPI-AT: Write data: 

+IPD,0,10

I (43239) HSPI-AT: Write data len: 9
I (43239) HSPI-AT: Write data: 

OK

>
I (43240) HSPI-AT: send_len: 9
I (43242) HSPI-AT: send_len: 4

```
So the AT core writes the IPD command followed by a data prompt in response to a previous CIPSEND.

I would have thought the last two lines should have read
send_len: 13
send_len: 9
with the IPD coming out first and the data prompt second.

The corresponding section in the logic analyzer trace is marked between the two markers.

The 9 byte data prompt comes out first (after the first marker) and the chopped 4-byte IPD comes out just before the second marker. In between is a data packet, but that isn't always the case. I've seen the problem happen without a data packet in the middle.

Because my application doesn't get the IPD, it doesn't ask for any data (its in passive mode) and keeps sending data to the server periodically wondering why its not responding.

Again, the problem is very repeatable. So any further recommendations about how to diagnose this further would be appreciated.

I'll also note that I'm logging via the USB port, not uart.
[Session 3.zip](https://github.com/espressif/esp-at/files/9809613/Session.3.zip)

Jacques-zhao, 2022-10-19:
@andrew7m This problem is very strange, if AT core sends data of length 13 and 9 to the stream buffer, then I should not have inconsistent lengths when reading from the stream buffer.
I have analyzed this part of the code and I suspect that the problem is caused by multiple task calling the write_data function at the same time, because the steam buffer is not thread-safe, resulting in abnormal write data.
This is easily verified by placing the spi_mutex_lock function above the xStreamBufferSend .

Just like this.

    /* Result of AT command, auto call when read_data get data */
    static int32_t at_spi_write_data(uint8_t* buf, int32_t len)
    {
    ......
    
        ESP_LOGI(TAG, "Write data len: %d", len);
        spi_mutex_lock();
        
        length = xStreamBufferSend(spi_slave_tx_ring_buf, buf, len, 0);
    
        if(length != len) {
            ESP_LOGE(TAG, "Stream buffer send error");
            return -1;
        }
    
        if (initiative_send_flag == 0) {
            initiative_send_flag = 1;
            spi_msg_t spi_msg = {
                .direct = SPI_SLAVE_WR,
            };
    
            if (xQueueSend(msg_queue, (void*)&spi_msg, 0) != pdPASS) {
                ESP_LOGE(TAG, "send WR queue for spi_write error");
            }
        }
        spi_mutex_unlock();
        return len;
    }

In fact, in real use, this problem should not exist, just because we added some print, resulting in print +IPD tasks and print OK tasks accessing the steam buffer at the same time

andrew7m, 2022-10-20:
@Jacques-zhao, this has fixed the problem. Moving the mutex lock to include the xStreamBufferSend() call, I get no more reads of chopped data, nor 4KB of buffer history.
I got a fresh copy of the AT project and made just that one change and it has been running for hours without issue, whereas it used to run for only a minute or so.
You can close this issue unless you have doubts and need more tests done.

Note that you might also add a mutex unlock call right before the "return -1;" where the return value of xStreamBufferSend() is checked, so the mutex doesn't remain locked if there's a send error.

Thanks for your help.

closed 2023-06-05 by ustccw