espressif/esp-at #674: 编译出现错误

Nuralimnur, 2022-10-18:
我按照官网进行编译固件的时候出现了错误
两个版本都试了，出现了同样的错误
A fatal error occurred: list index out of range

[image]

[image]

Nuralimnur, 2022-10-19:
could you please help to take a look~

ustccw, 2022-10-19:
@Nuralimnur seems you'are using a wrong IDF version (idf 4.4).
coud you please try to run the following commands and provide detailed compile logs:
```
rm -rf esp-idf build sdkconfig
./build.py install
./build.py build
```
and please read the https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html first.

Nuralimnur, 2022-10-19:
@ustccw 看一下呢，还是出现了同样的错误，并且往下运行 `./build menuconfig` 也会出现同样的错误
``` zsh
nur@nur-virtual-machine  ~/esp32/Projects/esp-at-2.4.2.0   master  rm -rf esp-idf build sdkconfig
 nur@nur-virtual-machine  ~/esp32/Projects/esp-at-2.4.2.0   master  ./build.py install        
Ready to install ESP-IDF prerequisites..
[sudo] nur 的密码： 
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
bison 已经是最新版 (2:3.5.1+dfsg-1)。
ccache 已经是最新版 (3.7.7-1)。
cmake 已经是最新版 (3.16.3-1ubuntu1)。
flex 已经是最新版 (2.6.4-6.2)。
libffi-dev 已经是最新版 (3.3-4)。
libusb-1.0-0 已经是最新版 (2:1.0.23-2build1)。
python3 已经是最新版 (3.8.2-0ubuntu2)。
python3-setuptools 已经是最新版 (45.2.0-1)。
dfu-util 已经是最新版 (0.9-1)。
gperf 已经是最新版 (3.1-1build1)。
ninja-build 已经是最新版 (1.10.0-1build1)。
git 已经是最新版 (1:2.25.1-1ubuntu3.6)。
libssl-dev 已经是最新版 (1.1.1f-1ubuntu2.16)。
wget 已经是最新版 (1.20.3-1ubuntu2)。
python3-pip 已经是最新版 (20.0.2-5ubuntu1.6)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 4 个软件包未被升级。
Ready to install ESP-AT prerequisites..
Requirement already satisfied: setuptools>=21 in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r requirements.txt (line 4)) (63.4.2)
Requirement already satisfied: cryptography<35,>=2.1.4 in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r requirements.txt (line 9)) (3.4.8)
Requirement already satisfied: pyyaml in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r requirements.txt (line 10)) (6.0)
Requirement already satisfied: xlrd in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r requirements.txt (line 11)) (2.0.1)
Requirement already satisfied: cffi>=1.12 in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cryptography<35,>=2.1.4->-r requirements.txt (line 9)) (1.15.1)
Requirement already satisfied: pycparser in /home/nur/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cffi>=1.12->cryptography<35,>=2.1.4->-r requirements.txt (line 9)) (2.21)
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
choose(range[1,2]):2

Module name:
1. MINI-1 (description: TX:7 RX:6)
2. ESP32C3-SPI
3. ESP32C3_QCLOUD (description: QCLOUD TX:7 RX:6)
choose(range[1,3]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):1
Platform name:ESP32C3   Module name:MINI-1      Silence:1
A fatal error occurred: list index out of range
```

ustccw, 2022-10-19:
1. 可以在 build.py 添加一点打印信息，看看是哪一行导致这样的错误。
2. 编译 AT 时候，不应该出现 `/home/nur/.espressif/python_env/idf4.4_py3.8_env`，可以排查下是不是哪里误设置了。

Nuralimnur, 2023-03-07:
这个问题解决了，我当时下载了release里source code压缩包导致的，我后来git clone后没问题了

closed 2023-03-07 by Nuralimnur