espressif/esp-at #675: MCU给ESP32发送的AT+CIPSEND命令，发送数据一部分没有被成功接收

SSSANTON, 2022-10-21:
使用官方的v2.4固件版本, 使用串口发送AT命令，波特率1000000bps，收发数据如下：【】内为时间戳，单位ms
1.【291 MCU->ESP32】AT+CIPSEND=0,274
2.【304 ESP32->MCU】OK
3.【314-353 MCU->EPS32】发送274字节数据
4.【771 MCU->ESP32】AT+CIPSEND=0,274
5.【4527 ESP32->MCU】+IPD,0,17 接收到17字节数据
6.【4538 MCU->ESP32】发送274字节数据
7.【4587 ESP32->MCU】返回busy p...
8.【4596 ESP32->MCU】返回Recv 274 bytes
9.【4605 ESP32->MCU】返回SENDOK

现在从服务器端看到的情况是，最后收到的数据是3的前半部分+4的全部+6的前半部分组成的。数据是我额外找两个硬件串口调试器接（RX分别接到ESP32和MCU通信的RX和TX线上）电脑打印观察的，代表是实际的收发数据。
ESP32的AT+CIPSEND命令发送后，应该按指定长度接收后面的数据，然后就发出吧？现在为何在第4步发送完274字节后，没有把数据发送到服务器，而且后面又丢弃了后半部分而拼接了别的数据? 目前看起来就好像第3步由MCU发送的数据后半部分没有被ESP32成功接收和处理。

ustccw, 2022-10-21:
在 Recv xxx bytes 后就会发出去。第四步需要等待第三步返回 SEND OK。网络问题可以抓包确认。

SSSANTON, 2022-10-21:
> 在 Recv xxx bytes 后就会发出去。第四步需要等待第三步返回 SEND OK。网络问题可以抓包确认。

该问题即使是正常等待SEND OK也会出现，比如，即使等待20秒超时了，后面发送的命令和数据也会拼接到前面AT+SEND命令需要发送的数据里面。有用WIRESHARK抓包对比数据，不觉得是网络原因。

ustccw, 2022-10-21:
烦请提供下具体的命令交互日志，和网络抓包。

SSSANTON, 2022-10-22:
> 烦请提供下具体的命令交互日志，和网络抓包。
你好，以下是RT-thread的AT命令原始数据打印日志和WIRESHARK的网络抓包记录，请留意网络日志10-22 21:35:02的数据帧和和txt日志106行附近的AT命令收发记录。

[AT命令收发日志.txt](https://github.com/espressif/esp-at/files/9844500/AT.txt)
[ESP32 busy问题网络抓包记录.zip](https://github.com/espressif/esp-at/files/9844502/ESP32.busy.zip)

ustccw, 2022-10-24:
@SSSANTON 从日志和抓包看，可能是 UART 的问题。
1. 由于日志的第 19 行发送了 AT+CIPSEND=0,274 命令，在 24 - 32 行的日志显示发送了 274 字节，但实际上 ESP32 并没有完整接收到该字节，这样导致后面发送的命令会被当做数据，直到接收到 274 字节，再发送。
打印 `Recv 274 bytes` 表示 ESP-AT 从 UART 接收到 274 字节，所以 19 行的命令，实际到 109 行才实际接收到 274 字节。

2. 从抓包看，24-32 行丢到的数据为日志里第 27 行 `68 50 0C` 开始到 31 行的 `67 43 BE`，以及最后的 `00 00 01` 到 `00 31 DB`。

可能的建议：
1. 再检测下 MCU 测发送的逻辑，是否有可能会导致 send buffer 覆盖的问题
2. 用逻辑分析仪抓下 MCU 测 TX 的信号，确保数据被正确发出来了
3. 硬件检查：是否有可能 UART 波特率过高，供压不足，连接线过长等可能因素
4. 可以尝试用 UART 硬件流控，[AT+UART_CUR=115200,8,1,0,3](https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#id17)

如果仍有问题，请提供详细的日志，和逻辑分析仪数据。带时间戳。

SSSANTON, 2022-10-28:
> @SSSANTON 从日志和抓包看，可能是 UART 的问题。
>
> 1. 由于日志的第 19 行发送了 AT+CIPSEND=0,274 命令，在 24 - 32 行的日志显示发送了 274 字节，但实际上 ESP32 并没有完整接收到该字节，这样导致后面发送的命令会被当做数据，直到接收到 274 字节，再发送。
>    打印 `Recv 274 bytes` 表示 ESP-AT 从 UART [...]

感谢，问题已解决。
专门写了一个裸机程序测试，发现串口数据的确发生了部分丢失和拼接，比如主控MCU已经发出1000字节数据，但ESP32接收有时候会少接了最后200字节，然后又从MCU下次发送的数据（不等于200字节）中补全200字节来凑够，导致这时候会提示busy p...。
用了串口硬件流控后，即使波特率到了2Mbps，也不会出现这种情况了。

closed 2022-10-28 by SSSANTON