espressif/esp-at #677: ESP32-S2-WROOM

conotto, 2022-10-28:
Hi,
I am trying to use AT commands on ESP32-S2-WROOM with ESP32-S2-WROOM_AT_Bin_V2.1.0.0 firmware
After flashing, some of the commands are not working and result in ERROR

AT+GMR
AT version:2.1.0.0(0b76313 - ESP32S2 - Aug 20 2020 05:57:43)
SDK version:v4.2-dev-2044-gdd3c032
compile time(b5e1674):Aug 21 2020 04:59:57
Bin version:2.1.0(WROOM)

These are just some of the commands that are failing
AT+CMD?
AT+SYSTEMP?
AT+USERRAM?
AT+CWSTATE?

Can you please shed some light on this ?

Thanks

ustccw, 2022-10-29:
@conotto please see https://espressif-docs.readthedocs-hosted.com/projects/esp-at/zh_CN/release-v2.1.0.0_esp32s2/ for ESP32-S2 v2.1.0 version.
cause the commands you mentioned are not supported in ESP32-S2 v2.1.0.

i would like to suggest you use ESP32-C3 or ESP32-C2 series chip instead.
~

conotto, 2022-10-29:
Thanks for your reply, i was going by the instructions provided here

https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp32/AT_Command_Set/Basic_AT_Commands.html

I think i see where the confusion is, even though the 2.2 docs mention ESP32-S2 and full command set, latest binary version available for the S2 is 2.1.

I will try using the C3

Thanks again

closed 2023-06-05 by ustccw