espressif/esp-at #678: AT+RFPOWER command not working on ble server mode

aykt-ks, 2022-11-14:
I'm using V2.4.2.0 version at C3-DevkitM-1. I need specify distance limits for my project with different power levels. I'm only monitoring advertisement packets RSSI level. When i execute RFPOWER command with different values nothing change on NRF Connect scanner (on RSSI level). Then i started to try old version firmwares. On V2.3.0.0 version command executed succcessfuly. I didn't test levels on wifi.

ustccw, 2022-11-17:
could you please provide more details amap. such as `AT+GMR`, logs, the commands, test environment, test steps, etc.

aykt-ks, 2022-11-17:
I am using _ESP32-C3-DevKitM-1_ as hardware. Connected with USB to TTL converter to pins 6th and 7th. Observing RSSI levels via nRF Connect Application on Android 12.

Commands and responses when using **_ESP32-C3-MINI-1_AT_Bin_V2.3.0.0_**  firmware

`AT

OK
AT+GMR
AT version:2.3.0.0(e98993f - ESP32C3 - Dec 23 2021 09:03:35)
SDK version:v4.3.1-354-g98d34e5
compile time(9ff1373):Dec 24 2021 11:22:05
Bin version:2.3.0(MINI-1)

OK
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300,0x02
+BLEGATTSCHAR:"desc",1,1,1,0x2901
+BLEGATTSCHAR:"char",1,2,0xC301,0x02
+BLEGATTSCHAR:"desc",1,2,1,0x2901
+BLEGATTSCHAR:"char",1,3,0xC302,0x08
+BLEGATTSCHAR:"desc",1,3,1,0x2901
+BLEGATTSCHAR:"char",1,4,0xC303,0x04
+BLEGATTSCHAR:"desc",1,4,1,0x2901
+BLEGATTSCHAR:"char",1,5,0xC304,0x08
+BLEGATTSCHAR:"char",1,6,0xC305,0x10
+BLEGATTSCHAR:"desc",1,6,1,0x2902
+BLEGATTSCHAR:"char",1,7,0xC306,0x20
+BLEGATTSCHAR:"desc",1,7,1,0x2902
+BLEGATTSCHAR:"char",1,8,0xC307,0x02
+BLEGATTSCHAR:"desc",1,8,1,0x2901
+BLEGATTSCHAR:"char",2,1,0xC400,0x02
+BLEGATTSCHAR:"desc",2,1,1,0x2901
+BLEGATTSCHAR:"char",2,2,0xC401,0x02
+BLEGATTSCHAR:"desc",2,2,1,0x2901

OK
AT+BLEADVPARAM=400,800,0,0,7,0

OK
AT+BLEADVSTART

OK
AT+RFPOWER=78,0,0,0 **(Observed RSSI: (-78, -86dBm))**

OK
AT+RFPOWER=78,15,15,15 **(Observed RSSI: (-25, -35dBm))**

OK`

Commands and responses when using **_ESP32-C3-MINI-1_AT_Bin_V2.4.2.0_** firmware

`AT

OK
AT+GMR
AT version:2.4.2.0(fb731d2 - ESP32C3 - Aug 25 2022 03:47:31)
SDK version:qa-test-v4.3.3-20220423
compile time(67966ba):Aug 26 2022 07:46:13
Bin version:2.4.2(MINI-1)

OK
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300,0x02
+BLEGATTSCHAR:"desc",1,1,1,0x2901
+BLEGATTSCHAR:"char",1,2,0xC301,0x02
+BLEGATTSCHAR:"desc",1,2,1,0x2901
+BLEGATTSCHAR:"char",1,3,0xC302,0x08
+BLEGATTSCHAR:"desc",1,3,1,0x2901
+BLEGATTSCHAR:"char",1,4,0xC303,0x04
+BLEGATTSCHAR:"desc",1,4,1,0x2901
+BLEGATTSCHAR:"char",1,5,0xC304,0x08
+BLEGATTSCHAR:"char",1,6,0xC305,0x10
+BLEGATTSCHAR:"desc",1,6,1,0x2902
+BLEGATTSCHAR:"char",1,7,0xC306,0x20
+BLEGATTSCHAR:"desc",1,7,1,0x2902
+BLEGATTSCHAR:"char",1,8,0xC307,0x02
+BLEGATTSCHAR:"desc",1,8,1,0x2901
+BLEGATTSCHAR:"char",2,1,0xC400,0x02
+BLEGATTSCHAR:"desc",2,1,1,0x2901
+BLEGATTSCHAR:"char",2,2,0xC401,0x02
+BLEGATTSCHAR:"desc",2,2,1,0x2901

OK
AT+BLEADVPARAM=400,800,0,0,7,0

OK
AT+BLEADVSTART

OK
AT+RFPOWER=78,0,0,0 **(Observed RSSI: (-44, -52dBm))**

OK
AT+RFPOWER=78,15,15,15 **(Observed RSSI: (-44, -52dBm))**

OK`

ustccw, 2022-11-18:
@aykt-ks thanks for the information.

@Luxin812 could you please take a look.

aykt-ks, 2022-12-08:
@ustccw Is there any solution to solve this problem?

Luxin812, 2022-12-08:
Maybe you can use [rfpower.zip](https://github.com/espressif/esp-at/files/10183364/rfpower.zip) file to compile a temporary test firmware based on the master branch to test whether this function is normal. Thank you for finding this problem.

> @ustccw Is there any solution to solve this problem?

Maybe you can use [rfpower.zip](https://github.com/espressif/esp-at/files/10183364/rfpower.zip) file to compile a temporary test firmware based on the master branch to test whether this function is normal. Thank you for finding this problem.

aykt-ks, 2023-01-17:
Sorry for late response. It is fixed. Thank you.

closed 2023-01-17 by aykt-ks

Luxin812, 2023-05-09:
1. The fix for this issue has been merged into the latest master version, but there is no backport to v2.4.2.0.
2. At present, the latest version of the release firmware of C3 is v2.4.2, but C3 will have a new version release in the near future.

mah-eiSmart, 2023-05-09:
@Luxin812 thank you for that very good information. What new release will it be? Based on v3.0.0.0 or based on v2.4.2.0 (so 2.4.3.0 for example)? Is there a date estimate, as we want to release our product in next 4 weeks.

mah-eiSmart, 2023-05-24:
> @Luxin812 thank you for that very good information. What new release will it be? Based on v3.0.0.0 or based on v2.4.2.0 (so 2.4.3.0 for example)? Is there a date estimate? We want to release our product in next 4 weeks.

@Luxin812 Do you have an answer on the questions above?

Luxin812, 2023-05-24:
> > @Luxin812 thank you for that very good information. What new release will it be? Based on v3.0.0.0 or based on v2.4.2.0 (so 2.4.3.0 for example)? Is there a date estimate? We want to release our product in next 4 weeks.
> 
> @Luxin812 Do you have an answer on the questions above?

The latest release prediction of ESP32C3 AT will be based on the master branch test and release after fix issues (the version number is expected to be v3.2.0.0), you can first test your product functions based on the master branch.

mah-eiSmart, 2023-06-05:
Hello @Luxin812 ,

there has been a release v3.1.0.0. For ESP32-C3 the recommended released version is still v2.4.2.0.

Will there be another release for ESP32-C3, or is v3.1.0.0 the release we've been talking about in the above conversation?

Thanks a lot for the support!