espressif/esp-at #679: 【TAG：V2.4.2 】SPI AT 方式AT+HTTPCPOST指令不能多次发送数据吗？

huangjinrui, 2022-11-16:
目前的现象是：若指定post 126字节数据，我测试多次发送的时候，第一次只发送了22字节，为何AT core 会自动补零，而不是等待数据接收完毕后再发起数据请求呢？
==================================================
--> AT+HTTPCPOST="http://120.234.211.xxx:11120/box/config/getBoxConfig",136,2,"connection: keep-alive","content-type: application/json"

[image]
[image]

Jacques-zhao, 2022-11-16:
Hi @huangjinrui 此次 SPI 发送已经指定了长度是136，那么必须要发送 136自己的数据，你可以认为写status和发送数据是一体的，不可以拆分，如果你只是想发送22字节，那么你需要将长度也指定为 22字节

Jacques-zhao, 2022-11-16:
不过我看了一下代码，这里我觉得很奇怪， “read len: 136”，代表SPI这边应该是收到了136字节，但是我实际拿数据时，却只拿到了22字节，所以才会打印那个错误。这部分你 MCU 的行为是什么，命令可以发出来给我们看看么

huangjinrui, 2022-11-16:
> 

您好，136字节只是我随便定义的数据长度而已，目的是为了测试AT+HTTPCPOST参考文档提到的可多次发送功能，实际项目的数据量可能会到达M量级，受限于系统资源只能多次发送。但在SPI AT测试过程中遇到了问题，UART AT使用是没有问题的，会按预期等待数据接收到指定的长度才发起数据请求。

huangjinrui, 2022-11-16:
> Hi @huangjinrui 此次 SPI 发送已经指定了长度是136，那么必须要发送 136自己的数据，你可以认为写status和发送数据是一体的，不可以拆分，如果你只是想发送22字节，那么你需要将长度也指定为 22字节

您好，136字节只是我随便定义的数据长度而已，目的是为了测试AT+HTTPCPOST参考文档提到的可多次发送功能，实际项目的数据量可能会到达M量级，受限于系统资源只能多次发送。但在SPI AT测试过程中遇到了问题，UART AT使用是没有问题的，会按预期等待数据接收到指定的长度才发起数据请求。

Jacques-zhao, 2022-11-16:
@huangjinrui 我觉得可能是这个原因，https://github.com/espressif/esp-at/blob/master/main/interface/hspi/at_hspi_task_esp32_series.c#L160 能否试一下把 return len; 改为 return ring_len; 然后再编译下载测试一下

huangjinrui, 2022-11-16:
> @huangjinrui 我觉得可能是这个原因，https://github.com/espressif/esp-at/blob/master/main/interface/hspi/at_hspi_task_esp32_series.c#L160 能否试一下把 return len; 改为 return ring_len; 然后再编译下载测试一下

@Jacques-zhao 谢谢您的解答，改为ring_len，问题得到解决了

closed 2022-11-16 by huangjinrui