espressif/esp-at #684: ESP32-C3 OTA gets stuck due to socket error

mah-eiSmart, 2022-11-28:
Hello,
I'm using ESP32-C3 mini-1 radio module and test and OTA (over the air update).

Therefore I'm setting up the device as access point and start the web server:
_AT+CWMODE=3,1
AT+CWSAP="my_ssid","",1,0
AT+CIPMUX=1
AT+WEBSERVER=1,80,25_

Then I connect to the radio module with a smart phone, open the web page and upload the new bin file, as described in
[https://docs.espressif.com/projects/esp-at/en/release-v2.4.0.0/esp32c3/AT_Command_Examples/Web_server_AT_Examples.html#ota-firmware-upgrade-using-a-browser](url)

The radio module returns:
_+STA_CONNECTED:"a8:9c:ed:be:76:ca"
+DIST_STA_IP:"a8:9c:ed:be:76:ca","192.168.4.2"
+WEBSERVERRSP:3_

What I see is that the OTA does not finish. The message _+WEBSERVERRSP:4_ does not show up.
When I check the debug log, I can see that at a certain time, there is a socket receive error, that is repeated forever.

_...
D (57238) httpd_txrx: httpd_req_recv: remaining length = 180576<27>[0m<\r><\n>
D (57243) httpd_txrx: httpd_req_recv: remaining length = 179296<27>[0m<\r><\n>
D (57247) httpd_txrx: httpd_req_recv: remaining length = 178816<27>[0m<\r><\n>
D (57256) httpd_txrx: httpd_req_recv: remaining length = 177536<27>[0m<\r><\n>
D (57262) httpd_txrx: httpd_req_recv: remaining length = 176256<27>[0m<\r><\n>
D (57264) httpd_txrx: httpd_req_recv: remaining length = 175936<27>[0m<\r><\n>
<27>[0;33mW (62268) httpd_txrx: httpd_sock_err: error in recv : 11<27>[0m<\r><\n>
D (62268) httpd_txrx: httpd_recv_with_opt: error in recv_fn<27>[0m<\r><\n>
D (62268) httpd_txrx: httpd_req_recv: error in httpd_recv<27>[0m<\r><\n>
D (62274) httpd_txrx: httpd_req_recv: remaining length = 175936<27>[0m<\r><\n>
<27>[0;33mW (67280) httpd_txrx: httpd_sock_err: error in recv : 11<27>[0m<\r><\n>
D (67280) httpd_txrx: httpd_recv_with_opt: error in recv_fn<27>[0m<\r><\n>
D (67280) httpd_txrx: httpd_req_recv: error in httpd_recv<27>[0m<\r><\n>
D (67286) httpd_txrx: httpd_req_recv: remaining length = 175936<27>[0m<\r><\n>
<27>[0;33mW (72292) httpd_txrx: httpd_sock_err: error in recv : 11<27>[0m<\r><\n>
D (72292) httpd_txrx: httpd_recv_with_opt: error in recv_fn<27>[0m<\r><\n>
D (72292) httpd_txrx: httpd_req_recv: error in httpd_recv<27>[0m<\r><\n>
D (72298) httpd_txrx: httpd_req_recv: remaining length = 175936<27>[0m<\r><\n>
..._
[full_log.txt](https://github.com/espressif/esp-at/files/10102959/full_log.txt)

How can we overcome this issue?

Thx in advance!

mah-eiSmart, 2022-11-28:
Update: When I shorten the length of each single transmission fragment (ESP_AT_WEB_SCRATCH_BUFSIZE) the transmission is running more stable. But in this case the watch dog resets the module, see the logs:

_D (111198) httpd_txrx: httpd_req_recv: remaining length = 126189<27>[0m<\r><\n>
D (111204) httpd_txrx: httpd_req_recv: remaining length = 126029<27>[0m<\r><\n>
D (111210) httpd_txrx: httpd_req_recv: remaining length = 125869<27>[0m<\r><\n>
D (111216) httpd_txrx: httpd_req_recv: remaining length = 125709<27>[0m<\r><\n>
D (111223) httpd_txrx: httpd_req_recv: remaining length = 125549<27>[0m<\r><\n>
D (111229) httpd_txrx: httpd_req_recv: remaining length = 125389<27>[0m<\r><\n>
D (111236) httpd_txrx: httpd_req_recv: remaining length = <27>[0;31mE (111239) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:<27>[0m<\r><\n>
<27>[0;31mE (111239) task_wdt:  - IDLE (CPU 0)<27>[0m<\r><\n>
<27>[0;31mE (111239) task_wdt: Tasks currently running:<27>[0m<\r><\n>
<27>[0;31mE (111239) task_wdt: CPU 0: esp_timer<27>[0m<\r><\n>
<27>[0;31mE (111239) task_wdt: Aborting.<27>[0m<\r><\n>
<\r><\n>
abort() was called at PC 0x420e2505 on core 0<\r><\n>
Core  0 register dump:<\r><\n>
MEPC    : 0x403806a8  RA      : 0x4038b752  SP      : 0x3fc97850  GP      : 0x3fc9388c  <\r><\n>
TP      : 0x3fc66a88  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  <\r><\n>
S0/FP   : 0x00000004  S1      : 0x3fc978b4  A0      : 0x3fc9787c  A1      : 0x3fc978b2  <\r><\n>
A2      : 0x00000000  A3      : 0x3fc978a9  A4      : 0x00000001  A5      : 0x3fca1000  <\r><\n>
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x0001b287  S3      : 0x00000001  <\r><\n>
S4      : 0x00000008  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  <\r><\n>
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  <\r><\n>
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  <\r><\n>
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  <\r><\n>
MHARTID : 0x00000000  <\r><\n>
<\r><\n>
Stack memory:<\r><\n>
3fc97850: 0x00000000 0x00000000 0x3fc978b0 0x4039218e 0x00000000 0x00000000 0x00000008 0x3fc96914<\r><\n>
3fc97870: 0x3fc978b4 0x3fc96930 0x3fc978b0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465<\r><\n>
3fc97890: 0x43502074 0x34783020 0x32653032 0x20353035 0x63206e6f 0x2065726f 0x00000030 0x420e0000<\r><\n>
3fc978b0: 0x3fce0030 0x65303234 0x35303532 0x00000000 0x600c2000 0x0001b287 0x00000001 0x420e2508<\r><\n>
3fc978d0: 0x00001881 0x80000007 0x3fc9e338 0x40392d84 0x60002000 0x3fcdfff4 0x00000000 0x00000000<\r><\n>
3fc978f0: 0x40380214 0x00000000 0x3fc9e338 0x403801f8 0x00000001 0x3fc97908 0xffffffff 0x3fca476c<\r><\n>
3fc97910: 0x3fca476c 0x00000000 0x3fc9791c 0xffffffff 0x3fc9791c 0x3fc9791c 0x00000000 0x3fc97930<\r><\n>
3fc97930: 0xffffffff 0x3fc97930 0x3fc97930 0x00000000 0x3fc97944 0xffffffff 0x3fc97944 0x3fc97944<\r><\n>
3fc97950: 0x00000000 0x3fc97958 0xffffffff 0x3fc97958 0x3fc97958 0x00000001 0x3fc9796c 0xffffffff<\r><\n>
3fc97970: 0x3fca3888 0x3fca3888 0x00000000 0x3fc97980 0xffffffff 0x3fc97980 0x3fc97980 0x00000000<\r><\n>
3fc97990: 0x3fc97994 0xffffffff 0x3fc97994 0x3fc97994 0x00000000 0x3fc979a8 0xffffffff 0x3fc979a8<\r><\n>
3fc979b0: 0x3fc979a8 0x00000000 0x3fc979bc 0xffffffff 0x3fc979bc 0x3fc979bc 0x00000000 0x3fc979d0<\r><\n>
3fc979d0: 0xffffffff 0x3fc979d0 0x3fc979d0 0x00000000 0x3fc979e4 0xffffffff 0x3fc979e4 0x3fc979e4<\r><\n>
3fc979f0: 0x00000000 0x3fc979f8 0xffffffff 0x3fc979f8 0x3fc979f8 0x00000000 0x3fc97a0c 0xffffffff<\r><\n>
3fc97a10: 0x3fc97a0c 0x3fc97a0c 0x00000000 0x3fc97a20 0xffffffff 0x3fc97a20 0x3fc97a20 0x00000000<\r><\n>
3fc97a30: 0x3fc97a34 0xffffffff 0x3fc97a34 0x3fc97a34 0x00000000 0x3fc97a48 0xffffffff 0x3fc97a48<\r><\n>
3fc97a50: 0x3fc97a48 0x00000000 0x3fc97a5c 0xffffffff 0x3fc97a5c 0x3fc97a5c 0x00000000 0x3fc97a70<\r><\n>
3fc97a70: 0xffffffff 0x3fc97a70 0x3fc97a70 0x00000000 0x3fc97a84 0xffffffff 0x3fc97a84 0x3fc97a84<\r><\n>
3fc97a90: 0x00000000 0x3fc97a98 0xffffffff 0x3fc97a98 0x3fc97a98 0x00000000 0x3fc97aac 0xffffffff<\r><\n>
3fc97ab0: 0x3fc97aac 0x3fc97aac 0x00000001 0x3fc97ac0 0xffffffff 0x3fca2ca0 0x3fca2ca0 0x00000000<\r><\n>
3fc97ad0: 0x3fc97ad4 0xffffffff 0x3fc97ad4 0x3fc97ad4 0x00000000 0x3fc97ae8 0xffffffff 0x3fc97ae8<\r><\n>
3fc97af0: 0x3fc97ae8 0x00000001 0x3fc97afc 0xffffffff 0x3fca7800 0x3fca7800 0x00000000 0x3fc97b10<\r><\n>
3fc97b10: 0xffffffff 0x3fc97b10 0x3fc97b10 0x00000000 0x3fc97b24 0xffffffff 0x3fc97b24 0x3fc97b24<\r><\n>
3fc97b30: 0x00000007 0x3fc97b38 0xffffffff 0x3fca50c8 0x3fcaa44c 0x00000000 0x3fc97b4c 0xffffffff<\r><\n>
3fc97b50: 0x3fc97b4c 0x3fc97b4c 0x00000000 0x3fc97b58 0x00000000 0x00000000 0x00000000 0x3fc97b70<\r><\n>
3fc97b70: 0xffffffff 0x3fc97b70 0x3fc97b70 0x00000000 0x3fc97b84 0xffffffff 0x3fc97b84 0x3fc97b84<\r><\n>
3fc97b90: 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000<\r><\n>
3fc97bb0: 0x3fc97bac 0x00000000 0x00000000 0x00000000 0x3fc97bc4 0xffffffff 0x3fc97bc4 0x3fc97bc4<\r><\n>
3fc97bd0: 0x00000000 0x3fc97bd8 0xffffffff 0x3fc97bd8 0x3fc97bd8 0x00000001 0x00000001 0x00000000<\r><\n>
3fc97bf0: 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca18c0 0x3fca1928 0x3fca1990<\r><\n>
3fc97c10: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x40052910 0x00000000<\r><\n>
3fc97c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000<\r><\n>
<\r><\n>
<\r><\n>
<\r><\n>
ELF file SHA256: 34dfe065f59136f2<\r><\n>
<\r><\n>
Rebooting...<\r><\n>_

What's wrong with the watch dog?

ustccw, 2022-11-29:
it seems like your wifi tx power is 70, right?  what if set AT+RFPOWER=84 and retry again.

mah-eiSmart, 2022-11-29:
AT+RFPOWER is 80, which should be the maximum according to documentation. The distance between the smart phone and the chip is about 50cm. I reduced the TX power to check whether there is cross talk, but no effect.

It's definitely a matter of the chip working as access point. In station mode it's much better.