espressif/esp-at #686: MQTTSUBRECV返回把一条mqtt消息分割了，而且内容还不对

zslzxm, 2022-11-29:
使用的标准AT固件：ESP32-WROOM-32_AT_Bin_V2.4.0.0

+MQTTSUBRECV:0,"xxx111/56000012/CollectPara/0/1",276,:4096,"TriggerType":1,"TriggerLimitList":[{"Channels":[1],"AccLimit":0.0,"VerLimit":0.0,"RpmLimit":0.0,"ShockLimit":0.0}]}],"RtuChannelList":[],"OutputChannelList":[{"ChNo":1,"IsOn":false,"DevId":0,"Limit":0.0},{"ChNo":2,"IsOn":true,"DevId":0,"Limit":0.0}],"StopLimitList":[]}
0/1",986,{"VibChannelList":[{"ChNo":1,"SpeedNo":0,"SensType":0,"CnstCurSrcState":0,"SensCoef":10.0,"DevId":1,"VibParaList":[{"LowFreq":0,"HighFreq":20000,"Length":0,"Unit":"m/s^2","PlusLength":0,"Dir":1,"Code":1},{"LowFreq":0,"HighFreq":12800,"Length":8192,"Unit":"m/s^2","PlusLength":0,"Dir":1,"Code":11},{"LowFreq":0,"HighFreq":1600,"Length":4096,"Unit":"m/s^2","PlusLength":2048,"Dir":1,"Code":12},{"LowFreq":0,"HighFreq":1600,"Length":2048,"Unit":"m/s^2","PlusLength":1024,"Dir":1,"Code":15},{"LowFreq":0,"HighFreq":1600,"Length":4096,"Unit":"m/s^2","PlusLength":2048,"Dir":3,"Code":12},{"LowFreq":0,"HighFreq":1600,"Length":2048,"Unit":"m/s^2","PlusLength":1024,"Dir":3,"Code":15},{"LowFreq":0,"HighFreq":1600,"Length":4096,"Unit":"m/s^2","PlusLength":2048,"Dir":2,"Code":12},{"LowFreq":0,"HighFreq":1600,"Length":2048,"Unit":"m/s^2","PlusLength":1024,"Dir":2,"Code":15}],"LimitParaList":[]}],"TechChannelList":[],"SpeedChannelList":[],"TriggerList":[{"TriggerGroup":1,"Freq":20000,"Length"
+MQTTSUBRECV:0,"xxx111/56000012/CollectPara/0/1",276,:4096,"TriggerType":1,"TriggerLimitList":[{"Channels":[1],"AccLimit":0.0,"VerLimit":0.0,"RpmLimit":0.0,"ShockLimit":0.0}]}],"RtuChannelList":[],"OutputChannelList":[{"ChNo":1,"IsOn":false,"DevId":0,"Limit":0.0},{"ChNo":2,"IsOn":true,"DevId":0,"Limit":0.0}],"StopLimitList":[]}

首先，能不能不要分割？
另外，就算分割，第一条也是错的啊，怎么解决？

zslzxm, 2022-11-29:
986这个是第一个分段，276是第二个分段，现在的结果是 276,986,276，这个没办法解析了

zslzxm, 2022-11-29:
忽略数据错误的说法，是我程序没搞对

但是，关于会分包的这个问题，有解决办法吗？有没有哪里可以设置这个大小的？

ustccw, 2022-11-29:
可以在这里设置：
`./build.py menuconfig` -> `Component config` -> `ESP-MQTT Configurations` -> `MQTT Using custom configurations` -> `Default MQTT buffer size`

zslzxm, 2022-11-29:
多谢大佬相助，我原本想省事直接用编译好的版本的，现在看来还得重新编译了，我先研究一下
Thanks!

closed 2023-04-20 by ustccw